NVIDIA/open-gpu-kernel-modules #735: No display on linux kernel 6.12

SheMelody, 2024-11-14:
### NVIDIA Open GPU Kernel Modules Version

565.57.01

### Please confirm this issue does not happen with the proprietary driver (of the same version). This issue tracker is only for bugs specific to the open kernel driver.

- [ ] I confirm that this does not happen with the proprietary driver package.

### Operating System and Version

Arch Linux

### Kernel Release

6.12.0-rc7-1-mainline #1 SMP PREEMPT_DYNAMIC Mon, 11 Nov 2024 05:44:29 +0000 x86_64 GNU/Linux

### Please confirm you are running a stable release kernel (e.g. not a -rc). We do not accept bug reports for unreleased kernels.

- [ ] I am running on a stable kernel release.

### Hardware: GPU

NVIDIA GeForce RTX 4070 SUPER (UUID: GPU-a1bb0f59-973a-389d-0857-63c4a8adce7b)

### Describe the bug

Installs successfully, but there's no display (X11/Wayland) except console. The login manager is unable to start.

### To Reproduce

1. Install linux mainline 6.12-rc7
2. Install nvidia open dkms modules
3. Boot into linux mainline 6.12-rc7

### Bug Incidence

Always

### nvidia-bug-report.log.gz

[nvidia-bug-report.log.gz](https://github.com/user-attachments/files/17749375/nvidia-bug-report.log.gz)

closed 2024-11-14 by SheMelody

ptr1337, 2024-11-14:
> Didn't actually see that you don't accept bug reports for rc. Closing.

There is a patch available, see here: https://github.com/NVIDIA/open-gpu-kernel-modules/issues/712#issuecomment-2475331834

I have pushed the fix already in the archlinux repository in "extra-testing".

SheMelody, 2024-11-14:
> > Didn't actually see that you don't accept bug reports for rc. Closing.
> 
> There is a patch available, see here: [#712 (comment)](https://github.com/NVIDIA/open-gpu-kernel-modules/issues/712#issuecomment-2475331834)
> 
> I have pushed the fix already in the archlinux repository in "extra-testing".

I saw in another issue, thank you so much. I'll wait for it to reach the stable repo.

SheMelody, 2024-11-14:
> > Didn't actually see that you don't accept bug reports for rc. Closing.
> 
> There is a patch available, see here: [#712 (comment)](https://github.com/NVIDIA/open-gpu-kernel-modules/issues/712#issuecomment-2475331834)
> 
> I have pushed the fix already in the archlinux repository in "extra-testing".

I confirm that the patch works, I downloaded the relevant files from the Arch Linux archive and installed them using `pacman -U ...`, effectively replacing the right packages. Thank you for the fix.

Binary-Eater, 2024-11-23:
@SheMelody I think the way to interpret our stance on rc related issues is feel free to open them but come in with the expectation that our releases may not be ready to handle rc kernels.

Binary-Eater, 2024-12-02:
Hi @SheMelody,

I am already making fixes for 6.13 (multiple needed) and hoping to have the changes in releases for our active driver branches before 6.13 is released. We are happy to take bug reports for 6.13, but we do not have the staffing currently to proactively provide patches for rc kernels publicly (multiple changes that are getting integrated in our development branch, run through QA, and then make their way into our release branches) while developing the DRM KMS stack as well as the OpenGL/Vulkan usermode components. As of right now, we are unable to proactively support rc kernels, but we are happy to take bug reports.

Binary-Eater, 2024-12-02:
For your specific issue, this is being tracked in #746.