NVIDIA/open-gpu-kernel-modules #738: NVRM: nvAssertFailedNoLog: Assertion failed: 0 @ osapi.c:1904 Accompanied with 30w power limits

rjojjr, 2024-11-15:
Hi, 

I am running the latest 565 open kernel module on the latest Pop OS on a Dell G16 7630 with RTX 4070 8GB. I have an issue in both Windows and Pop OS in which when the GPU is under load, the power for the card randomly drops down to 30W and later randomly returns back to the stock level. In pop os I have captured the following msg. from dmsg(corresponding to each power limit change):

` NVRM: nvAssertFailedNoLog: Assertion failed: 0 @ osapi.c:1904 Accompanied with 30w power limits`
 
 Any idea what the cause could be? This makes this laptop unbearable to use.

[nvidia-bug-report.log.gz](https://github.com/user-attachments/files/17765890/nvidia-bug-report.log.gz)

_Originally posted by @rjojjr in https://github.com/NVIDIA/open-gpu-kernel-modules/discussions/737_

rjojjr, 2024-11-15:
I should mention that the random 30w power limit occurs on Windows 11 with latest prop. "Game Ready" drivers

thesword53, 2024-12-16:
Hello,

I have the exact opposite issue even in both Linux and Windows: https://github.com/NVIDIA/open-gpu-kernel-modules/issues/752