NVIDIA/open-gpu-kernel-modules #746: Kernel 6.13: implicit declaration of function ‘crypto_akcipher_verify’; did you mean ‘crypto_akcipher_reqtfm’?

philmmanjaro, 2024-11-26:
### NVIDIA Open GPU Kernel Modules Version

565.57.01

### Operating System and Version

Manjaro Linux

### Kernel Release

6.13.0-rc0-1-MANJARO

### Please confirm you are running a stable release kernel (e.g. not a -rc). We do not accept bug reports for unreleased kernels.

- [ ] I am running on a stable kernel release.

### Build Command

sudo chrootbuild -p nvidia -b unstable

### Terminal output/Build Log

Build log:

```
(5/6) Updating linux initcpios...
==> Building image from preset: /etc/mkinitcpio.d/linux613.preset: 'default'
==> Using default configuration file: '/etc/mkinitcpio.conf'
  -> -k /boot/vmlinuz-6.13-x86_64 -g /boot/initramfs-6.13-x86_64.img
==> Starting build: '6.13.0-rc0-1-MANJARO'
  -> Running build hook: [base]
  -> Running build hook: [udev]
  -> Running build hook: [autodetect]
  -> Running build hook: [microcode]
  -> Running build hook: [modconf]
  -> Running build hook: [kms]
  -> Running build hook: [keyboard]
  -> Running build hook: [keymap]
  -> Running build hook: [consolefont]
==> WARNING: consolefont: no font found in configuration
  -> Running build hook: [block]
  -> Running build hook: [filesystems]
  -> Running build hook: [fsck]
==> Generating module dependencies
==> Creating zstd-compressed initcpio image: '/boot/initramfs-6.13-x86_64.img'
  -> Early uncompressed CPIO image generation successful
==> Initcpio image generation successful
==> Building image from preset: /etc/mkinitcpio.d/linux613.preset: 'fallback'
==> Using default configuration file: '/etc/mkinitcpio.conf'
  -> -k /boot/vmlinuz-6.13-x86_64 -g /boot/initramfs-6.13-x86_64-fallback.img -S autodetect
==> Starting build: '6.13.0-rc0-1-MANJARO'
  -> Running build hook: [base]
  -> Running build hook: [udev]
  -> Running build hook: [microcode]
  -> Running build hook: [modconf]
  -> Running build hook: [kms]
==> WARNING: Possibly missing firmware for module: 'ast'
==> WARNING: Possibly missing firmware for module: 'i915'
==> WARNING: Possibly missing firmware for module: 'nouveau'
==> WARNING: Possibly missing firmware for module: 'amdgpu'
==> WARNING: Possibly missing firmware for module: 'radeon'
==> WARNING: Possibly missing firmware for module: 'xe'
  -> Running build hook: [keyboard]
==> WARNING: Possibly missing firmware for module: 'xhci_pci_renesas'
  -> Running build hook: [keymap]
  -> Running build hook: [consolefont]
==> WARNING: consolefont: no font found in configuration
  -> Running build hook: [block]
==> WARNING: Possibly missing firmware for module: 'wd719x'
==> WARNING: Possibly missing firmware for module: 'aic94xx'
[... 573 lines cut ...]
   34 | bool libspdm_random_seed(const uint8_t *seed, size_t seed_size)
      |      ^~~~~~~~~~~~~~~~~~~
make[2]: *** [/usr/lib/modules/6.13.0-rc0-1-MANJARO/build/Makefile:1936: /build/linux613-nvidia/src/NVIDIA-Linux-x86_64-565.57.01-no-compat32/kernel-open] Error 2
make[1]: *** [Makefile:224: __sub-make] Error 2
make[1]: Leaving directory '/usr/lib/modules/6.13.0-rc0-1-MANJARO/build'
make: *** [Makefile:89: modules] Error 2
make: Leaving directory '/build/linux613-nvidia/src/NVIDIA-Linux-x86_64-565.57.01-no-compat32/kernel-open'
==> ERROR: A failure occurred in build().
    Aborting...
```

### More Info

With commit https://github.com/torvalds/linux/commit/6b34562f0cfe81f1f207fc7c146c4ff4b31eb625 the following changes were made upstream:

```
crypto: akcipher - Drop sign/verify operations
A sig_alg backend has just been introduced and all asymmetric
sign/verify algorithms have been migrated to it.

The sign/verify operations can thus be dropped from akcipher_alg.
It is now purely for asymmetric encrypt/decrypt.
```

Nvidia driver and Nvidia-Open driver needs to adopt to these changes.

ttabi, 2024-11-26:
"We do not accept bug reports for unreleased kernels."

Please don't ignore this line.

philmmanjaro, 2024-11-26:
Possible solution: https://gitlab.manjaro.org/packages/extra/linux613-extramodules/nvidia/-/commit/bcde3efc1d401c3763988c619cb85fb258044683

steven-bellock, 2024-11-26:
This has been tracked internally as bug 489882.

@philmmanjaro that workaround is fine if the system is not in confidential compute mode. In that case an even simpler workaround is just `err = 0`. However if the system is in confidential compute mode then things will fail.

Binary-Eater, 2024-11-26:
We are tracking this as well in internal bug 4898827.

SheMelody, 2024-12-02:
> This has been tracked internally as bug 4898827.
>
> [@philmmanjaro](https://github.com/philmmanjaro) that workaround is fine if the system is not in confidential compute mode. In that case an even simpler workaround is just `err = 0`. However if the system is in confidential compute mode then things will fail.
It also makes wrong symlinks so it doesn't install. To quickly workaround the problem I just made and ran this loop script while the kernel installed:
```sh
#!/bin/bash
while true
do
    cat /var/lib/dkms/nvidia/565.57.01/build/kernel-open/nvidia/nv-kernel.o > /dev/null
    if [[ $? != 0 ]]; then
        sudo rm /var/lib/dkms/nvidia/565.57.01/build/kernel-open/nvidia/nv-kernel.o
        sudo cp --force /var/lib/dkms/nvidia/565.57.01/build/src/nvidia/_out/Linux_x86_64/nv-kernel.o /var/lib/dkms/nvidia/565.57.01/build/kernel-open/nvidia/nv-kernel.o
    fi
    cat /var/lib/dkms/nvidia/565.57.01/build/kernel-open/nvidia-modeset/nv-modeset-kernel.o > /dev/null
    if [[ $? != 0 ]]; then
        sudo rm /var/lib/dkms/nvidia/565.57.01/build/kernel-open/nvidia-modeset/nv-modeset-kernel.o
        sudo cp --force /var/lib/dkms/nvidia/565.57.01/build/src/nvidia-modeset/_out/Linux_x86_64/nv-modeset-kernel.o /var/lib/dkms/nvidia/565.57.01/build/kernel-open/nvidia-modeset/nv-modeset-kernel.o
    fi
done
```
I know, it's probably not the best solution, but it works and I'm in.

philmmanjaro, 2024-12-02:
@SheMelody you can use this patch instead: https://gist.github.com/joanbm/d1f89391a4b20f4b56ba931ef6ca62da

SheMelody, 2024-12-03:
> [@SheMelody](https://github.com/SheMelody) you can use this patch instead: https://gist.github.com/joanbm/d1f89391a4b20f4b56ba931ef6ca62da

Thanks, that's basically the problem that was happening.

SheMelody, 2024-12-10:
For 565.77 I also had to comment `MODULE_IMPORT_NS(DMA_BUF);` at line 130 of `nv.c`. No idea what this will cause but I'll see. All the workarounds above are still required.

ptr1337, 2024-12-10:
> For 565.77 I also had to comment `MODULE_IMPORT_NS(DMA_BUF);` at line 130 of `nv.c`. No idea what this will cause but I'll see. All the workarounds above are still required.

Here a better way todo:
https://raw.githubusercontent.com/CachyOS/CachyOS-PKGBUILDS/refs/heads/nvidia-565-6.13/nvidia/nvidia-utils/0009-nvidia-nv-Convert-symbol-namespace-to-string-literal.patch

SheMelody, 2024-12-10:
> Here a better way todo: https://raw.githubusercontent.com/CachyOS/CachyOS-PKGBUILDS/refs/heads/nvidia-565-6.13/nvidia/nvidia-utils/0009-nvidia-nv-Convert-symbol-namespace-to-string-literal.patch

Thanks. Will there be problems if that entire line is just commented out instead? I'm having no issues so far but I'd like to know, because I'm too lazy and busy to reinstall.

l1k, 2024-12-26:
The patches linked above which replace `crypto_akcipher_verify()` with `crypto_akcipher_encrypt()` are wrong. They may unbreak the build, but they won't result in a working libspdm os_stub.

Signature generation/verification was moved out of `akcipher` into the new `sig` transform starting with v6.13-rc1:
https://lore.kernel.org/all/cover.1725972333.git.lukas@wunner.de/

So the proper solution is to call `crypto_sig_verify()` now. Be aware that it accepts kernel buffers instead of scatterlists.

I note that `lkca_ecdsa_verify()` constructs an ASN.1 encoding of the signature before verification:
https://github.com/NVIDIA/open-gpu-kernel-modules/blob/main/kernel-open/nvidia/libspdm_ecc.c#L281

This is now no longer necessary as the above-linked patch series introduced support for the P1363 encoding which is used natively by SPDM. To take advantage of it, you just need to amend `libspdm_ec_new_by_nid()` like this:

```
-        ctx->name = "ecdsa-nist-p256";
+        ctx->name = "p1363(ecdsa-nist-p256)";
```

And then you can just remove the manual construction of the ASN.1 encoding.

Support for P1363 was introduced in preparation for the in-kernel SPDM library:
https://lore.kernel.org/all/cover.1719771133.git.lukas@wunner.de/

closed 2025-03-02 by Binary-Eater