NVIDIA/open-gpu-kernel-modules #747: ld: cannot find ./nvidia/nv-kernel.o due to Kbuild changes in 6.13

philmmanjaro, 2024-12-02:
### NVIDIA Open GPU Kernel Modules Version

565.57.01

### Operating System and Version

Manjaro 24.2

### Kernel Release

6.13.0-rc1-1-MANJARO

### Please confirm you are running a stable release kernel (e.g. not a -rc). We do not accept bug reports for unreleased kernels.

- [ ] I am running on a stable kernel release.

### Build Command

The [Kbuild changes](https://github.com/torvalds/linux/compare/0e287d31b...6a34dfa15d6edf7e78b8118d862d2db0889cf669) merged in 6.13-rc1 prevent the nvidia drivers to compile.

### Terminal output/Build Log

```
==> Starting build()...
make: Entering directory '/build/linux613-nvidia/src/NVIDIA-Linux-x86_64-565.57.01-no-compat32/kernel-open'
make[1]: Entering directory '/usr/lib/modules/6.13.0-rc0-4-MANJARO/build'
make[2]: Entering directory '/build/linux613-nvidia/src/NVIDIA-Linux-x86_64-565.57.01-no-compat32/kernel-open'
warning: the compiler differs from the one used to build the kernel
  The kernel was built by: gcc (GCC) 14.2.1 20240910
  You are using:           cc (GCC) 14.2.1 20240910
  SYMLINK nvidia/nv-kernel.o
  SYMLINK nvidia-modeset/nv-modeset-kernel.o
 CONFTEST: hash__remap_4k_pfn
 CONFTEST: set_pages_uc
 CONFTEST: list_is_first
 CONFTEST: set_memory_uc
 CONFTEST: set_memory_array_uc
 CONFTEST: set_pages_array_uc
 CONFTEST: ioremap_cache
 CONFTEST: ioremap_wc
 CONFTEST: ioremap_driver_hardened
 CONFTEST: ioremap_driver_hardened_wc
 CONFTEST: ioremap_cache_shared
 CONFTEST: pci_get_domain_bus_and_slot
 CONFTEST: get_num_physpages
 CONFTEST: pde_data
 CONFTEST: xen_ioemu_inject_msi
 CONFTEST: phys_to_dma
 CONFTEST: get_dma_ops
 CONFTEST: dma_attr_macros
 CONFTEST: dma_map_page_attrs
 CONFTEST: write_cr4
 CONFTEST: of_find_node_by_phandle
 CONFTEST: of_node_to_nid
 CONFTEST: pnv_pci_get_npu_dev
 CONFTEST: of_get_ibm_chip_id
 CONFTEST: pci_stop_and_remove_bus_device
 CONFTEST: pci_rebar_get_possible_sizes
 CONFTEST: wait_for_random_bytes
 CONFTEST: register_cpu_notifier
 CONFTEST: cpuhp_setup_state
 CONFTEST: dma_map_resource
 CONFTEST: get_backlight_device_by_name
 CONFTEST: timer_setup
 CONFTEST: pci_enable_msix_range
 CONFTEST: kernel_read_has_pointer_pos_arg
 CONFTEST: kernel_write_has_pointer_pos_arg
 CONFTEST: dma_direct_map_resource
 CONFTEST: tegra_get_platform
 CONFTEST: tegra_bpmp_send_receive
[... 698 lines cut ...]
  CC [M]  nvidia-drm/nvidia-drm-format.o
  CC [M]  nvidia-drm/nvidia-drm-os-interface.o
  CC [M]  nvidia-drm/nvidia-drm-linux.o
  CC [M]  nvidia-peermem/nvidia-peermem.o
ld -r -o nvidia/nv-interface.o nvidia/nv.o nvidia/nv-pci.o nvidia/nv-dmabuf.o nvidia/nv-nano-timer.o nvidia/nv-acpi.o nvidia/nv-cray.o nvidia/nv-dma.o nvidia/nv-i2c.o nvidia/nv-mmap.o nvidia/nv-p2p.o nvidia/nv-pat.o nvidia/nv-procfs.o nvidia/nv-usermap.o nvidia/nv-vm.o nvidia/nv-vtophys.o nvidia/os-interface.o nvidia/os-mlock.o nvidia/os-pci.o nvidia/os-registry.o nvidia/os-usermap.o nvidia/nv-modeset-interface.o nvidia/nv-pci-table.o nvidia/nv-kthread-q.o nvidia/nv-memdbg.o nvidia/nv-ibmnpu.o nvidia/nv-report-err.o nvidia/nv-rsync.o nvidia/nv-msi.o nvidia/nv-caps.o nvidia/nv-caps-imex.o nvidia/nv-host1x.o nvidia/nv_uvm_interface.o nvidia/libspdm_aead.o nvidia/libspdm_ecc.o nvidia/libspdm_hkdf.o nvidia/libspdm_rand.o nvidia/libspdm_shash.o nvidia/libspdm_rsa.o nvidia/libspdm_aead_aes_gcm.o nvidia/libspdm_sha.o nvidia/libspdm_hmac_sha.o nvidia/libspdm_internal_crypt_lib.o nvidia/libspdm_hkdf_sha.o nvidia/libspdm_ec.o nvidia/libspdm_x509.o nvidia/libspdm_rsa_ext.o nvidia/nvlink_linux.o nvidia/nvlink_caps.o nvidia/linux_nvswitch.o nvidia/procfs_nvswitch.o nvidia/i2c_nvswitch.o
ld -r -o nvidia-modeset/nv-modeset-interface.o nvidia-modeset/nvidia-modeset-linux.o nvidia-modeset/nv-kthread-q.o
  LD [M]  nvidia.o
ld: cannot find ./nvidia/nv-kernel.o: No such file or directory
make[4]: *** [/usr/lib/modules/6.13.0-rc0-4-MANJARO/build/scripts/Makefile.build:387: nvidia.o] Error 1
make[4]: *** Waiting for unfinished jobs....
  LD [M]  nvidia-modeset.o
ld: cannot find ./nvidia-modeset/nv-modeset-kernel.o: No such file or directory
make[4]: *** [/usr/lib/modules/6.13.0-rc0-4-MANJARO/build/scripts/Makefile.build:387: nvidia-modeset.o] Error 1
make[3]: *** [/usr/lib/modules/6.13.0-rc0-4-MANJARO/build/Makefile:1989: .] Error 2
make[2]: *** [/usr/lib/modules/6.13.0-rc0-4-MANJARO/build/Makefile:251: __sub-make] Error 2
make[2]: Leaving directory '/build/linux613-nvidia/src/NVIDIA-Linux-x86_64-565.57.01-no-compat32/kernel-open'
make[1]: *** [Makefile:251: __sub-make] Error 2
make[1]: Leaving directory '/usr/lib/modules/6.13.0-rc0-4-MANJARO/build'
make: *** [Makefile:89: modules] Error 2
make: Leaving directory '/build/linux613-nvidia/src/NVIDIA-Linux-x86_64-565.57.01-no-compat32/kernel-open'
```

### More Info

When I find time, I might do a git-bisect to find the commit causing this.

aritger, 2024-12-02:
Thank you for reporting this.  We're also tracking this internally as NVIDIA bug 4989092.  If you can do a git-bisect, that would be very useful.  Thanks.

philmmanjaro, 2024-12-02:
Guy's you should step up your game and check. Seems the symlinks get done wrongly as mentioned here: https://github.com/NVIDIA/open-gpu-kernel-modules/issues/746#issuecomment-2511645255. For 470xx series the community already has a patch: https://gist.github.com/joanbm/d1f89391a4b20f4b56ba931ef6ca62da. Should apply to newer and older Nvidia drivers just fine.

philmmanjaro, 2024-12-02:
@aritger the issue is `commit 80f289101690 "kbuild: change working directory to external module directory with M=" (Masahiro Yamada, 10 Nov 2024)`

mtijanic, 2024-12-05:
> While it seems like it's considered acceptable by nvidia in this case, I'm curious if someone can explain to me why the bug reports for release candidates are not allowed, or why they would be accepted in this particular case?

Hi there. We'll need to revisit our policy on this, but I imagine it could look something roughly like:

- Bug reports for RC kernels are welcome as a way to provide a good, searchable location for anyone hitting the same issue and a place to post patches as above.
- RC kernels are not actually supported, and we may not respond to these bugs.

I believe there are two reasons for the original policy. One is that because the early RCs are sometimes very unstable, we don't want to spend time chasing down issues that are not part of our stack.

The other is that by far the overwhelming majority of these RC bug reports are due to some API changes that make the NV driver fail to build/load. This is something we are actively tracking internally, so these bug reports don't actually help us, it is just extra work to triage and respond. However, they do help everyone else, so I think the policy might get changed. Maybe we'll provide a separate bug template and tag for these issues.

philmmanjaro, 2024-12-05:
@mtijanic the RC1 is the most important kernel version there is. It normally includes all changes made in that kernel series. The following RCs are to polish the regressions or features introduced in that development cycle. Nvidia can simply join that development cycle similar as other companies like Intel and AMD do. Also post their fixes as soon as they have some to the public on given issues we may open. As distro maintainers we have to prepare our systems as soon as possible. Some distributions, especially rolling ones, might adopt software as soon as they got tagged stable.

Therefore I always ignored the rules of not posting bugs for not released kernels. It simply makes no sense to have such a rule, as you have to change the code anyway.

Sure, QA and official support might take longer for Nvidia to do. However, if Nvidia decides to do open kernel modules like AMD, Nvidia has to join and develop in the development cycle at given merge windows as everybody else has to do too.

So why not start early as the trend is in that direction anyway.

mtijanic, 2024-12-05:
As an out of tree driver we adapt our release (and thus QA) schedule not just to linux mainline, but also HW releases, security rollouts across multiple LTS driver branches, enterprise customer schedules, LTS kernel releases, and even Windows and the big game title releases. Unfortunately, this means we cannot always provide latest mainline compatible releases.

There's certainly an argument to be made that we can still handle this much better even within all the above constraints, such as maybe pushing the kernel-compatibility patches out of band as they are developed. We are actively discussing this internally and looking to improve the situation for bleeding edge rolling release distros.

philmmanjaro, 2024-12-05:
That would be really good. As most of the patches the user community provides get tested even on older Nvidia drivers which are EOL, as the 390xx series as an example. You can follow those efforts for example at the AUR from Arch:

https://aur.archlinux.org/cgit/aur.git/log/?h=nvidia-470xx-utils
https://aur.archlinux.org/cgit/aur.git/log/?h=nvidia-390xx-utils

Also Joan provides some patches for the 470xx driver series, as he still has to use that driver. Most of them also apply to newer or older drivers:

https://gist.github.com/joanbm

Most bleeding edge Distros are able to patch drivers as needed. The only issue here might be, that it might violate the licenses of the proprietary drivers. However, we only modify those source available parts anyway. So the binary blobs are never touched. Might be a gray area. So if that can be also added to the internal discussions it would help us a lot.

ttabi, 2024-12-05:
> So why not start early as the trend is in that direction anyway.

That's the point that Milos is making -- we already have started.  We track not only the -rc kernels internally, but also linux-next, so we are typically aware of -rc1 build failures before -rc1 is released.  So as he said:  "these bug reports don't actually help us, it is just extra work to triage and respond."

So no, you're not helping us by filing these bugs, and you're not telling us anything we don't already know.

philmmanjaro, 2024-12-06:
@ttabi the issue here is that Nvidia may know about the issues due to their internal bug tracker. However the Distro maintainers and users who want to use the latest kernels or have to ship the drivers in combination with the latest kernels would want to know about the issue and possible patches already existing. So having those issues publicly posted helps the community and users of the Nvidia drivers.

Posts like https://forums.developer.nvidia.com/t/patch-for-565-57-01-linux-kernel-6-12/313260 are great a great example to publish patches as needed as soon as they are known and created.

And even when the community is posting again issues you already know, the proper response would be: hey here is the patch, will be officially released in 565.xxx, which is scheduled for release at <some date>. We also track it in our internal tracker with <some number>. Apply the patch on your own risk, as we may change it as needed.

igorgue, 2025-01-22:
Got hit by this since update in NixOS just hit...

ptr1337, 2025-01-22:
> Got hit by this since update in NixOS just hit...

https://gitlab.archlinux.org/archlinux/packaging/packages/nvidia-utils/-/merge_requests/22

That is how we handle it now in archlinux. Its not sexy, but works!

dlouzan, 2025-04-01:
Just got bit by this simply updating f40 stable to latest available versions of kernel (`6.13.9-100.fc40`) and nvidia-driver (`3:565.57.01-1.fc40` from `@cuda-fedora40-x86_64`).

We had to downgrade the kernel to 6.8 to be able to build and use the driver.

barneysspeedshop, 2025-04-01:
Try `sudo dkms build -m nvidia -v (version # - i have 550.144.03) -k (kernel version - I have 6.14.0)`

So for example `sudo dkms build -m nvidia -v 550.144.03 -k 6.14.0`