NVIDIA/open-gpu-kernel-modules #749: fbdev option is not available on Linux 6.13-rc1

xtexChooser, 2024-12-07:
### NVIDIA Open GPU Kernel Modules Version

565.77

### Please confirm this issue does not happen with the proprietary driver (of the same version). This issue tracker is only for bugs specific to the open kernel driver.

- [ ] I confirm that this does not happen with the proprietary driver package.

### Operating System and Version

AOSC OS

### Kernel Release

Linux xtex1 6.13.0-aosc-main #1 SMP PREEMPT_DYNAMIC Wed Dec  4 20:00:43 UTC 2024 x86_64 GNU/Linux

### Please confirm you are running a stable release kernel (e.g. not a -rc). We do not accept bug reports for unreleased kernels.

- [ ] I am running on a stable kernel release.

### Hardware: GPU

GPU 0: NVIDIA GeForce GTX 750 Ti

### Describe the bug

`drm_aperture_remove_conflicting_pci_framebuffers` is removed in Linux 6.13-rc1 (https://github.com/torvalds/linux/commit/689274a56c0c088796d359f6c6267323931a2429), and `NV_DRM_FBDEV_AVAILABLE` is not defined:
https://github.com/NVIDIA/open-gpu-kernel-modules/blob/9d0b0414a5304c3679c5db9d44d2afba8e58cc1b/kernel-open/nvidia-drm/nvidia-drm-os-interface.h#L61-L69

### To Reproduce

Build on Linux kernel 6.13.0-rc1 and set `fbdev=1` parameter for `nvidia_drm` module.

### Bug Incidence

Always

### nvidia-bug-report.log.gz

```
DKMS make.log for nvidia-565.77 for kernel 6.13.0-aosc-main (x86_64)
Sat Dec  7 09:25:49 PM CST 2024
make[1]: Entering directory '/usr/src/linux-headers-6.13.0-aosc-main'
make[2]: Entering directory '/var/lib/dkms/nvidia/565.77/build'
  SYMLINK nvidia/nv-kernel.o
  SYMLINK nvidia-modeset/nv-modeset-kernel.o
 CONFTEST: hash__remap_4k_pfn
 CONFTEST: set_pages_uc
 CONFTEST: list_is_first
 CONFTEST: set_memory_uc
 CONFTEST: set_memory_array_uc
 CONFTEST: set_pages_array_uc
 CONFTEST: ioremap_cache
 CONFTEST: ioremap_wc
 CONFTEST: ioremap_driver_hardened
 CONFTEST: ioremap_driver_hardened_wc
 CONFTEST: ioremap_cache_shared
 CONFTEST: pci_get_domain_bus_and_slot
 CONFTEST: get_num_physpages
 CONFTEST: pde_data
 CONFTEST: xen_ioemu_inject_msi
 CONFTEST: phys_to_dma
 CONFTEST: get_dma_ops
 CONFTEST: dma_attr_macros
 CONFTEST: dma_map_page_attrs
 CONFTEST: write_cr4
 CONFTEST: of_find_node_by_phandle
 CONFTEST: of_node_to_nid
 CONFTEST: pnv_pci_get_npu_dev
 CONFTEST: of_get_ibm_chip_id
[... 644 lines cut ...]
  MODPOST Module.symvers
  CC [M]  nvidia.mod.o
  CC [M]  .module-common.o
  CC [M]  nvidia-uvm.mod.o
  CC [M]  nvidia-modeset.mod.o
  CC [M]  nvidia-drm.mod.o
  CC [M]  nvidia-peermem.mod.o
  LD [M]  nvidia.ko
  LD [M]  nvidia-uvm.ko
  LD [M]  nvidia-modeset.ko
  LD [M]  nvidia-drm.ko
  LD [M]  nvidia-peermem.ko
  BTF [M] nvidia-peermem.ko
Skipping BTF generation for nvidia-peermem.ko due to unavailability of vmlinux
  BTF [M] nvidia-modeset.ko
Skipping BTF generation for nvidia-modeset.ko due to unavailability of vmlinux
  BTF [M] nvidia-drm.ko
Skipping BTF generation for nvidia-drm.ko due to unavailability of vmlinux
  BTF [M] nvidia-uvm.ko
Skipping BTF generation for nvidia-uvm.ko due to unavailability of vmlinux
  BTF [M] nvidia.ko
Skipping BTF generation for nvidia.ko due to unavailability of vmlinux
make[2]: Leaving directory '/var/lib/dkms/nvidia/565.77/build'
make[1]: Leaving directory '/usr/src/linux-headers-6.13.0-aosc-main'
```

xtexChooser, 2024-12-07:
( I made a tentative fix for 565.77: https://gist.github.com/xtexChooser/da92d9df902788b75f746f348552ae80

SheMelody, 2024-12-11:
> ( I made a tentative fix for 565.77: https://gist.github.com/xtexChooser/da92d9df902788b75f746f348552ae80

This works.

closed 2025-03-02 by Binary-Eater