NVIDIA/open-gpu-kernel-modules #751: MODULE_IMPORT_NS should use string literal on Linux kernel 6.13-rc2+

xtexChooser, 2024-12-14:
### NVIDIA Open GPU Kernel Modules Version

565.77

### Operating System and Version

AOSC OS

### Kernel Release

6.13.0-aosc-main

### Please confirm you are running a stable release kernel (e.g. not a -rc). We do not accept bug reports for unreleased kernels.

- [ ] I am running on a stable kernel release.

### Build Command

dkms install

### Terminal output/Build Log

```
In file included from /usr/src/linux-headers-6.13.0-aosc-main/include/linux/module.h:22,
                 from nvidia/nv.c:24:
nvidia/nv.c:130:18: error: expected ‘,’ or ‘;’ before ‘DMA_BUF’
  130 | MODULE_IMPORT_NS(DMA_BUF);
      |                  ^~~~~~~
/usr/src/linux-headers-6.13.0-aosc-main/include/linux/moduleparam.h:26:61: note: in definition of macro ‘__MODULE_INFO’
   26 |                 = __MODULE_INFO_PREFIX __stringify(tag) "=" info
      |                                                             ^~~~
/usr/src/linux-headers-6.13.0-aosc-main/include/linux/module.h:299:33: note: in expansion of macro ‘MODULE_INFO’
  299 | #define MODULE_IMPORT_NS(ns)    MODULE_INFO(import_ns, ns)
      |                                 ^~~~~~~~~~~
nvidia/nv.c:130:1: note: in expansion of macro ‘MODULE_IMPORT_NS’
  130 | MODULE_IMPORT_NS(DMA_BUF);
      | ^~~~~~~~~~~~~~~~
```

### More Info

The parameter of `MODULE_IMPORT_NS` has been changed from constants to string literal: https://github.com/torvalds/linux/commit/cdd30ebb1b9f36159d66f088b61aee264e649d7a.

noshricardo, 2025-01-23:
This is still an issue on the 6.13 stable release.

ptr1337, 2025-01-23:
> This is still an issue on the 6.13 stable release.

https://gitlab.archlinux.org/archlinux/packaging/packages/nvidia-utils/-/commit/101b429bbf1ff62947f50a7e983d0086db8f2df9

Here is the patches, we use on archlinux for 6.13 support. 565 will not get a release, and the next expected release will be together with the client blackwell release.

noshricardo, 2025-01-23:
> https://gitlab.archlinux.org/archlinux/packaging/packages/nvidia-utils/-/commit/101b429bbf1ff62947f50a7e983d0086db8f2df9
>
> Here is the patches, we use on archlinux for 6.13 support. 565 will not get [...]

has this been applied to nvidia-open-dkms?

ptr1337, 2025-01-23:
> > > This is still an issue on the 6.13 stable release.
> > 
> > 
> > [gitlab.archlinux.org/archlinux/packaging/packages/nvidia-utils/-/commit/101b429bbf1ff62947f50a7e983d0086db8f2df9](https://gitlab.archlinux.org/archlinux/packaging/packages/nvidia-utils/-/commit/101b429bbf1ff62947f50a7e983d0086db8f2df9)
> > Here is the patches, we use on archlinux for 6.13 support. 565 will not get a release, and the next expected release will be together with the client blackwell release.
> 
> has this been applied to nvidia-open-dkms?

Yes, most of them are applied to nvidia-open. You can see it in the patch list in the PKGBUILD.

aritger, 2025-01-23:
My apologies for the delays in releasing a driver with Linux kernel 6.13 compatibility.  I hope a release will be made soon.  Thank you for your patience, and for the patches made by @ptr1337 .

For posterity, this MODULE_IMPORT_NS issue was tracked internally by NVIDIA as bug 5017197.

aritger, 2025-01-24:
Hi @pallaswept , sorry: the CUDA and graphics sides of the company aren't always able to perfectly coordinate their releases.  I think some 570 releases have started to trickle out from CUDA.  An official 570 release will hopefully be released soon.

mtijanic, 2025-01-25:
It sounds like a distro packaging issue. As far as I know nothing was pulled on the nvidia side. A 570.xx driver was published out-of-band with a CUDA toolkit release. Normally distros don't package these for regular users. My guess is that your distro packaged it up by mistake, assuming it was the actual driver release and then pulled it when they realized it's not. This is all outside of our control, so you really need to take it up with whoever was handling the packaging for your repos.

Sorry, I know this sucks but I really don't see what we could do here.

mtijanic, 2025-01-25:
Ah, understood, sorry. I'm not very versed in the CUDA release process, as evidenced by the fact that I too was caught by surprise with the 570 driver release 😃 .

I'll see if I can find someone to clarify what got (temporarily?) pulled and why.

closed 2025-03-02 by Binary-Eater