NVIDIA/open-gpu-kernel-modules #752: GPU stays at maximum power state when using multiple monitor

thesword53, 2024-12-15:
### NVIDIA Open GPU Kernel Modules Version

565.77 (older versions are also affected)

### Please confirm this issue does not happen with the proprietary driver (of the same version). This issue tracker is only for bugs specific to the open kernel driver.

- [ ] I confirm that this does not happen with the proprietary driver package.

### Operating System and Version

Arch Linux

### Kernel Release

6.12.1-arch1-1

### Please confirm you are running a stable release kernel (e.g. not a -rc). We do not accept bug reports for unreleased kernels.

- [x] I am running on a stable kernel release.

### Hardware: GPU

NVIDIA GeForce RTX 2080 SUPER

### Describe the bug

With certain combinations of monitor refresh rate, the GPU clocks won’t ramp down at idle (no app using the GPU open) even if I set:

- Powermizer mode to Adaptive or Auto in with both Xorg, Wayland and even TTY.

My GPU clocks never go bellow 1575 and 7750 Mhz at idle and `nvidia-smi` is reporting P0 in Perf section and 60W, it should be P8 and 22W at idle.

I have 3 monitors:

- Monitor 1: Alienware AW2725DF 2560x1440 360 Hz, Adaptive-Sync, DisplayPort
- Monitor 2: MSI MAG271CQR: 2560x1440 144 Hz, Adaptive-Sync, DisplayPort
- Monitor 3: Iiyama G2250HS: 1920x1080 75Hz, Adaptive-Sync, DisplayPort

The only configuration that doesn’t trigger the issue with more than one monitor is to only use monitor 2 at 144 Hz and monitor 3 at 75 Hz and disabling monitor 1 but this setup with VRR is causing another issue: ps://forums.developer.nvidia.com/t/monitors-literally-stutter-when-vrr-g-sync-is-enabled/256836

GPU doesn't even ramp down even if I am on a TTY console without any processes using GPU running.

### To Reproduce

- Have a RTX 2080 SUPER
AND
- Connect monitors 1, 2 and 3 and set them at 360 Hz, 144 Hz and 75 Hz respectively
OR
- Connect monitors 1 and 2 and set them at 360 Hz and 144 Hz respectively
OR
- Connect monitors 2 and 3 and set them at 144 Hz and 60 Hz respectively

### Bug Incidence

Always

### nvidia-bug-report.log.gz

[nvidia-bug-report.log.gz](https://github.com/user-attachments/files/18030466/nvidia-bug-report.log.gz)

### More Info

- That issue behave as if i enable `disable_vrr_memclk_switch` in `nvidia-modeset` module.
- With this issue, the GPU doesn't even go to P2 state when CUDA apps are running.

Binary-Eater, 2025-08-25:
@thesword53 can you do me a favor and confirm if you see the issue at the TTY console level if you set the `fbdev=0` kernel parameter for `nvidia-drm`?

ChristianKlass, 2026-09-07:
Not sure if this is the same bug you're all hitting or a slightly different one, so feel free to
tell me to open a separate issue.

Short version: I've got the same "GPU sits at max clocks forever" symptom on an RTX 4070, but
unlike the original report mine idles perfectly fine at the TTY console. What seems to trigger
it here is just having a Wayland compositor running at all, and I can't tie it to any particular
refresh rate combination.

Machine is an RTX 4070 on 610.57.04 (open kernel module), kernel 7.2.1-ogc4.1.fc44.x86_64 (Fedora
44 based, distro kernel with vendor patches, so let me know if that rules this out), Plasma 6.7.4 on
Wayland. `nvidia_drm` is `modeset=Y fbdev=Y`. Three screens, all on the 4070: 3440x1440@165 (DP),
1920x1080@144 rotated to portrait (DP), and 3840x1100@60 (HDMI).

Same three connectors enabled with DPMS on in all of these, and nothing using the GPU:

| state | pstate | core | memory | power | util |
|---|---|---|---|---|---|
| TTY console (nvidia-drm fbdev) | P8 | 210 MHz | 405 MHz | 10.4 W | 0% |
| KWin / Plasma session | P0 | 2880 MHz | 10501 MHz | 51-56 W | 0-1% |
| niri 26.04, every output at 60 Hz | P0 | 2880 MHz | 10501 MHz | 52 W | 0% |

Stopping the compositor puts it straight back to P8. When I started niri it went from 10.5 W to
52 W about two seconds after it grabbed the outputs, and stayed there doing nothing.

I also logged the whole boot once a second from a systemd unit ordered before the display manager.
`gpuclients` there is just the row count from `nvidia-smi --query-compute-apps`:

```
17:53:52  P5,  360 MHz,   810 MHz, 16.26 W, 0%  gpuclients=0   before display init
17:53:53  P0, 2850 MHz, 10501 MHz, 26.21 W, 0%  gpuclients=0   memory clock pins here
17:54:07  P0,  210 MHz, 10501 MHz, 28.61 W, 0%  gpuclients=1   login greeter up
17:54:16  P0,  210 MHz, 10501 MHz, 28.89 W, 0%  gpuclients=1   core still idling down
17:54:20  P0, 2850 MHz, 10501 MHz, 44.47 W, 2%  gpuclients=1   Plasma session starts
17:55:58  P0, 2880 MHz, 10501 MHz, 56.02 W, 1%  gpuclients=1   still pinned, doing nothing
```

The bit in the middle is what I find interesting. While the login greeter was up (which is a KWin
instance too) the core clock was still happily dropping to 210 MHz, but the memory clock had
already stuck at maximum. Once a full session starts, both stay up and never come back down.

Stuff I tried that made no difference:

- Refresh rate. niri brought all three outputs up at their preferred 60 Hz modes and it still
  pinned. Dropping the 144 Hz panel to 60 Hz under Plasma did nothing either.
- Monitor count. Single display at 60 Hz with the other two disabled, still P0.
- A client holding a decode context. I quit the browser that was using NVDEC, which left the
  compositor as the only context, and nothing changed.
- `nvidia-persistenced`, stopped it for 20s, no change.
- PowerMizer, `GPUPowerMizerMode` is 0 (adaptive), so it isn't forced to max performance.
- Blanking all the displays with DPMS off while the session runs, still P0 at about 48 W.

On the `nvidia-drm fbdev=0` question further up: I can't really answer it as asked, because I
never get into the state you're probing for, my console idles fine as it is. It might still be
useful though, since on this machine a console framebuffer that's actively scanning out clearly
isn't enough on its own to pin the clocks. `fbdev` is on and nvidia-drm owns my console:

```
fbcon: nvidia-drmdrmfb (fb0) is primary device
nvidia 0000:01:00.0: [drm] fb0: nvidia-drmdrmfb frame buffer device
fbcon: Taking over console
```

and with that live on all three connectors the card sits at P8 and 10.4 W. I can boot with
`nvidia-drm.fbdev=0` and report back if you need it.

thesword53, 2026-09-11:
The driver 615.71.09 is out and there is a new `nvidia` module parameter named `RmDisableDisplayGlitchPerfLimit` which fixed the issue for me on both Linux and Windows but when the GPU clock ramps up or down my 360Hz QHD monitor briefly loses video signal.

ElXreno, 2026-11-11:
Adding a data point that might narrow this down, because it happens here with a single display and no multi-monitor setup at all.

The machine is an MSI Titan 18 HX (MS-1824) with an RTX 5090 Laptop GPU (GB203M, 10de:2c18). The MUX is in discrete mode, so the internal eDP panel (BOE NE180QA1, 3840x2400, VRR 48-240 Hz) is driven by the NVIDIA card and nothing else is connected. Driver 620.70.06 (open modules), kernel 7.3.3, niri on Wayland.

At the panel's native 240 Hz the idle desktop sits in P0 with the memory clock at 14001 MHz and never leaves it. Switching the same panel to its 120 Hz mode, with VRR still on and no registry keys, makes the card start moving on its own within a few seconds:

```
240 Hz   P0   memory 14001 MHz   24.7 W
120 Hz   P8   memory   405 MHz   13-19 W   (walks P4/P5/P8, 15 memory clock changes in one minute)
```

Those switches at 120 Hz are invisible, including the jumps from 405 to 14001 MHz when a load starts.

I first assumed the vertical blanking was too short at 240 Hz, since both EDID modes share the same vertical total (2440, 40 blank lines), which gives 137 us of blanking at 120 Hz and only 68 us at 240 Hz. To test that I fed the driver an EDID override with a mode that keeps the 240 Hz pixel clock but doubles the blanking:

```
stock 240 Hz    pclk 2342.4 MHz   vtotal 2440   vblank  68 us   P0, pinned
custom 236 Hz   pclk 2342.4 MHz   vtotal 2480   vblank 137 us   P0, pinned
stock 120 Hz    pclk 1171.2 MHz   vtotal 2440   vblank 137 us   dynamic
```

The 236 Hz mode was accepted and displayed fine, but the card stayed in P0 for the whole test, so blanking time is not what decides it here. What is left is the pixel clock. My guess is that this mode is driven as 2Heads1OR and the driver treats it like the multi-head case from this issue, but I have not confirmed that the mode actually uses two heads.

With `RmDisableDisplayGlitchPerfLimit=1` the card does switch dynamically at 240 Hz, roughly ten memory clock changes a minute on an idle desktop. On this panel there is no signal loss like on the 360 Hz monitor mentioned above, but there is a visible stutter every time the card comes back up from P8 after sitting idle, which makes it unpleasant for desktop use.

The cost of the pinned state is also larger than nvidia-smi suggests. `nvidia-smi -lmc` works at runtime on this card, so I locked the memory clock at three levels and measured at the wall with a smart plug (CPU package power subtracted, three rounds of 60 s each, idle desktop):

```
memory clock   wall minus CPU package   nvidia-smi power.draw
14001 MHz      70.3 W                   25.4 W
 9001 MHz      63.4 W                   18.9 W
  810 MHz      52.5 W                   21.1 W
```

So the pinned P0 state costs about 18 W at the wall compared to 810 MHz, while nvidia-smi shows a difference of a few watts. On this laptop "GPU Memory Power Readings" and "Module Power Readings" are N/A, so the memory side is apparently not part of `power.draw`.

For what it is worth, 9001 MHz is perfectly smooth for a 4K 240 Hz desktop here, 810 MHz stutters badly and 405 MHz is unusable, so the hardware clearly does not need 14001 MHz to drive this panel.

Is the 2Heads1OR path what blocks the VRR based memory clock switch in this configuration? If it is, a single VRR panel driven by two locked heads looks like a case where the switch could be allowed, the same way it already works at 120 Hz.