NVIDIA/open-gpu-kernel-modules #760: egl-x11 confilict while install the beta driver

tom-cat-mao, 2025-01-03:
### NVIDIA Open GPU Kernel Modules Version

 565.57.01-1

### Operating System and Version

Arch Linux x86_64

### Kernel Release

Kernel: 6.12.7-arch1-1

### Please confirm you are running a stable release kernel (e.g. not a -rc). We do not accept bug reports for unreleased kernels.

- [ ] I am running on a stable kernel release.

### Build Command

yay -S nvidia-settings-beta

### Terminal output/Build Log

[image]

### More Info

I need to uninstall the egl-x11 and all the dependency to install nvidia-settings-beta 565.77-1
There might be something wrong with the dependency of your driver.
Maybe it should use the egl-x11 from the system or you need to provide your egl-x11.
Hope you can help me out.

fish4terrisa-MSDSM, 2025-01-03:
It's a dirtro related error, so probably shouldn't be asked at here.
However, a simple fix is to add `--overwrite '*'` after `yay -S nvidia-settings-beta`

ptr1337, 2025-01-03:
> It's a dirtro related error, so probably shouldn't be asked at here. However, a simple fix is to add `--overwrite '*'` after `yay -S nvidia-settings-beta`

Hi,

archlinux NVIDIA Package Maintainer. I would suggest you, as long there is no new beta to use the official packages, since these are properly maintained and also is better for bugreports to the offical nvidia tracker to have a base.

aritger, 2025-01-03:
Yes, this is a packaging issue, and out of scope of open-gpu-kernel-modules.  Closing; sorry.

closed 2025-01-03 by aritger