NVIDIA/open-gpu-kernel-modules #765: `nv_module_resources_init` tried to execute NX-protected page

SyntheticBird45, 2025-01-09:
### NVIDIA Open GPU Kernel Modules Version

565.77

### Please confirm this issue does not happen with the proprietary driver (of the same version). This issue tracker is only for bugs specific to the open kernel driver.

- [x] I confirm that this does not happen with the proprietary driver package.

### Operating System and Version

Artix Linux

### Kernel Release

Linux unknown 6.11.10-hardened1-1-hardened

### Please confirm you are running a stable release kernel (e.g. not a -rc). We do not accept bug reports for unreleased kernels.

- [x] I am running on a stable kernel release.

### Hardware: GPU

Nvidia Geforce RTX 4070 (nvidia-smi don't work since driver can't load)

### Describe the bug

On linux kernel 6.11.10 (and 6.12.6) ([linux-hardened](https://github.com/anthraxx/linux-hardened) patches applied), built with Clang CFI and Thin LTO, the open driver tries to execute an NX page.

I was able to reproduce this on 6.12.6 but [another unrelated bug](https://gitlab.freedesktop.org/drm/amd/-/issues/3787) make me unable to profit from any GPU. 
I downgraded to 6.11.10 but issue stay the same. 
This issue DO NOT happen on the official [linux-hardened arch linux package compiled with GCC](https://archlinux.org/packages/extra/x86_64/linux-hardened/).

Kernel stack trace (truncated for privacy concerns, can add additional details if required):
[dmesg_nvidia_open.log](https://github.com/user-attachments/files/18363445/dmesg_nvidia_open.log)

### To Reproduce

- Pull up a release from https://github.com/anthraxx/linux-hardened/releases.
- Configure kernel for using clang, enable CFI and Thin LTO.
- Compile it and install it
- Install NVIDIA open driver through DKMS or manually.
- `sudo modprobe nvidia`
- Should say `Killed`
- `dmesg` will give this stack trace.

### Bug Incidence

Always

### nvidia-bug-report.log.gz

No point since neither nvidia driver loads, neither is it runtime feature related. Also i don't like sharing my whole PCI tree in public github.

SyntheticBird45, 2025-01-11:
Thanks @TheBetterSolution, I'll try soon

SyntheticBird45, 2025-01-11:
Well doing this change is provoking a cascade of compile error (it compiles fine without the change):

[cantremovemacro.txt](https://github.com/user-attachments/files/18386147/cantremovemacro.txt)

SyntheticBird45, 2025-01-11:
I've tried commenting the macro but it then completely went haywire on the compile errors (see logs below):

[compilerror.txt](https://github.com/user-attachments/files/18386187/compilerror.txt)

I also tried to `#define NV_KTIME_GET_RAW_TS64_PRESENT` but as one could have guessed, the kernel just hanged at boot.

TheBetterSolution, 2025-01-12:
Excuse me @SyntheticBird45 
I researched the calls in nv_module_resources_init (and reference the offset) that execute the code outside the module, I guess  the exception is raised at NV_KMEM_CACHE_CREATE:
https://github.com/NVIDIA/open-gpu-kernel-modules/blob/9d0b0414a5304c3679c5db9d44d2afba8e58cc1b/kernel-open/nvidia/nv.c#L414

aritger, 2025-01-15:
Hi.  I don't know if it is related, but the mention of CFI here reminded me of this issue:
https://github.com/NVIDIA/open-gpu-kernel-modules/issues/439

In that issue, the problem is that the core of nvidia.ko isn't being built by kbuild, and thus isn't getting kbuild's extra CFLAGS for CFI.  In the current issue, I don't think we're even getting far enough to execute any of the "core" of nvidia.ko, so there is probably something else going on here.  But, you may trip on issue/439 once we resolve the current problem.  Or, something about the core of nvidia.ko not being built with kbuild's CFLAGS could be confusing things and triggering the current problem.

I see:
```
[ 1543.298921] Tainted: [O]=OOT_MODULE, [T]=RANDSTRUCT
```
It might be a useful experiment to test without RANDSTRUCT.

Looking at your call trace:
```
[ 1543.298943]  ? asm_exc_page_fault+0x26/0x30
[ 1543.298945]  ? nv_module_resources_init+0x80/0x160 [nvidia c7e7bb6b8f2be675d60a29e17e81cd37292395a3]
[ 1543.299005]  ? nv_module_init+0xf/0x150 [nvidia c7e7bb6b8f2be675d60a29e17e81cd37292395a3]
[ 1543.299047]  ? init_module+0x102/0x2cb [nvidia c7e7bb6b8f2be675d60a29e17e81cd37292395a3]
```
And @TheBetterSolution's speculation, I guess you are concerned about this path?
```
    nv_module_resources_init()
     NV_KMEM_CACHE_CREATE()
      nv_kmem_cache_create()
       nv_ktime_get_raw_ns()
        ktime_get_raw_ts64()
```
I suppose you could check the Module.symvers for your kernel (typically /usr/lib/modules/`uname -r`/build/Module.symvers) to find what module provides ktime_get_raw_ts64.  E.g., for me:
```
$ grep ktime_get_raw_ts64 ./build/Module.symvers
0x00000000      ktime_get_raw_ts64      vmlinux EXPORT_SYMBOL
```
(i.e., part of vmlinux, not a separate kernel module)

Was there specific reason to suspect ktime_get_raw_ts64(), or was that just speculation?

It might be easiest to sprinkle from printks in nv_module_resources_init() and its callees to determine where exactly the asm_exc_page_fault is happening.

    printk(KERN_ERR "file:%s, line:%d\n", __FILE__, __LINE__);