NVIDIA/open-gpu-kernel-modules #767: Expensive drmModeCloseFB due to framebuffers being destroyed immediately

Gert-dev, 2025-01-16:
### NVIDIA Open GPU Kernel Modules Version

565.77

### Please confirm this issue does not happen with the proprietary driver (of the same version). This issue tracker is only for bugs specific to the open kernel driver.

- [x] I confirm that this does not happen with the proprietary driver package.

### Operating System and Version

Arch Linux

### Kernel Release

6.12.8

### Please confirm you are running a stable release kernel (e.g. not a -rc). We do not accept bug reports for unreleased kernels.

- [x] I am running on a stable kernel release.

### Hardware: GPU

NVIDIA GeForce RTX 4090 Laptop GPU

### Describe the bug

(This is an upstream report of [mutter issue 3579](https://gitlab.gnome.org/GNOME/mutter/-/issues/3579).)

In mutter we are noticing that screen changes, e.g. caused by moving the mouse cursor or even spinner animations running, is causing rather high CPU usage located in `drmModeCloseFB`. Using sysprof to analyze the root, it ends up being due to `nv_drm_framebuffer_destroy` and its callees:

[image]

From the profile it seems as if every close of the framebuffer a buffer is immediately being destroyed. I'm not familiar with the callees in the profile, but the destroy may or may not be a red herring as it looks as if this code ends up relying on some kind of lock across threads underneath that blocks for longer than expected, resulting in high CPU usage.

This happens on single-GPU NVIDIA systems as well as systems with a monitor attached to the NVIDIA dGPU. As can be seen in the discussions on the ticket we did some testing on various NVIDIA drivers and it seems less problematic with the GSP firmware disabled on the closed driver. It's also not happening on other graphics drivers and seems to be unique to this driver.

### To Reproduce

1. Ensure you are on an empty desktop in GNOME.
2. Start a terminal, start `top`.
3. Observe baseline of `gnome-shell` CPU use. For me it is about 0% to 1%.
4. Keep moving the mouse over the background wallpaper whilst observing `top`.
5. Notice gnome-shell spiking. For me it goes to 15% or higher.

(Having a clean desktop/workspace ensures damage of applications that you are moving the mouse over is not influencing the result.)

### Bug Incidence

Always

### nvidia-bug-report.log.gz

[nvidia-bug-report.log.gz](https://github.com/user-attachments/files/18444301/nvidia-bug-report.log.gz)

Binary-Eater, 2025-01-25:
Thanks @Gert-dev for reporting this. We just opened NvBug 5074562 for tracking this.

Binary-Eater, 2025-02-01:
Just a heads up, we believe we have root caused the issue and are integrating the fix into current releases. Currently, nvidia-drm tries to idle the display engines whenever unregistering a surface. This is not necessary for nvidia-drm since we can guarantee all flips have been completed before nvidia-drm tries to unregister the surface.

Gert-dev, 2025-03-01:
Hi @Binary-Eater 

Did a fix for this by any chance land in 570.124.04? I'm asking because I don't think I can reproduce any more with this driver.

Binary-Eater, 2025-03-01:
@Gert-dev the fix did land in 570.124.04.

closed 2025-03-02 by Gert-dev