NVIDIA/open-gpu-kernel-modules #768: HDR distorted colours when using displayport, works fine when connected to HDMI

XapaJIaMnu, 2025-01-19:
### NVIDIA Open GPU Kernel Modules Version

nvidia-open-565.77-11 (From archlinux.org

### Please confirm this issue does not happen with the proprietary driver (of the same version). This issue tracker is only for bugs specific to the open kernel driver.

- [ ] I confirm that this does not happen with the proprietary driver package.

### Operating System and Version

Arch Linux

### Kernel Release

6.12.9-arch1-1

### Please confirm you are running a stable release kernel (e.g. not a -rc). We do not accept bug reports for unreleased kernels.

- [x] I am running on a stable kernel release.

### Hardware: GPU

NVIDIA GeForce RTX 3090

### Describe the bug

565.77 fixed HDMI HDR colours (which now work as expected), however if a monitor is connected to the display port, the colours are washed out (as described in this bug: https://github.com/NVIDIA/egl-wayland/issues/108 )
Monitor:  LG - UltraGear 27GP950 4K Ultra HD 27" 

Are you aware of the issue?

Thanks,

### To Reproduce

Take an HDR capable monitor.
Connect it via HDMI -> HDR works
Coneect it via DP       -> HDR washed out

Proprietary and Open Source drivers behave the same with this issue.

### Bug Incidence

Always

### nvidia-bug-report.log.gz

[nvidia-bug-report.log.gz](https://github.com/user-attachments/files/18466824/nvidia-bug-report.log.gz)

Binary-Eater, 2025-01-25:
Thanks for reporting this. We filed bug 5074583 internally for tracking this.

AlexGoinsNV, 2025-01-27:
@XapaJIaMnu ,

Are you able to describe in more detail, or possibly even take comparison photos of a screen that demonstrates the problem?

In https://github.com/NVIDIA/egl-wayland/issues/108, the problem was a lack of saturation (washed out). I know you are describing what you see as "washed out", but just to confirm, is it visually the same problem here (as seen on HDMI with pre-565 drivers), or is the display just dim but with adequate saturation?

XapaJIaMnu, 2025-01-28:
@AlexGoinsNV I'm away from my home for a few days so I can't provide pictures, but yes, it's literally the same behaviour as pre-565. I can provide some photos later on if necessary.

AlexGoinsNV, 2025-01-30:
@XapaJIaMnu , thanks! I would still be interested in photos when you are able.

So far I have been unable to replicate washed out colors on DP, though I haven't yet been able to test on your exact display. The fix for NVIDIA/egl-wayland#108 in 565 is working for me, both with HDMI and DP.

I did run into a problem where kwin was defaulting to a lower brightness on DP compared to HDMI on the same display, which was fixed by adjusting the "Display Brightness" slider in the system tray. Would you mind trying that and see if it helps?

XapaJIaMnu, 2025-01-31:
@AlexGoinsNV I've recorded a video, but it's really not obvious because the phone camera adapts to the colour space...

https://www.youtube.com/watch?v=lnjm5gT5hSI

AlexGoinsNV, 2025-01-31:
Thanks, @XapaJIaMnu .

You're right that the camera makes it hard to tell, but I think your video shows consistent color saturation between HDMI and DP. However, with DP the display is significantly dimmer. In your video, I can see that your secondary display appears brighter after you switch the primary display to DP. That tells me that the camera is adjusting to the lower brightness of your primary display after switching to DP, which should lend support to this theory.

With the issue in https://github.com/NVIDIA/egl-wayland/issues/108, you would see desaturated colors (i.e. washed out / too gray) as the display would interpret the incoming BT.2020 video signal as BT.709, but the overall display brightness (compare using a white screen, where BT.709 and BT.2020 match as they use the same D65 white point) would be unaffected.

Given the above, I think you may be seeing the same issue I encountered in https://github.com/NVIDIA/open-gpu-kernel-modules/issues/768#issuecomment-2625598222 . On the lower right, next to the volume symbol, click the "sun" icon and try adjusting the display brightness down and then back up. Does that help?

XapaJIaMnu, 2025-01-31:
Yes, sorry, thank you, @AlexGoinsNV .

It is indeed User error, the hdmi brightness is just better by default and the whole default hdmi settings work better. Closing the issue now as not an issue, thank you again for helping me identify it.

AlexGoinsNV, 2025-01-31:
Thanks for confirming.

I don't think this is necessarily user error. I found the same, that DP by default was set to a lower brightness even though I never adjusted it that way. It may be a bug with KDE / KWin.

As a side note, Wikipedia has an article on "colorfulness" which helps explain why a dimmer display would appear "washed out" to the eye: https://en.wikipedia.org/wiki/Colorfulness . Reduced brightness results in lower perceived "colorfulness", but "saturation" refers to "colorfulness of an area judged in proportion to its brightness".

XapaJIaMnu, 2025-01-31:
Thank you! That's so interesting! I guess the phone fixing the colour space should have been my first queue that it was a brightness issue and not a colour space issue.