NVIDIA/open-gpu-kernel-modules #773: nvidia driver 570.86.16 regression causes black screens when adaptive sync is enabled

ryao, 2025-02-01:
### NVIDIA Open GPU Kernel Modules Version

570.86.16

### Please confirm this issue does not happen with the proprietary driver (of the same version). This issue tracker is only for bugs specific to the open kernel driver.

- [ ] I confirm that this does not happen with the proprietary driver package.

### Operating System and Version

Gentoo Linux

### Kernel Release

6.12.6-gentoo-x86_64

### Please confirm you are running a stable release kernel (e.g. not a -rc). We do not accept bug reports for unreleased kernels.

- [x] I am running on a stable kernel release.

### Hardware: GPU

NVIDIA GeForce RTX 3090 Ti

### Describe the bug

Updating from 565.77 to 570.86.16 causes black screens whenever adaptive sync is enabled. Initially, I had a black screen in SDDM. I added the following entry to `/usr/share/nvidia/nvidia-application-profiles-570.86.16-rc`, which resolved the black screen in SDDM:

```
        { "pattern" : { "feature" : "procname", "matches" : "sddm-greeter-qt6" },           "profile" : "No VRR/OSD" },
```

I had been able to log into my machine by entering my password prior to doing that, but entering a password blindly was annoying, so I used that workaround.

After logging into my machine, I started Marvel Rivals, only to see another black screen. I went into KDE's system settings and set adaptive sync from automatic to never. This allowed me to see the contents of Marvel Rivals. I also could see it before change the adaptive sync setting when holding Alt+Tab, as that causes KDE's overlay appears, making kwin disable adaptive sync temporarily.

Finally, I tried setting Adaptive Sync to Always, which caused an immediate black screen.

### To Reproduce

1. Get a machine with a freesync/gsync/VRR screen. The screen must be configured with VRR on. Ideally, you would test this using a Dough Spectrum One (ES07D03) configured to use VRR (see more info), but other monitors likely will reproduce this.
2. Install a Linux distribution that uses KDE Plasma 6.2.5 / KDE Frameworks 6.10.0 / Qt version 6.8.1 (other versions likely also work for reproduction) and Nvidia driver 565.77
3. Boot it into SDDM, select a Wayland session and log into the machine.
4. Observe everything works fine, regardless of Adaptive Sync setting in KDE's settings. Leave Adaptive Sync at Automatic.
5. Upgrade to Nvidia driver 570.86.16
6. Observe that you now have a black screen at SDDM.
7. Observe that setting Adaptive Sync to Always causes a black screen. (hitting escape immediately will cancel the change; or you can wait for a 15 second timeout)
8. Observe that starting any game causes a black screen unless you set adaptive sync to Never.

### Bug Incidence

Always

### nvidia-bug-report.log.gz

[nvidia-bug-report.log.gz](https://github.com/user-attachments/files/18630522/nvidia-bug-report.log.gz)

### More Info

This also affects the Xorg X11 server given that it affects SDDM, which runs on top of the Xorg X11 server. However, I did not do of KDE on the Xorg X11 server to confirm it.

Furthermore, while SDDM appears on 565.77, it actually has some periodic flickering on my machine. Presumably, it needs the application profile too, although it just happens to work well enough with my Dough Spectrum monitor. The Dough Spectrum (ES07D03) is particularly finicky with Nvidia hardware on Linux, so I might see more problems on it than Nvidia QA does on other monitors.

Interestingly, my xorg.conf file has "metamodes" "3840x2160_144 +0+0 {AllowGSYNCCompatible=On}", but removing that line had no effect on the issue. I assume that is because it is obsolete at this point.

ptr1337, 2025-02-01:
Do you use hdmi? There has been some reports with HDMI and VRR, so far I know.
DP seems to work fine.

ryao, 2025-02-01:
> Do you use hdmi? There has been some reports with HDMI and VRR, so far I know. DP seems to work fine.

Yes. I am using HDMI. It is the only output that supports 144Hz 4K on my monitor.

adolfotregosa, 2025-02-03:
On your bug report you say it doesn't happen with proprietary driver.

Are you sure??

aritger, 2025-02-03:
Thanks for the report.  The 570 HDMI + VRR problems are being tracked internally within as bug 5088161.

My understanding is that issue affects both the closed and open kernel modules.  It would help confirm that what you are seeing is the same as bug 5088161, to test if the symptom described here also reproduces with the closed kernel modules.

Ac5000, 2025-02-04:
To add an additional data point: 

I too am having a similar set of issues as of today. Began once I updated `nvidia-open` from 565.77.12 -> 570.86.16-2

### Operating System and Version

Arch Linux x86_64

### Kernel Release

Linux 6.13.1-arch1-1

### Hardware: GPU

NVIDIA GeForce RTX 3090

### Description of the Issue and Troubleshooting Steps

Experiencing flickering on one of my attached monitors (both attached with DP connections.) Appears related to VRR. If I stop moving mouse/only show static material on my screen, the flickering stops. "Flickering" can be described as though my brightness on my monitor is bouncing +/- a couple percent brightness when going from static to refreshing. Disabling VRR removes the flickering issue.

ryao, 2025-02-04:
> On your bug report you say it doesn't happen with proprietary driver.
> 
> Are you sure??

I did not say that. I said that I did not test that.

> Thanks for the report. The 570 HDMI + VRR problems are being tracked internally within as bug 5088161.
>
> My understanding is that issue affects both the closed and open kernel modules. It would help [...]

I will make an effort to test this when I have some downtime.

My monitor is somewhat finicky with HDMI/DisplayPort link establishment with Nvidia hardware on Linux, so until the monitor manufacturer releases a firmware update that fixes that, testing different configurations with it requires me to properly isolate any driver regression from the monitor's behavior to avoid making a bad report.

cm-pony, 2025-02-05:
The same thing happens on the proprietary driver as well.
Kernel: Linux 6.13.1-arch1-1
GPU: NVIDIA GeForce RTX 4090

adolfotregosa, 2025-02-10:
> > On your bug report you say it doesn't happen with proprietary driver.
> > Are you sure??
> 
> I did not say that. I said that I did not test that.
> 

uhh ? You did.

[image]

ryao, 2025-02-22:
> > > On your bug report you say it doesn't happen with proprietary driver.
> > > Are you sure??
> > 
> > 
> > I did not say that. I said that I did not test that.
> 
> uhh ? You did by not testing it. It happens on both closed and open unfortunately.
> 
> [image]

That is why I left that box unchecked.

adolfotregosa, 2025-02-23:
> > > > On your bug report you say it doesn't happen with proprietary driver.
> > > > Are you sure??
> > > 
> > > 
> > > I did not say that. I said that I did not test that.
> > 
> > 
> > uhh ? You did by not testing it. It happens on both closed and open unfortunately.
> > [image]
> 
> That is why I left that box unchecked. Unchecked can either mean that it was not tested or it does happen.

...and you still don't get it! I will pass the word, as I was also taught by the mods myself. Look closely at what I highlighted bellow.

Please confirm this issue does not happen with the proprietary driver (of the same version). **_This issue tracker is only for bugs specific to the open kernel driver._**

ryao, 2025-03-04:
> > > > > On your bug report you say it doesn't happen with proprietary driver.
> > > > > Are you sure??
> > > > 
> > > > 
> > > > I did not say that. I said that I did not test that.
> > > 
> > > 
> > > uhh ? You did by not testing it. It happens on both closed and open unfortunately.
> > > [image]
> > 
> > 
> > That is why I left that box unchecked. Unchecked can either mean that it was not tested or it does happen.
> 
> ...and you still don't get it! I will pass the word, as I was also taught by the mods myself. Look closely at what I highlighted bellow.
> 
> Please confirm this issue does not happen with the proprietary driver (of the same version). **_This issue tracker is only for bugs specific to the open kernel driver._**
> 
> and this issue is NOT specific to the open kernel driver.

I realized that weeks ago after posting this, but they were happy to take the report as evidenced above. Most maintainers appreciate a well done bug report.

Ape, 2025-03-04:
The issue has been fixed in 570.124.04.

mtijanic, 2025-03-04:
Closing per ☝ .

As for the checkbox in the bug template, it is not really serving much purpose anymore now that the open modules are the primary flavor, and even the closed ones use GSP where possible. We should edit the template to be clearer.

closed 2025-03-04 by mtijanic

Kessi11, 2025-06-23:
Hey there, I would like to inquire how confident you are that this is actually fixed. Thing is, I am still getting this exact black with adaptive sync enabled on 575.57.08, but only after starting from standby (or at least screen turned off due to inactivity).

If you want me to, I will gladly file a new issue with all details I have, but so far I found little to go on.