NVIDIA/open-gpu-kernel-modules #777: Stutter in KDE Plasma Desktop when using GSP firmware

svenmauch, 2025-02-03:
### NVIDIA Open GPU Kernel Modules Version

570.86.16

### Please confirm this issue does not happen with the proprietary driver (of the same version). This issue tracker is only for bugs specific to the open kernel driver.

- [x] I confirm that this does not happen with the proprietary driver package.

### Operating System and Version

Arch Linux 6.13.1-arch1-1

### Kernel Release

Linux 6.13.1-arch1-1

### Please confirm you are running a stable release kernel (e.g. not a -rc). We do not accept bug reports for unreleased kernels.

- [x] I am running on a stable kernel release.

### Hardware: GPU

NVIDIA GeForce RTX 3070

### Describe the bug

Some actions in the KDE Plasma desktop environment suffer from very noticeable stutter. For example opening the application launcher, resizing windows (especially Firefox with a loaded website) and mouse movements on the lower refresh monitors.

### To Reproduce

Occurs with the package ``nvidia-open``

Does not occur with the package ``nvidia`` and kernel parameter ``nvidia.NVreg_EnableGpuFirmware=0`` set

### Bug Incidence

Always

### nvidia-bug-report.log.gz

(will provide shortly)

### More Info

KDE Plasma 6.2.5-1
KWin 6.2.5-3
Wayland
2x 60 Hz monitor, 1x 144 Hz monitor

I opened this issue because https://github.com/NVIDIA/open-gpu-kernel-modules/issues/693 seems like a different bug pattern and in https://github.com/NVIDIA/open-gpu-kernel-modules/issues/538 we were asked to create a new issue should it arise again.

mtijanic, 2025-02-04:
Hey there! Yes, thank you for opening a new issue, both of those are not the right place to put this.

Unfortunately, there is not much to go on in your report above. This issue happens when there are two programs hitting the GSP at the same time and serializing on that. Under normal circumstances nothing is actively talking to GSP, so when you e.g. start a new app it has free passage to it.

However, there are tools like the KDE system monitor widget that will constantly poke GSP in the background. Again, during steady state this monitor is the only thing touching GSP so there is no contention, but as soon as a second caller comes along things can get stuttery. Do you know if you're running any such monitoring? Things like coolercontrol, ksysmon/ksysguard, `nvidia-smi pmon`, mangohud?

It would be helpful to provide a full list of `ps aux` in addition to the `nvidia-bug-report.log`. Feel free to send it to `linux-bugs@nvidia.com` instead of uploading here.

Hopefully this is not a case of some script doing `while :; do nvidia-smi -q | something; sleep 1; done` where the monitoring program constantly comes and goes, because then we won't be able to capture it in a snapshot. If the above doesn't reveal anything, I'll kindly ask you to run a tracing script and reproduce the problem.

Thanks for reporting!

svenmauch, 2025-02-04:
Hi, thank you for the quick repsonse!

As far as I can tell, I don't have any monitoring tools or widgets running. I remember the stutter being much worse with version 560, but it's still definitely there and easy to distinguish from the performance with the proprietary modules. The animation when maximizing Dolphin windows is another easy way to reproduce the stutter, a program which is otherwise super smooth.

I have sent both files to ``linux-bugs@nvidia.com`` since I couldn't immediately tell how much sensitive information the logs might contain. Thank you for offering this!

svenmauch, 2025-02-04:
I sent the logs again with your mail in CC. The sender address is sven@mauch.me. Thank you.

mtijanic, 2025-02-04:
Nope, sorry. Neither the bugs list nor personal mailbox. Not in the spam folders either. I'll check with our IT.

mtijanic, 2025-02-04:
Okay, email issue resolved (will send details privately), got the logs. I'll look them over in a day or two and get back to you.

mtijanic, 2025-02-05:
From the logs, I do see a thing where we are hitting the GSP a lot more than we should. We actually fixed this already while debugging #650 but it didn't make the cut for this release. It should be in one of the next 570.xx releases. I don't know if this will solve the issue fully though.

If you're willing to try, I can extract it as a patch and you can build your own nvidia_modeset.ko and see if it fixes it. Or, wait a bit for the official release and try that.

If that doesn't fix it, I would have to ask you to run a bpftrace script and reproduce the problem and send those logs as well.

Thanks again for the report and the logs!

svenmauch, 2025-02-06:
I am travelling at the moment and could test a patch on 11.02. if that helps. Otherwise I'd wait for the release and use the old modules in the meantime.
Thank you for looking into this!

VeilSilence, 2025-02-08:
> 
> If you're willing to try, I can extract it as a patch and you can build your own nvidia_modeset.ko and see if it fixes it. Or, wait a bit for the official release and try that.
> 
I'm also experiencing lags and stuttering with the open NVIDIA driver (570) + GSP enabled on KWin Wayland, but I'm also facing another issue.

My "average time to render a frame" in OBS stats is much higher with the open NVIDIA driver + GSP enabled compared to GSP disabled/proprietary driver.

<details><summary>GSP OFF</summary>
<p>

[image]

</p>
</details> 

<details><summary>GSP ON</summary>
<p>

[image]

</p>
</details> 

Sometimes, OBS even misses frames due to rendering lags in the NVIDIA open driver. The average time constantly changes from 0.6 to ~3.5 ms or higher. I'm not sure if this is the same issue as described above, so I might need to create a separate issue about that.

Can you share the patch you mentioned earlier so I can test if it fixes my issue with OBS?

L-Acacia, 2025-02-17:
I have the same issue on Nixos with kernel 6.13.2, driver version 570.86.16, and plasma 6.3.0 if I turn on the GSP firmware.
I can also replicate it on arch. If you need any help to test patches, I would be glad to help.

urbenlegend, 2025-02-27:
Testing with 570.124.04-1 on Arch with the GSP on still shows stuttering while moving windows around on the KDE desktop. They're small microstutters that occur once every 2-3 seconds or so.

mtijanic, 2025-02-27:
Thanks for that report. 570.124 does contain the fixes for the issues reported in the OP, so if there are still stutters persisting, we'll need a new batch of profiling data. I'll provide the tools to do this in the next few days and will kindly ask you to take it for a spin.

Till then, two questions:
- @urbenlegend can you check what the output of `nvidia-smi --query-gpu="pstate" --format=csv` when it stutters? And if it's >= P5, can you please run something like `__GL_SYNC_TO_VBLANK=0 glxgears` to get the GPU to full power and see if it still stutters while glxgears is running in the background.

- @svenmauch Can you check the latest 570.124 and see if the original issue still repros?

Thanks!

urbenlegend, 2025-02-28:
@mtijanic I ran `watch -n 1 nvidia-smi --query-gpu="pstate" --format=csv` to run that command on a loop and yes during the stutters, it never got higher than a P5.

Once I ran `__GL_SYNC_TO_VBLANK=0 glxgears`, then it went straight to P0 and I no longer get the stutters.

Please keep in mind, the stutters can be subtle sometimes, like bad frame-pacing in a video game, but you can usually tell the difference by continuously dragging a window around for a few seconds. Eventually, the motion of the window will judder, whereas with GSP off, it remains perfectly smooth the entire time.

I tested with a Brave browser window this time around, but I have replicated it with all types of windows.

mtijanic, 2025-02-28:
Thanks @urbenlegend. Since the stutters go away at P0, I'm thinking this might be the same root cause as #650 (nvbug 4830125). We are testing a fix for that, but since it's a bit involved it might not make it into a 570.xx release and will be for one of the next major ones, so it can get more QA cycles.

I'll think about a good way to verify that it is indeed the same bug, so we don't waste time waiting for a fix that turns out to be irrelevant.

svenmauch, 2025-03-01:
I just updated to 570.124.04 and to me, the performance on GSP seems indistinguishable from the old modules now. Thank you!

There's still some stutters when coming from low power states (starting to scroll in firefox after idling a bit), but since these also occur with the old modules, they are probably tracked internally, correct?

DragonSWDev, 2025-03-02:
I just tried 570.124.04 on RTX 3060 and it's noticeably better than it was before. There are still stutters after system idles for some time but after a while everything is smooth just like with closed kernel module with disabled GSP.

mtijanic, 2025-03-03:
Thanks all, I'm going to close this issue as fixed in 570.122.04 (fyi, relevant code [here](https://github.com/NVIDIA/open-gpu-kernel-modules/blob/129479b1b7345f0055016ab3aa42ae78a4ca2048/src/nvidia-modeset/interface/nvkms-api.h#L2486-L2495) and [here](https://github.com/NVIDIA/open-gpu-kernel-modules/blob/129479b1b7345f0055016ab3aa42ae78a4ca2048/src/nvidia-modeset/kapi/src/nvkms-kapi.c#L2342-L2348)).

There is an ongoing issue where running at low power adds a delay in vblank notification delivery to the compositor, resulting in dropped frames, which is tracked in #650 .

There is also the ongoing issue where the GPU being in _deep_ idle mode is slower to wake than with GSP enabled. Some explanation here: https://github.com/NVIDIA/open-gpu-kernel-modules/issues/693#issuecomment-2295877351. This one is going to be a slow burn and I expect it will slowly be getting better over the next several releases. Because there is simply more work that needs to be done to wake the GPU now, I can't promise that it will ever be as good as with GSP-disabled (especially since the improvements we make will likely improve the other case as well, so it's a moving target), but I hope we'll get to a point where it will not be a hindrance. 

Please open a new issue if there is any stuttery behavior still present that is not covered by the two explanations above.

closed 2025-03-03 by mtijanic