NVIDIA/open-gpu-kernel-modules #779: HDR not working with RTX 5080

takemikazuchi-mar, 2025-02-08:
### NVIDIA Open GPU Kernel Modules Version

570.86.16

### Please confirm this issue does not happen with the proprietary driver (of the same version). This issue tracker is only for bugs specific to the open kernel driver.

- [ ] I confirm that this does not happen with the proprietary driver package.

### Operating System and Version

Bazzite 41 (FROM Fedora Kinoite)

### Kernel Release

6.12.12-203.bazzite.fc41.x86_64

### Please confirm you are running a stable release kernel (e.g. not a -rc). We do not accept bug reports for unreleased kernels.

- [x] I am running on a stable kernel release.

### Hardware: GPU

RTX 5080

### Describe the bug

When turning HDR on KDE Plasma, the output looks washed out and it seems like HDR isn't turning on (my TV doesn't show the HDR logo).
It used to work on my RTX 3080 with the same drivers.

### To Reproduce

Turn HDR on the settings

### Bug Incidence

Always

### nvidia-bug-report.log.gz

[nvidia-bug-report.log.gz](https://github.com/user-attachments/files/18717620/nvidia-bug-report.log.gz)

AlexGoinsNV, 2025-02-10:
Hi @takemikazuchi-mar ,

Bug 5099580 has been filed internally to track this issue.

Unfortunately, I haven't been able to reproduce your problem. Would you mind sharing some more details?

- What display are you using, and is it connected via HDMI or DP? You state that it's a TV, so I would guess HDMI, but are you using the native HDMI port, or perhaps a DP to HDMI active adapter?

- What version of KDE / kwin are you using?

- If you adjust the display brightness in the system tray ("sun" logo next to volume/bluetooth/clock), does it make a difference? What about the "SDR Brightness" setting in the Display Configuration panel?

takemikazuchi-mar, 2025-02-11:
Hello @AlexGoinsNV 

I'm using a LG C2 42 inch, the RTX 5080 is connected with HDMI.
The KDE Plasma version is 6.2

For the record, it was working in the same environnement with a RTX 3080, also in HDMI.

When using a displaylink adapter, HDR works. But with HDMI, it doesn't. The HDR just doesn't activate on the screen.

AlexGoinsNV, 2025-02-22:
Thanks, @takemikazuchi-mar . The bug is root caused and will be fixed in a future release.

clinche, 2026-08-19:
I think I've found the root cause. If your issue is still happening, would you mind testing and confirming the one-line fix in #1308 works? 

Thanks