NVIDIA/open-gpu-kernel-modules #781: DLSS with Frame generation caps GPU clock to 780MHz

zebulon2, 2025-02-11:
### NVIDIA Open GPU Kernel Modules Version

570.86.16

### Please confirm this issue does not happen with the proprietary driver (of the same version). This issue tracker is only for bugs specific to the open kernel driver.

- [x] I confirm that this does not happen with the proprietary driver package.

### Operating System and Version

Arch Linux

### Kernel Release

Linux arch 6.13.2-arch1-1 #1 SMP PREEMPT_DYNAMIC Sat, 08 Feb 2025 18:54:55 +0000 x86_64 GNU/Linux

### Please confirm you are running a stable release kernel (e.g. not a -rc). We do not accept bug reports for unreleased kernels.

- [x] I am running on a stable kernel release.

### Hardware: GPU

GPU 0: NVIDIA GeForce RTX 5080

### Describe the bug

In some games like Cyberpunk 2077 GPU clock tops to 780MHz. This dos not happen with some other games, but has also been reported on the nvidia forums for Hogwarts Legacy (see https://forums.developer.nvidia.com/t/clocking-issues-in-some-games-with-rtx-5080/322364).

In Cyberpunk 2077, and using Proton Experimental, Hotfix or proton-ge, the bug only happens when using DLSS Frame generation. With DLSS without frame generation, then frequency and power are boosted (to 2737MHz), even with Reflex enabled. However, restarting the game is necessary after changing these options, which makes testing tedious (especially one cannot skip the intro screens). There are also some weirdness, depending on the proton version, DLSS options are different (but I gather it may be the capability for these proton versions to tell the software DLSS3 is supported). And even after a restart, sometimes the texture quality option (low/high) under DLSS frame generation appears, or not. Same with the frame generation factor (2,3,4X). 

### To Reproduce

Launch Cyberpunk 2077 under Proton Experimental or Proton-ge, with mangohud, and change Graphical settings to: everything max, DLSS on, DLSS frame generation on (2, 3 or 4X). The GPU clock is locked to 780 MHz as soon as you reach the main menu.

### Bug Incidence

Always

### nvidia-bug-report.log.gz

[nvidia-bug-report.log.gz](https://github.com/user-attachments/files/18758803/nvidia-bug-report.log.gz)

zebulon2, 2025-02-12:
I have some more information that may help scoping the problem. Actually the 780 MHz lock up only happens when restarting Cyberpunk with DLSS frame generation 4X mode activated. If it is set to 2X or 3X, and the game is restarted, then the boost works (2737 Hz). Does this help? Can other people with other games try to set DLSS 2X or 3X in settings?

zebulon2, 2025-02-12:
Update: the plot thickens: it may happen during game in 3X after all. However, contrary to 4X, this does not happen immediately, but at some point during play, without reproducibility. Once it is locked to 780 MHz, however, there is no coming back, even if settings reduce frame generation (or disabling it), until game is restarted. It is as if there was some throttling happening and then lock down to 780 MHz.

zebulon2, 2025-02-13:
Another user has reported the issue here: https://forums.developer.nvidia.com/t/clocking-issues-in-some-games-with-rtx-5080/322364/9. He wrote:

In Hogwarts Legacy, which will allow you to change settings in game without restarting, this issue can be bypassed by disabling frame generation / relfex, which will immediately cause the GPU to throttle back up to full power, then reenabling frame gen / reflex and the GPU will stay at full power and perform as expected.
However, somethimes after some gaming, at random unforseeable intervals, the GPU will still fall back to this 780 MHz limit and stay there until doing this disable / enable cycle again.

Note that as far as I know one cannot disable frame gen and reflex in game in Cyberpunk (it requires restarting the game).

Armhole1225, 2025-02-19:
I'm experiencing the same issue. with the 5080 on Arch Linux with several games.

Armhole1225, 2025-02-20:
> Problem has been reproduced at nvidia and engineers are looking into it. Hopefully this will be fixed in the next driver release.

Thanks for the update.

zebulon2, 2025-03-21:
Apparently this is fixed in 570.133.07? I am awaiting for a package in Archlinux to test this.

ptr1337, 2025-03-21:
> Apparently this is fixed in 570.133.07? I am awaiting for a package in Archlinux to test this.

Ive pushed them to extra-testing already. You can also manually pull it

zebulon2, 2025-03-26:
> Ive pushed them to extra-testing already. You can also manually pull it

Thank you very much. And yes it is fixed, so I close the bug. Many thanks to developers.

closed 2025-03-26 by zebulon2