NVIDIA/open-gpu-kernel-modules #788: fbdev parameter is not available on Linux 6.14-rc3

XZVB12, 2025-02-18:
### NVIDIA Open GPU Kernel Modules Version

570.86.16

### Please confirm this issue does not happen with the proprietary driver (of the same version). This issue tracker is only for bugs specific to the open kernel driver.

- [ ] I confirm that this does not happen with the proprietary driver package.

### Operating System and Version

Manjaro Linux 25 / Arch Linux

### Kernel Release

6.14-rc3

### Please confirm you are running a stable release kernel (e.g. not a -rc). We do not accept bug reports for unreleased kernels.

- [ ] I am running on a stable kernel release.

### Hardware: GPU

NVIDIA GeForce RTX 3060

### Describe the bug

```
sudo cat /sys/module/nvidia_drm/parameters/fbdev
return No such file or directory
```

This problem has been repeated many times, for example for 6.13  https://github.com/NVIDIA/open-gpu-kernel-modules/issues/749

Maybe someone can publish a fix as it was before.
thanks

### To Reproduce

1)start system

2) sudo cat /sys/module/nvidia_drm/parameters/fbdev

### Bug Incidence

Always

### nvidia-bug-report.log.gz

[nvidia-bug-report.log.gz](https://github.com/user-attachments/files/18847110/nvidia-bug-report.log.gz)

Binary-Eater, 2025-02-18:
This happens when the upstream kernel changes the interface for DRM fbdev. What we do is handle the changes needed in our future releases. For example, 570.86.16 handles 6.13. we already have an internal fix for this, so a future release will handle 6.14. We don't support rc candidates at the time a rc candidate is tagged but appreciate the bug reports none the less.

XZVB12, 2025-02-18:
> we already have an internal fix for this

Can you publish it? This would allow you to use the system right now and not wait driver release, as it is not even known when this will happen.

ptr1337, 2025-02-18:
> > we already have an internal fix for this
> 
> Can you publish it? This would allow to use the system right now and not wait driver release, as it is not even known when this will happen.

https://raw.githubusercontent.com/CachyOS/CachyOS-PKGBUILDS/92308660049a15c326a59ad169e9c5b7b26085ee/nvidia/nvidia-utils/0003-drm-Remove-driver-data-from-struct-drm_driver.patch

Here you can find a patch, which should make it work.

XZVB12, 2025-02-27:
fixed in 570.124.04

closed 2025-02-27 by XZVB12