NVIDIA/open-gpu-kernel-modules #795: Failure to resume with 570.86.16 beta and 6.13.4

birdie-github, 2025-02-26:
### NVIDIA Open GPU Kernel Modules Version

570.86.16

### Please confirm this issue does not happen with the proprietary driver (of the same version). This issue tracker is only for bugs specific to the open kernel driver.

- [ ] I confirm that this does not happen with the proprietary driver package.

### Operating System and Version

Fedora release 41 (Forty One)

### Kernel Release

Linux zen 6.13.4-zen3 #2 SMP PREEMPT_DYNAMIC Tue Feb 25 10:32:20 UTC 2025 x86_64 GNU/Linux

### Please confirm you are running a stable release kernel (e.g. not a -rc). We do not accept bug reports for unreleased kernels.

- [ ] I am running on a stable kernel release.

### Hardware: GPU

GPU 0: NVIDIA GeForce RTX 4070 SUPER

### Describe the bug

I upgraded both the kernel and the driver, so I've no idea what led to this regression but on resume the system is dead:
```
 kernel: NVRM: gpuWaitForGfwBootComplete_TU102: failed to wait for GFW_BOOT: (progress 0x3)
 kernel: NVRM: kgspWaitForGfwBootOk_TU102: failed to wait for GFW boot complete: 0x55 VBIOS version 95.04.69.40.36
 kernel: NVRM: kgspWaitForGfwBootOk_TU102: (the GPU may be in a bad state and may need to be reset)
 kernel: NVRM: _kgspLogXid119: ********************************* GSP Timeout **********************************
 kernel: NVRM: _kgspLogXid119: Note: Please also check logs above.
 kernel: NVRM: GPU at PCI:0000:09:00: GPU-c3ad889c-6ef1-2ea2-2635-079a8d083975
 kernel: NVRM: Xid (PCI:0000:09:00): 119, pid=55768, name=nvidia-sleep.sh, Timeout after 6s of waiting for RPC response from GPU0 GSP! Expected function 76 (GSP_RM_CONTROL) (0x20801117 0x1).
 kernel: NVRM: GPU0 GSP RPC buffer contains function 76 (GSP_RM_CONTROL) and data 0x0000000020801117 0x0000000000000001.
 kernel: NVRM: GPU0 RPC history (CPU -> GSP):
 kernel: NVRM:     entry function                   data0              data1              ts_start           ts_end             duration actively_polling
 kernel: NVRM:      0    76   GSP_RM_CONTROL        0x0000000020801117 0x0000000000000001 0x00062f0b794056b3 0x0000000000000000          y
 kernel: NVRM:     -1    47   UNLOADING_GUEST_DRIVE 0x0000000000000000 0x0000000000000000 0x00062f09365e4005 0x00062f09365f299e  59801us  
 kernel: NVRM:     -2    10   FREE                  0x00000000c1e00b62 0x0000000000000000 0x00062f09365e3ea6 0x00062f09365e3fb6    272us  
 kernel: NVRM:     -3    10   FREE                  0x000000000000000a 0x0000000000000000 0x00062f09365e3d3e 0x00062f09365e3ea4    358us  
 kernel: NVRM:     -4    10   FREE                  0x000000000000000b 0x0000000000000000 0x00062f09365e3c21 0x00062f09365e3cc8    167us  
 kernel: NVRM:     -5    10   FREE                  0x0000000000000006 0x0000000000000000 0x00062f09365e39ec 0x00062f09365e3c1a    558us  
 kernel: NVRM:     -6    10   FREE                  0x0000000000000002 0x0000000000000000 0x00062f09365e345e 0x00062f09365e39a1   1347us  
 kernel: NVRM:     -7    10   FREE                  0x0000000000000005 0x0000000000000000 0x00062f09365e30b9 0x00062f09365e3459    928us  
 kernel: NVRM: GPU0 RPC event history (CPU <- GSP):
 kernel: NVRM:     entry function                   data0              data1              ts_start           ts_end             duration during_incomplete_rpc
 kernel: NVRM:      0    4108 UCODE_LIBOS_PRINT     0x0000000000000000 0x0000000000000000 0x00062f09365eb072 0x00062f09365eb072           
 kernel: NVRM:     -1    4128 GSP_POST_NOCAT_RECORD 0x0000000000000002 0x0000000000000028 0x00062f09365e5d70 0x00062f09365e5d72      2us  
 kernel: NVRM:     -2    4111 PERF_BRIDGELESS_INFO_ 0x0000000000000000 0x0000000000000000 0x00062f09365c4bd6 0x00062f09365c4bd7      1us  
 kernel: NVRM:     -3    4128 GSP_POST_NOCAT_RECORD 0x0000000000000005 0x00000285017d990c 0x00062f09365c1d34 0x00062f09365c1d35      1us  
 kernel: NVRM:     -4    4099 POST_EVENT            0x0000000000000000 0x0000000000000000 0x00062f09365c08c2 0x00062f09365c08c5      3us  
 kernel: NVRM:     -5    4128 GSP_POST_NOCAT_RECORD 0x0000000000000005 0x00000285017d990c 0x00062f09365bcd2b 0x00062f09365bcd2c      1us  
 kernel: NVRM:     -6    4099 POST_EVENT            0x0000000000000000 0x0000000000000000 0x00062f09365bb964 0x00062f09365bb966      2us  
 kernel: NVRM:     -7    4099 POST_EVENT            0x0000000000000000 0x0000000000000000 0x00062f09365b9e4a 0x00062f09365b9e4c      2us  
 kernel: CPU: 13 UID: 0 PID: 55768 Comm: nvidia-sleep.sh Tainted: G           O       6.13.4-zen3 #2
 kernel: Tainted: [O]=OOT_MODULE
 kernel: Hardware name: System manufacturer System Product Name/TUF GAMING X570-PLUS (WI-FI), BIOS 5013 03/22/2024
 kernel: Call Trace:
 kernel:  <TASK>
 kernel:  dump_stack_lvl+0x4c/0x70
 kernel:  dump_stack+0x10/0x16
 kernel:  os_dump_stack+0x9/0x10 [nvidia]
 kernel:  _kgspRpcRecvPoll+0x54c/0x620 [nvidia]
 kernel:  _issueRpcAndWait+0x71/0x360 [nvidia]
 kernel:  rpcRmApiControl_GSP+0x67e/0xa10 [nvidia]
 kernel:  ? os_release_spinlock+0x15/0x20 [nvidia]
 kernel:  ? _tlsThreadEntryGet+0x82/0x90 [nvidia]
 kernel:  ? osGetCurrentThread+0x26/0x60 [nvidia]
[... 125 lines cut ...]
 kernel:  entry_SYSCALL_64_after_hwframe+0x6c/0x74
 kernel: RIP: 0033:0x7fdddf751504
 kernel: Code: c7 00 16 00 00 00 b8 ff ff ff ff c3 66 2e 0f 1f 84 00 00 00 00 00 f3 0f 1e fa 80 3d c5 8b 10 00 00 74 13 b8 01 00 00 00 0f 05 <48> 3d 00 f0 ff ff 77 54 c3 0f 1f 00 55 48 89 e5 48 83 ec 20 48 89
 kernel: RSP: 002b:00007ffd8f06b418 EFLAGS: 00000202 ORIG_RAX: 0000000000000001
 kernel: RAX: ffffffffffffffda RBX: 0000000000000007 RCX: 00007fdddf751504
 kernel: RDX: 0000000000000007 RSI: 000056118bba3740 RDI: 0000000000000001
 kernel: RBP: 00007ffd8f06b440 R08: 0000000000000410 R09: 0000000000000001
 kernel: R10: 0000000000000004 R11: 0000000000000202 R12: 0000000000000007
 kernel: R13: 000056118bba3740 R14: 00007fdddf8535c0 R15: 00007fdddf850e80
 kernel:  </TASK>
 kernel: ---[ end trace 0000000000000000 ]---
```

### To Reproduce

 * Suspend
 * Try to resume

### Bug Incidence

Always

### nvidia-bug-report.log.gz

[nvidia-bug-report.log.gz](https://github.com/user-attachments/files/18984941/nvidia-bug-report.log.gz)

Ac5000, 2025-02-26:
I am starting to believe that my issue [here](https://bbs.archlinux.org/viewtopic.php?id=303420) is also related.

I've been trying to troubleshoot and tack down exactly what is happening but have made no progress. While I haven't had the random hangs I was having before, my system still seems to lock up randomly after receiving dpms off signal.

birdie-github, 2025-02-27:
This seems to be fixed in 570.124.04.

Thanks NVIDIA.

closed 2025-02-27 by birdie-github