NVIDIA/open-gpu-kernel-modules #797: CUDA initialization failiure on latest release drivers

E-Allie, 2025-02-27:
### NVIDIA Open GPU Kernel Modules Version

570.86.16 Release

### Please confirm this issue does not happen with the proprietary driver (of the same version). This issue tracker is only for bugs specific to the open kernel driver.

- [x] I confirm that this does not happen with the proprietary driver package.

### Operating System and Version

NixOS 25.05 (Warbler)

### Kernel Release

Linux 6.13.4 #1-NixOS SMP PREEMPT_DYNAMIC Fri Feb 21 13:11:21 UTC 2025 x86_64 GNU/Linux

### Please confirm you are running a stable release kernel (e.g. not a -rc). We do not accept bug reports for unreleased kernels.

- [x] I am running on a stable kernel release.

### Hardware: GPU

NVIDIA GeForce RTX 3080 Laptop GPU

### Describe the bug

The specific issue seems to have first been described here: https://github.com/NVIDIA/open-gpu-kernel-modules/discussions/780

When attempting to initialize NVDEC/CUDA in any capacity:

`cu->cuInit(0) failed -> CUDA_ERROR_NOT_INITIALIZED: initialization`

is returned.

This does not occur on proprietary kernel.

Setting "uvm_disable_hmm=1" on open kernel, as described in the above link, also worksaround/solves the issue.

### To Reproduce

This occurred when using any program trying to use CUDA/NVDEC.

I ran into it while using:

`ffmpeg -y -vsync 0 -hwaccel cuda -hwaccel_output_format cuda -i input.mp4 -c:a copy -c:v h264_nvenc -b:v 5M output.mp4`

which would return

`[AVHWDeviceContext @ 0x359865c0] cu->cuInit(0) failed -> CUDA_ERROR_NOT_INITIALIZED: initialization error`.

Similar initialization error would occur when invoking "vainfo" [through nvidia-vaapi-driver]

### Bug Incidence

Always

### nvidia-bug-report.log.gz

Will generate shortly after initial posting:

E-Allie, 2025-02-27:
After reading over all 3 generated bug reports, it seems that

`/sys/module/nvidia_uvm/parameters/uvm_disable_hmm:N` is constant, which is decidedly odd as I'm definitely setting it to 1/Y in modprobe under success

The key difference is that the addressing mode reported in nvidia-smi is HMM when failing, and None when working

Diatrus, 2025-03-04:
I can also corroborate this issue on Debian, Intel i5-12600KF + RTX 4070

E-Allie, 2025-03-07:
Different issue I've been encountering that seems to be root caused by this:

For a while I could not use DLSS through Proton, and just assumed it was a limitation of whatever games I was attempting to play etc etc, and had very non-descript error messages most of the time except "Crash on Streamline initialization"

The uvm-disable workaround outlined here also seems to have resolved "Cannot properly initialize DLSS" type issues

E-Allie, 2025-03-27:
Still identically failing on 570.133.07

E-Allie, 2025-04-24:
Update: Still identically failing on 575.52.01

Diatrus, 2025-10-16:
Also attempted using `nokaslr` as a workaround, but issue still occurs on kernel 6.14.11 unless I disable hmm.

Diatrus, 2025-10-21:
Confirmed issue persists with 580.95.05 on kernel 6.17.1.

Qubitium, 2025-10-23:
Nvidia, please fix this! I wasted 4+ hours fixing this only to find this github issue.. Thank god I found it. All thanks to @E-Allie for finding the source cause. Nvidia needs to fix the actual bug now.

plopresti, 2026-02-05:
> After checking some upstream commit, I tried following 2 commits.
> 
> 7ffb791423c7 ("x86/kaslr: Reduce KASLR entropy on most x86 systems") 7170130e4c72 ("x86/mm/init: Handle the special case of device private pages in add_pages(), to not increase max_pfn and trigger dma_addressing_limited() bounce buffers")
> 
> I created test custom kernel with these 2 patches and tested on same system with reproducer and now CUDA works without the nvidia_uvm workaround.

Thanks @vishal14051992. Red Hat has been tracking this as RHEL-129452, I think. That is the reference that shows up in the ChangeLog for kernel-6.12.0-124.31.1, which incorporated these fixes for EL10.

jcelerier, 2026-03-17:
@E-Allie I'm on 6.19.6 and still have the problem

```
$ nvidia-smi
Tue Mar 17 16:49:26 2026       
+-----------------------------------------------------------------------------------------+
| NVIDIA-SMI 590.48.01              Driver Version: 590.48.01      CUDA Version: 13.1     |
+-----------------------------------------+------------------------+----------------------+
| GPU  Name                 Persistence-M | Bus-Id          Disp.A | Volatile Uncorr. ECC |
| Fan  Temp   Perf          Pwr:Usage/Cap |           Memory-Usage | GPU-Util  Compute M. |
|                                         |                        |               MIG M. |
|=========================================+========================+======================|
|   0  NVIDIA GeForce RTX 3060 ...    On  |   00000000:01:00.0 Off |                  N/A |
| N/A   49C    P8             12W /   80W |       1MiB /   6144MiB |      0%      Default |
|                                         |                        |                  N/A |
+-----------------------------------------+------------------------+----------------------+
```