NVIDIA/open-gpu-kernel-modules #799: Rendering freeze randomly

jobs-git, 2025-03-02:
### NVIDIA Open GPU Kernel Modules Version

570.124

### Please confirm this issue does not happen with the proprietary driver (of the same version). This issue tracker is only for bugs specific to the open kernel driver.

- [ ] I confirm that this does not happen with the proprietary driver package.

### Operating System and Version

Description:	Ubuntu 24.04.2 LTS

### Kernel Release

6.8.0-53-generic

### Please confirm you are running a stable release kernel (e.g. not a -rc). We do not accept bug reports for unreleased kernels.

- [x] I am running on a stable kernel release.

### Hardware: GPU

RTX 4070

### Describe the bug

Description:	Ubuntu 24.04.2 LTS

### To Reproduce

Launch steam game and wait

### Bug Incidence

Always

### nvidia-bug-report.log.gz

n/a

tlneondo, 2025-03-06:
Is this while using gamescope or Wine-Wayland?

merrkry, 2025-03-07:
Same here. Happens with gamescope + minecraft / proton. This doesn't happen when running on iGPU.

jobs-git, 2025-03-07:
> Is this while using gamescope or Wine-Wayland?

Currently using Xorg

jobs-git, 2025-03-07:
Issue has also been reproduced with Fedora with 570.x driver. This issue does not happen with 550

Grinyaka, 2025-03-23:
Same here. Wayland and Xorg - problem appears in both cases. However, had no problems with old games like Skyrim, but both Split Fiction and Marvel Rivals freezes.
And, It seems like everything above 550 not working for me. Even with kernel upgrade (couldn't upgrade higher than 6.11, but someone said that 6.12 helped him)