NVIDIA/open-gpu-kernel-modules #800: HDMI issues on Lenovo Legion 5 with nVidia RTX 3060 (570.124.04) (UEFI Workaround Partially Fixes)

pippo111, 2025-03-02:
### NVIDIA Open GPU Kernel Modules Version

570.124.04

### Please confirm this issue does not happen with the proprietary driver (of the same version). This issue tracker is only for bugs specific to the open kernel driver.

- [ ] I confirm that this does not happen with the proprietary driver package.

### Operating System and Version

CachyOS 24.05

### Kernel Release

Linux 6.13.5-2-cachyos

### Please confirm you are running a stable release kernel (e.g. not a -rc). We do not accept bug reports for unreleased kernels.

- [x] I am running on a stable kernel release.

### Hardware: GPU

GPU 0: NVIDIA GeForce RTX 3060 Laptop GPU (UUID: GPU-178fb331-491c-2baa-27d4-83bf23e9b1f7)

### Describe the bug

Hi, after a fresh installation of CachyOS (KDE Wayland) on my laptop (Lenovo Legion 5 15IAH7H), I cannot get my external monitor to work when connecting it to the HDMI port. The monitor wakes up briefly, but the screen remains black before turning off again.

I partially resolved the issue by changing the UEFI settings to use only the dedicated GPU. However, this is not an optimal solution, and I was also unable to change the screen resolution on my laptop afterward.

Logs: https://paste.cachyos.org/p/42194e9.log

Here are some outputs, but I can provide more if needed:

```
OS: CachyOS x86_64
Host: 82RB (Legion 5 15IAH7H)
Kernel: Linux 6.13.5-2-cachyos
Uptime: 7 mins
Packages: 1335 (pacman)
Shell: fish 3.7.1
Display (eDP-1): 1920x1080 @ 60 Hz in 16" [Built-in]
DE: KDE Plasma 6.3.2
WM: KWin (Wayland)
WM Theme: Breeze
Theme: Breeze (Dark) [Qt], Breeze-Dark [GTK2], Breeze [GTK3]
Icons: breeze-dark [Qt], breeze-dark [GTK2/3/4]
Font: Noto Sans (10pt) [Qt], Noto Sans (10pt) [GTK2/3/4]
Cursor: capitaine (24px)
Terminal: konsole 24.12.2
CPU: 12th Gen Intel(R) Core(TM) i7-12700H (20) @ 4.70 GHz
GPU 1: NVIDIA GeForce RTX 3060 Mobile / Max-Q [Discrete]
GPU 2: Intel Iris Xe Graphics @ 1.40 GHz [Integrated]
Memory: 4.90 GiB / 31.07 GiB (16%)
Swap: 4.25 MiB / 35.07 GiB (0%)
Disk (/): 203.51 GiB / 478.16 GiB (43%) - ext4
Local IP (enp52s0): 192.168.0.223/24
Battery (L21B4PC0): 100% [AC Connected]
Locale: pl_PL.UTF-8
```

lspci:
```
00:00.0 Host bridge: Intel Corporation 12th Gen Core Processor Host Bridge/DRAM Registers (rev 02)
00:01.0 PCI bridge: Intel Corporation 12th Gen Core Processor PCI Express x16 Controller #1 (rev 02)
00:02.0 VGA compatible controller: Intel Corporation Alder Lake-P GT2 [Iris Xe Graphics] (rev 0c)
00:04.0 Signal processing controller: Intel Corporation Alder Lake Innovation Platform Framework Processor Participant (rev 02)
00:06.0 PCI bridge: Intel Corporation 12th Gen Core Processor PCI Express x4 Controller #0 (rev 02)
00:06.2 PCI bridge: Intel Corporation 12th Gen Core Processor PCI Express x4 Controller #2 (rev 02)
00:07.0 PCI bridge: Intel Corporation Alder Lake-P Thunderbolt 4 PCI Express Root Port #0 (rev 02)
00:0a.0 Signal processing controller: Intel Corporation Platform Monitoring Technology (rev 01)
00:0d.0 USB controller: Intel Corporation Alder Lake-P Thunderbolt 4 USB Controller (rev 02)
00:0d.2 USB controller: Intel Corporation Alder Lake-P Thunderbolt 4 NHI #0 (rev 02)
00:14.0 USB controller: Intel Corporation Alder Lake PCH USB 3.2 xHCI Host Controller (rev 01)
00:14.2 RAM memory: Intel Corporation Alder Lake PCH Shared SRAM (rev 01)
00:14.3 Network controller: Intel Corporation Alder Lake-P PCH CNVi WiFi (rev 01)
00:15.0 Serial bus controller: Intel Corporation Alder Lake PCH Serial IO I2C Controller #0 (rev 01)
00:15.1 Serial bus controller: Intel Corporation Alder Lake PCH Serial IO I2C Controller #1 (rev 01)
00:16.0 Communication controller: Intel Corporation Alder Lake PCH HECI Controller (rev 01)
00:1d.0 PCI bridge: Intel Corporation Alder Lake PCI Express Root Port #9 (rev 01)
00:1d.1 PCI bridge: Intel Corporation Alder Lake PCI Express x1 Root Port #10 (rev 01)
00:1f.0 ISA bridge: Intel Corporation Alder Lake PCH eSPI Controller (rev 01)
00:1f.3 Audio device: Intel Corporation Alder Lake PCH-P High Definition Audio Controller (rev 01)
00:1f.4 SMBus: Intel Corporation Alder Lake PCH-P SMBus Host Controller (rev 01)
00:1f.5 Serial bus controller: Intel Corporation Alder Lake-P PCH SPI Controller (rev 01)
01:00.0 VGA compatible controller: NVIDIA Corporation GA106M [GeForce RTX 3060 Mobile / Max-Q] (rev a1)
01:00.1 Audio device: NVIDIA Corporation GA106 High Definition Audio Controller (rev a1)
06:00.0 Non-Volatile memory controller: Samsung Electronics Co Ltd NVMe SSD Controller PM9A1/PM9A3/980PRO
34:00.0 Ethernet controller: Realtek Semiconductor Co., Ltd. RTL8111/8168/8211/8411 PCI Express Gigabit Ethernet Controller (rev 15)
```

Moreover sometimes the system hangs during the boot but restaring it solves the issue. Logs:

```
mar 01 16:28:09 filip-laptop kernel: [drm] [nvidia-drm] [GPU ID 0x00000100] Loading driver
mar 01 16:28:09 filip-laptop kernel: [drm:nv_drm_load [nvidia_drm]] *ERROR* [nvidia-drm] [GPU ID 0x00000100] Failed to allocate NvKmsKapiDevice
mar 01 16:28:09 filip-laptop kernel: [drm:nv_drm_register_drm_device [nvidia_drm]] *ERROR* [nvidia-drm] [GPU ID 0x00000100] Failed to register device
```

What may be the issue? How to fix it?

### To Reproduce

- install CachyOS with mentioned drivers
- try to connect external monitor via HDMI

### Bug Incidence

Always

### nvidia-bug-report.log.gz

[nvidia-bug-report.log.gz](https://github.com/user-attachments/files/19043559/nvidia-bug-report.log.gz)

Nycz-lab, 2025-03-03:
i have the same issue however i am using Display Port so seems not to be limited to hdmi

illegalargumentexc, 2025-03-18:
i have the same issue on a 2nd gen X1 Extreme with a 1650 (max-Q design)