NVIDIA/open-gpu-kernel-modules #801: 570.124.04 high refresh rate causes rendering issues with gpu accelerated apps (wayland)

Nycz-lab, 2025-03-03:
### NVIDIA Open GPU Kernel Modules Version

570.124.04

### Please confirm this issue does not happen with the proprietary driver (of the same version). This issue tracker is only for bugs specific to the open kernel driver.

- [ ] I confirm that this does not happen with the proprietary driver package.

### Operating System and Version

NixOS 24.11 (Vicuna)

### Kernel Release

Linux nixos 6.13.4-cachyos #1-NixOS SMP PREEMPT_DYNAMIC Fri Feb 21 13:11:21 UTC 2025 x86_64 GNU/Linux

### Please confirm you are running a stable release kernel (e.g. not a -rc). We do not accept bug reports for unreleased kernels.

- [x] I am running on a stable kernel release.

### Hardware: GPU

NVIDIA GeForce RTX 4090

### Describe the bug

When my refresh rate is set to 240Hz and I'm using a GPU-accelerated program like Firefox, my desktop occasionally renders through the application (e.g., the wallpaper becomes visible behind it). I am using KDE Plasma 6, and this issue does not occur when using driver version 565.77.

### To Reproduce

Use driver version 570.124.04 on a Desktop Environment like KDE Plasma 6, use a monitor with a refresh rate of 240hz, open a gpu accelerated program like firefox and open for example youtube (sometimes scrolling seems to increase the effect)

### Bug Incidence

Once

### nvidia-bug-report.log.gz

[nvidia-bug-report.log.gz](https://github.com/user-attachments/files/19060036/nvidia-bug-report.log.gz)

### More Info

I am also using a 4k Monitor and hdr is turned on, however i dont think this influences the problem. Also i should mention the visual "glitches" appear only in the lower right corner of my screen

marcbull, 2025-03-11:
I have the same problem with an NVIDIA RTX 4000 and a 120Hz refresh rate. The problem occurs under both Wayland and X11. I'm also running a KDE Plasma6 desktop on EndeavourOS Linux with the nvidia-open-dkms driver 570.124.04-1.

For me the bug incidence is: multiple times a day

[nvidia-bug-report.log.gz](https://github.com/user-attachments/files/19181151/nvidia-bug-report.log.gz)

syzzori, 2025-03-12:
Having the same issue, it seems like I'm having some sort of brightness flickering, alongside with graphical glitches in some games. Having Anti Aliasing issue in some games and other visual glitches. Arch Linux, linux-zen 6.13.6.zen1-1, nvidia-open-dkms 570.124.04-1, X11 KDE Plasma 6

chaNcharge, 2025-03-23:
Similar case happening, though only on 240hz and not on 120hz. Only seems to show on the bottom right of the screen where the panel is on KDE, not sure why specifically there. Arch Linux, 6.13.7-arch1-1, Wayland KDE Plasma 6, RTX 4080 Super, nvidia-open 570.133.07

Nycz-lab, 2025-04-10:
this issue persists on 570.133.07

francoism90, 2025-04-15:
I have the same issue, did anyone tried the closed driver to check if it doesn't happen on that one?

rinbega, 2025-06-21:
Similar issue with a 4k@160Hz monitor. In Hyprland it looks like a horizontal line somewhere in the bottom right corner that causes tearing when scrolling, in niri it's more consistent and looks like the wallpaper sprinkled with checkerboard patterns. Seems to be "fixed" by limiting the monitor to 144Hz.

Arch Linux, 6.16.0-rc2-4-cachyos-rc, 4070 Ti, both closed and open 575.64

NCX-85, 2025-07-01:
Issue persisting on 575.64, Bazzite Linux, 6.14.6-109.bazzite.fc42.x86_64 (64-bit), 4070 Ti Super, Displayport. Happens at 360hz with a Asus PG27AQN but I can get it to stop by setting the refresh to 240hz.

Nycz-lab, 2025-10-06:
Issue persists on 580.95.05

NCX-85, 2025-12-29:
@Nycz-lab that does explain why it looks like it disappeared for me.

Further research on the components in Zirconium - aside from Niri for window management it uses DankMaterialShell for its desktop shell. DMS is based off of Quickshell, which in turn is built on the QtQuick framework. So this seems to be an issue with a Qt desktop + Chromium apps specifically.

illusaen, 2026-05-11:
Also seeing the same issue with AMD 7800x3D + Nvidia 4700 on Nix + DMS + Niri. No issue when replacing Niri with Hyprland.

n0099, 2026-10-28:
https://forums.developer.nvidia.com/t/displayport-dsc-4k-240hz-flickering-artifacts/294490
I'm encountering this on KDE Plasma 6.6.6 with Wayland, RTX 4060 Ti with open kernel module `615.78.08` (also on several older version but I can't recall them exactly), 4k 160hz (not reproducible on 144hz and lower) monitor plugged in a Display Port port with version 1.4.

---
With `/sys/module/nvidia_modeset/parameters/debug` enabled and switching refresh rate I saw the following `dmesg`:
144hz:
```
nvidia-modeset: DPCONN> Notify detach begin
nvidia-modeset: DPCONN> Notify detach end
nvidia-modeset: DP-DEV> Disabling DSC decompression on device - 
nvidia-modeset: DPCONN> Notify Attach Begin (Head 3, pclk 1279840000 raster 4000 x 2222  312 bpp, DSC 1, FEC 1)
nvidia-modeset: DPCONN> |  (VIDEO) |
nvidia-modeset: DPCONN> DSC Mode = SINGLE
nvidia-modeset: DP-DEV> Enabling DSC decompression on device - 
nvidia-modeset: DPCONN> Notify Attach End
```
160hz:
```
nvidia-modeset: DPCONN> Delayed HDCP Cap read called.
nvidia-modeset: DPCONN> Notify detach begin
nvidia-modeset: DPCONN> Notify detach end
nvidia-modeset: DP-DEV> Disabling DSC decompression on device - 
nvidia-modeset: DPCONN> Notify Attach Begin (Head 0, pclk 1398400000 raster 4000 x 2185  256 bpp, DSC 1, FEC 1)
nvidia-modeset: DPCONN> |  (VIDEO) |
nvidia-modeset: DPCONN> DSC Mode = DUAL
nvidia-modeset: DP-DEV> Enabling DSC decompression on device - 
nvidia-modeset: DPCONN> Notify Attach End
nvidia-modeset: DPCONN> Delayed HDCP Cap read called.
```

---
Both refresh rate have `DSC 1` on, but `DSC Mode` is different (`DUAL` when 160hz), so I start digging its definition:
https://github.com/NVIDIA/open-gpu-kernel-modules/blob/ca2d03e6fafd4d4b56b383497d931c25b283bfcf/src/common/displayport/src/dp_connectorimpl2x.cpp#L888
https://github.com/NVIDIA/open-gpu-kernel-modules/blob/ca2d03e6fafd4d4b56b383497d931c25b283bfcf/src/common/inc/displayport/displayport.h#L310-L316
https://github.com/NVIDIA/open-gpu-kernel-modules/blob/ca2d03e6fafd4d4b56b383497d931c25b283bfcf/src/common/displayport/src/dp_groupimpl.cpp#L438
https://github.com/NVIDIA/open-gpu-kernel-modules/blob/ca2d03e6fafd4d4b56b383497d931c25b283bfcf/src/nvidia-modeset/src/dp/nvdp-connector.cpp#L479-L480
There're ~30 literal match for [2Head1OR](https://github.com/search?q=repo%3ANVIDIA%2Fopen-gpu-kernel-modules+2Head1OR&type=code) and [2Heads1OR](https://github.com/search?q=repo%3ANVIDIA%2Fopen-gpu-kernel-modules+2Heads1OR&type=code), among them I found a `bDualHeadMode`:
https://github.com/NVIDIA/open-gpu-kernel-modules/blob/ca2d03e6fafd4d4b56b383497d931c25b283bfcf/src/nvidia-modeset/src/nvkms-hdmi.c#L2576
Searching [that](https://github.com/search?q=repo%3ANVIDIA/open-gpu-kernel-modules%20bDualHeadMode&type=code) gives [bDualMode](https://github.com/search?q=repo%3ANVIDIA/open-gpu-kernel-modules%20bDualMode&type=code):
https://github.com/NVIDIA/open-gpu-kernel-modules/blob/ca2d03e6fafd4d4b56b383497d931c25b283bfcf/src/common/modeset/hdmipacket/nvhdmipkt_C671.c#L587
Also an abbr [2H1OR](https://github.com/search?q=repo%3ANVIDIA%2Fopen-gpu-kernel-modules+2H1OR&type=code) for it:
https://github.com/NVIDIA/open-gpu-kernel-modules/blob/ca2d03e6fafd4d4b56b383497d931c25b283bfcf/src/common/modeset/hdmipacket/nvhdmi_frlInterface.h#L71

---
Despite all these string finding, 
https://nvidia.custhelp.com/app/answers/detail/a_id/5338
> When a display is connected to the GPU and is set to DSC mode, the GPU may use two internal heads to drive the display when the pixel rate needed to drive the display mode exceeds the GPU’s single head limit. This may affect your display topology when using multiple monitors.  For example if two displays with support for DSC are connected to a single GeForce GPU, all 4 internal heads will be utilized and you will not be able to use a third monitor with the GPU at the same time.

https://github.com/NVIDIA/open-gpu-kernel-modules/discussions/396#discussioncomment-3937173
> [...]but in some of the most extreme cases a single head cannot fetch and produce pixels fast enough to feed the OR (IIRC, the limit is somewhere around 1.3G pixels/second, but my numbers may be off). Those are the cases where we need to use two heads.

In the previous `dmesg`, the `Head` changed from `3` to `0`, and the `pclk` increased from `1279840000` to `1398400000`, I don't know what's that [`pclk`](https://github.com/search?q=repo%3ANVIDIA%2Fopen-gpu-kernel-modules+pclk&type=code) but it seems larger than the `1.3G` mentioned above.

---
I also tried switch the `Color accuracy` to `Prefer color accuracy` in the KDE `Display Configuration` and all flickering disappeared, this workaround has been found by:
- 2025-04-03 https://forums.developer.nvidia.com/t/displayport-dsc-4k-240hz-flickering-artifacts/294490/16 but they just used a ICC profile, for me selecting the `Built-in` profile and keeping `Color accuracy` as `Prefer efficiency` also would fix it.
- 2025-04-28 https://forums.developer.nvidia.com/t/flickering-graphics-corruption-and-loss-of-signal-with-nvidia-4090-and-5k2k-screen-lg-45gx950a-b-aus/331111/2 https://bugs.kde.org/show_bug.cgi?id=503458
https://github.com/NVIDIA/open-gpu-kernel-modules/issues/801#issuecomment-3937193183
- 2026-02-20 https://github.com/NVIDIA/open-gpu-kernel-modules/issues/801#issuecomment-3937193183 a comment above me
- 2026-04-08 https://forums.developer.nvidia.com/t/displayport-dsc-4k-240hz-flickering-artifacts/294490/75

---
[... cut ...]
https://github.com/hyprwm/Hyprland/blob/aabab87c387bd6b6b1e5b443308b25cafa16225e/src/render/GLRenderer.cpp#L156-L160
> ```
> // nvidia doesn't have implicit sync, so we have to explicitly wait here, llvmpipe and other software renderer seems to bug out as well.
> if ((isNvidia() && *PNVIDIAANTIFLICKER) || isSoftware())
>     glFinish();
> else
>     glFlush(); // mark an implicit sync point
> ```

---
@amrit1711 if you are the moderator `@amrits` on https://forums.developer.nvidia.com, may you follow up these details to NVIDIA engineering team for this long-lived bug?

dMopp, 2026-10-28:
Some more findings:

Using HDR instead of SDR is fixing the issue in Umbriel (Wayland compositor). Maybe that helps