NVIDIA/open-gpu-kernel-modules #804: nvidia-open only allows 60Hz maximum for my TV - Smart TV LG OLED 55'' 4K 120Hz oled55c1psa

FernandoMMuniz, 2025-03-05:
### NVIDIA Open GPU Kernel Modules Version

nvidia-open 570.124.04-2

### Please confirm this issue does not happen with the proprietary driver (of the same version). This issue tracker is only for bugs specific to the open kernel driver.

- [ ] I confirm that this does not happen with the proprietary driver package.

### Operating System and Version

Arch Linux

### Kernel Release

Linux archlinux 6.13.5-arch1-1 #1 SMP PREEMPT_DYNAMIC Thu, 27 Feb 2025 18:09:44 +0000 x86_64 GNU/Linux

### Please confirm you are running a stable release kernel (e.g. not a -rc). We do not accept bug reports for unreleased kernels.

- [ ] I am running on a stable kernel release.

### Hardware: GPU

GPU 0: NVIDIA GeForce GTX 1650 (UUID: GPU-994c04e0-f852-0941-1dc8-f3f74a68f2d8)

### Describe the bug

Even though my TV allows 120Hz/FPS, my system is limited to 60Hz. The HDMI cable I'm using is the same one I use for my PS5, which I saw it run some games at 120FPS with VRR.
https://bugs.kde.org/show_bug.cgi?id=501090

[image]

Operating System: Arch Linux 
KDE Plasma Version: 6.3.2
KDE Frameworks Version: 6.11.0
Qt Version: 6.8.2
Kernel Version: 6.13.5-arch1-1 (64-bit)
Graphics Platform: Wayland
Processors: 8 × 11th Gen Intel® Core™ i5-11300H @ 3.10GHz
Memory: 8.1 GB of RAM
Graphics Processor 1: NVIDIA GeForce GTX 1650
Graphics Processor 2: Intel® Iris® Xe Graphics
Manufacturer: LENOVO
Product Name: 82MG
System Version: IdeaPad Gaming 3 15IHU6

### To Reproduce

1- Connect a PS5 HDMI cable on your notebook IdeaPad Gaming 3 15IHU6
2- Go to System Settings and try to raise your Hertz from 60 to 120

### Bug Incidence

Always

### nvidia-bug-report.log.gz

nvidia-bug-report.sh will now collect information about your
system and create the file 'nvidia-bug-report.log.gz' in the current
directory.  It may take several seconds to run.  In some
cases, it may hang trying to capture data generated dynamically
by the Linux kernel and/or the NVIDIA kernel module.  While
the bug report log file will be incomplete if this happens, it
may still contain enough data to diagnose your problem.

If nvidia-bug-report.sh hangs, consider running with the --safe-mode
and --extra-system-data command line arguments.

Please include the 'nvidia-bug-report.log.gz' log file when reporting
your bug via the NVIDIA Linux forum (see forums.developer.nvidia.com)
or by sending email to 'linux-bugs@nvidia.com'.

By delivering 'nvidia-bug-report.log.gz' to NVIDIA, you acknowledge
and agree that personal information may inadvertently be included in
the output.  Notwithstanding the foregoing, NVIDIA will use the
output only for the purpose of investigating your reported issue.

Running nvidia-bug-report.sh... complete.

Summary of Skipped Sections:

Skipped Component                   | Details
================================================================================
ibstat output                       | ibstat not found 
--------------------------------------------------------------------------------
acpidump output                     | acpidump not found 
--------------------------------------------------------------------------------
mst output                          | mst not found 
--------------------------------------------------------------------------------
nvlsm-bug-report.sh output          | nvlsm-bug-report.sh not found 
--------------------------------------------------------------------------------

Summary of Errors:

Error Component                     | Details                                                      | Resolution          
=========================================================================================================================

ptr1337, 2025-03-06:
HDMI 2.0 only Support for 4k material (max. 4096 x 2160) at 50 and 60 Hz specified.

You would need a newer GPU for that, which supports HDMI 2.1a

Binary-Eater, 2025-03-09:
HDMI 2.1 support is required for modelines greater than or equal to 4k resolution with refresh rates beyond 60Hz.

- https://web.archive.org/web/20170106195344/http://www.hdmi.org/manufacturer/hdmi_2_1/index.aspx
- https://www.nvidia.com/en-us/geforce/news/geforce-rtx-hdmi-2-1-support/

closed 2025-03-09 by Binary-Eater