NVIDIA/open-gpu-kernel-modules #807: Monitor freezes shortly after boot

arroyo-pl, 2025-03-12:
### NVIDIA Open GPU Kernel Modules Version

570.124.04

### Please confirm this issue does not happen with the proprietary driver (of the same version). This issue tracker is only for bugs specific to the open kernel driver.

- [x] I confirm that this does not happen with the proprietary driver package.

### Operating System and Version

Arch Linux

### Kernel Release

6.13.6-arch1-1 #1 SMP PREEMPT_DYNAMIC  x86_64 GNU/Linux

### Please confirm you are running a stable release kernel (e.g. not a -rc). We do not accept bug reports for unreleased kernels.

- [x] I am running on a stable kernel release.

### Hardware: GPU

NVIDIA GeForce RTX 4070 Laptop GPU (UUID: GPU-85fade4b-34aa-55e3-2fe2-caacc7060d25)

### Describe the bug

I have a setup with three screens: a laptop display and two external monitors (according to xrandr: DP-3 and DP-4).
Shortly after boot, the laptop screen froze.
About a minute later, the remaining two monitors also froze, and I had to reboot using the SysRq key combination (I was unable to switch to a text-mode console).

### To Reproduce

The issue occurred while I was simply preparing for work—I wasn't doing anything specific or resource-intensive. Unfortunately, I don't know how to reproduce the problem.

### Bug Incidence

Once

### nvidia-bug-report.log.gz

Regarding nvidia-bug-report.log.gz: dmesg and the journal have been cleared, as they are not relevant— the freeze occurred earlier. Instead, I have attached a separate log containing the relevant event.

[journal.log.gz](https://github.com/user-attachments/files/19206597/journal.log.gz)
[nvidia-bug-report.log.gz](https://github.com/user-attachments/files/19206598/nvidia-bug-report.log.gz)

mrintrepide, 2025-03-13:
Same with Fedora 41, Linux 6.13.6 and Nvidia 570.124.04.

```
[17034.231970] NVRM: nvAssertFailedNoLog: Assertion failed: CliGetEventInfo(rpc_params->hClient, rpc_params->hEvent, &pEvent) @ kernel_gsp.c:466
[17034.231982] NVRM: _kgspProcessRpcEvent: Failed to process received event 0x1003 (POST_EVENT) from GPU0: status=0x57
[17087.872996] fbcon: Taking over console
[17090.888839] [drm:nv_drm_atomic_commit [nvidia_drm]] *ERROR* [nvidia-drm] [GPU ID 0x00000100] Flip event timeout on head 0
[17094.088384] [drm:nv_drm_atomic_commit [nvidia_drm]] *ERROR* [nvidia-drm] [GPU ID 0x00000100] Flip event timeout on head 0
[17094.088489] Console: switching to colour frame buffer device 320x90
```

```
Linux version 6.13.6-200.fc41.x86_64 (mockbuild@079f93d6724f4f2d92c4bdc0aa9b8288) (gcc (GCC) 14.2.1 20250110 (Red Hat 14.2.1-7), GNU ld version 2.43.1-5.fc41) #1 SMP PREEMPT_DYNAMIC Fri Mar  7 21:33:48 UTC 2025
NVRM: loading NVIDIA UNIX Open Kernel Module for x86_64  570.124.04  Release Build  (dvs-builder@U22-I3-AF04-14-5)  Tue Feb 25 03:49:44 UTC 2025
```

mtijanic, 2025-03-14:
Duplicate of #739 which is already triaged and being worked on. Sorry, I don't have any update at this time.

Closing this, please continue on the other thread.

closed 2025-03-14 by mtijanic

SaigyoujiYuyuko233, 2025-03-30:
> Same with Fedora 41, Linux 6.13.6 and Nvidia 570.124.04.
>
> ```
> [17034.231970] NVRM: nvAssertFailedNoLog: Assertion failed: CliGetEventInfo(rpc_params->hClient, rpc_params->hEvent, &pEvent) @ [...]

Same behavior. After switching between sessions (ctrl+alt+F4 to tty4 then ctrl+alt+F2 back), I'm able to unfreeze it.

ValeZAA, 2025-04-13:
> Same behavior. After switching between sessions (ctrl+alt+F4 to tty4 then ctrl+alt+F2 back), I'm able to unfreeze it.

... Then I have kinda the same issue, I fixed it by disabling x11 and moving to wayland. First time installing Nvidia GPL driver by compiling it :) GSP novueu broke in 6.13.8 kernel in GSP loading, and pymol OpenGL broke....

Nvidia driver enables x11 by default, to disable it you need to add a kernel parameter to grub nvidia-drm.modeset=1 and delete  /usr/lib/udev/rules.d/61-gdm.rules