NVIDIA/open-gpu-kernel-modules #814: Building with gcc-15 fails

heitbaum, 2025-03-19:
### NVIDIA Open GPU Kernel Modules Version

570.124.06

### Operating System and Version

Ubuntu 24.04

### Kernel Release

6.12

### Please confirm you are running a stable release kernel (e.g. not a -rc). We do not accept bug reports for unreleased kernels.

- [x] I am running on a stable kernel release.

### Build Command

Build using gcc-14.2 completes.
Build with gcc-15.x fails with errors that were warnings in previous gcc.

Workaround is to add the following no-error statements, until the code is updated to avoid these warnings.

`EXTRA_CFLAGS += -Wno-error=return-type -Wno-implicit-function-declaration -Wno-int-conversion -Wno-incompatible-pointer-types`

### Terminal output/Build Log

As above

Also errors

./common/inc/nv-mm.h:29:13: error: conflicting types for 'vm_fault_t'; have 'int'
   29 | typedef int vm_fault_t;
      |             ^~~~~~~~~~

./common/inc/nv-time.h:113:19: error: conflicting types for 'timespec64_to_ns'; have 's64(struct timespec64 *)' {aka 'long long int(struct timespec64 *)'}
  113 | static inline s64 timespec64_to_ns(struct timespec64 *ts)
      |                   ^~~~~~~~~~~~~~~~

### More Info

https://github.com/NVIDIA/open-gpu-kernel-modules/blob/25bef4626e6c5ccf5b433e1c22b6b1bd59e6f1bd/kernel-open/Kbuild#L89

edpil02, 2025-03-19:
Builds  fine  with gcc-15 for me  ( 570.124.04 and  last update  570.133.07).

~$ gcc --version
gcc (GCC) 15.0.1 20250313 (Red Hat 15.0.1-0)
Copyright © 2025 Free Software Foundation, Inc.

[image]

edpil02, 2025-03-19:
I'm using the development version of Fedora (rawhide), which started using gcc-15 in early January.
I had a lot of problems at first, but Fedora has worked hard to stabilize it.

dekomote, 2025-04-17:
Fedora 42, not building neither 570 nor 575.

ptr1337, 2025-04-17:
> Fedora 42, not building neither 570 nor 575.

Maybe report the issue to the rpmfusion maintainers to apply the patches.

dekomote, 2025-04-17:
> > Fedora 42, not building neither 570 nor 575.
> 
> Maybe report the issue to the rpmfusion maintainers to apply the patches.

This is when trying the .run file or compiling it by hand, not using the rpmfusion packages.

edpil02, 2025-04-18:
Rpmfusion added a patch (nvidia-kernel-ccflags-y.patch)  for the latest  575.51.02 and fc43 rawhide ; but there is no build yet for fc42.

heitbaum, 2025-04-18:
I can confirm that using https://github.com/negativo17/nvidia-kmod/blob/fedora-42/nvidia-kernel-ccflags-y.patch allows for compile with gcc-15 and linux-6.14.1

heitbaum, 2025-04-27:
- Fixed with -std=gnu17 in #841

heitbaum, 2025-10-31:
Fixed in upstream

closed 2026-10-30 by Binary-Eater