NVIDIA/open-gpu-kernel-modules #815: Sporadic black screen as a result of flip event timeout

orzen, 2025-03-22:
### NVIDIA Open GPU Kernel Modules Version

nvidia-open-dkms 570.124.04-1

### Please confirm this issue does not happen with the proprietary driver (of the same version). This issue tracker is only for bugs specific to the open kernel driver.

- [ ] I confirm that this does not happen with the proprietary driver package.

### Operating System and Version

Arch Linux

### Kernel Release

Linux 6.13.7-arch1-1 #1 SMP PREEMPT_DYNAMIC Thu, 13 Mar 2025 18:12:00 +0000 x86_64 GNU/Linux

### Please confirm you are running a stable release kernel (e.g. not a -rc). We do not accept bug reports for unreleased kernels.

- [x] I am running on a stable kernel release.

### Hardware: GPU

GPU 0: NVIDIA GeForce RTX 4070

### Describe the bug

This issue only occurs after I left the computer for a while and the screen's power saver kicks in and turns off the display(LG 34'' UltraGear 34GN850). The display is connected with DisplayPort. The issue results in a black screen and the only way to recover is to force a power cycle.

This log entry is the only thing I can go on
```
[drm:nv_drm_atomic_commit [nvidia_drm]] *ERROR* [nvidia-drm] [GPU ID 0x00000800] Flip event timeout on head 0
```

### To Reproduce

The issue is hard to reproduce. Just leave the computer until the display's builtin power saver kicks in.

(This happens around 3 times per week)

### Bug Incidence

Sometimes

### nvidia-bug-report.log.gz

I unfortunately cannot run `nvidia-bug-report.sh` due to the issue.

ninetailedtori, 2025-06-21:
CR - see bootlog.

[bootlog.txt](https://github.com/user-attachments/files/20844162/bootlog.txt)

alicercedigital, 2026-09-25:
Still happening on 615.71.09 (RTX 3060, nvidia-open, kernel 7.2, Hyprland, one HDMI monitor, S3). After every resume the monitor stays black, and the log shows `Flip event timeout on head 0` the next time the compositor commits.

While it was black I read the monitor's HDMI status register over I2C (SCDC 0x40). It reported no clock at all, so nothing reaches the monitor even though DRM says the output is on. A mode change to another refresh rate did not bring it back.

Turning the output off and on from the compositor brings it back every time. The off hits the 3 s flip timeout, then the head restarts and the picture returns about a second later. I run it after every resume:

```
# hypridle.conf
general {
    after_sleep_cmd = hyprctl dispatch dpms off && sleep 1 && hyprctl dispatch dpms on
}
```

The same off/on for KDE, GNOME, sway and X11, and a small script that runs it after resume, are in https://gist.github.com/alicercedigital/3f4d99f7a6636b6e3de9a5d27a344d6c. I only tested the Hyprland one.