NVIDIA/open-gpu-kernel-modules #816: RTX 5090 cannot do 7680x2160p@240hz via DP2.1 on Linux

matte-schwartz, 2025-03-22:
### NVIDIA Open GPU Kernel Modules Version

570.133.07

### Please confirm this issue does not happen with the proprietary driver (of the same version). This issue tracker is only for bugs specific to the open kernel driver.

- [x] I confirm that this does not happen with the proprietary driver package.

### Operating System and Version

CachyOS

### Kernel Release

6.13.7-3-cachyos

### Please confirm you are running a stable release kernel (e.g. not a -rc). We do not accept bug reports for unreleased kernels.

- [x] I am running on a stable kernel release.

### Hardware: GPU

NVIDIA GeForce RTX 5090 (UUID: GPU-addf70ec-1a4e-f8db-39bc-59d1ab271798)

### Describe the bug

When connecting my RTX 5090 to my Samsung G95NC running Plasma 6 Wayland, I am unable to utilize my monitor's 7680x2160@240hz modeset via DisplayPort 2.1 on Linux. Instead it caps out at 7680x2160p@60Hz. However, using the same cable on Windows with my 5090 does work at that same full resolution and refresh rate.

Monitor link: https://www.samsung.com/us/computing/monitors/gaming/57-odyssey-neo-g9-dual-4k-uhd-quantum-mini-led-240hz-1ms-hdr-1000-curved-gaming-monitor-ls57cg952nnxza/
DisplayPort 2.1 cable link: https://www.amazon.com/dp/B0DQ7KY639

### To Reproduce

1. Connect RTX 5090 to Samsung G95NC via DisplayPort 2.1 cable
2. Make sure the G95NC is in 240Hz mode in the monitor's Menu -> Game -> Refresh Rate and DP 2.1 mode in Menu -> System -> DisplayPort. Ver.
3. Turn on the PC and log into Plasma 6 Wayland

KDE's display settings should report the monitor only supporting a static 60Hz rather than letting you choose between 60Hz and 240Hz like it does on Windows.

### Bug Incidence

Always

### nvidia-bug-report.log.gz

[nvidia-bug-report.log.gz](https://github.com/user-attachments/files/19406192/nvidia-bug-report.log.gz)

abchauhan-nv, 2025-03-24:
Hi @matte-schwartz ,
Thank you for the report. Tracking this on internal bug - #5185798.

matte-schwartz, 2025-03-25:
I booted with `drm.debug=0x1e` and grabbed a new NVIDIA bug report: 

[nvidia-bug-report.log.gz](https://github.com/user-attachments/files/19440977/nvidia-bug-report.log.gz)

It looks like it's not recognizing 7680x2160@240 as a mode at all in `drm_helper_probe_single_connector_modes`, although when I probe the edid and decode it with `di-edid-decode` everything looks fine with the edid itself: [edid-decode-240.log](https://github.com/user-attachments/files/19441017/edid-decode-240.log)

Since I can switch my monitor to a separate 120/60Hz mode with its own edid, and on that mode 120Hz and 60Hz both work, I'm including a readout of that edid too in case it has any useful information about why this 120/60Hz mode works perfectly on Linux while the 240Hz/60Hz mode does not: [edid-decode-120.log](https://github.com/user-attachments/files/19441033/edid-decode-120.log)

abchauhan-nv, 2025-03-26:
Can you please check if loading nvidia-modeset with overlay_layers_enabled set to 0 allow 7840x2160@240?

matte-schwartz, 2025-03-26:
`/sys/module/nvidia_modeset/parameters/enable_overlay_layers:N` doesn't appear to have made a difference in the issue.

[nvidia-bug-report.log.gz](https://github.com/user-attachments/files/19458864/nvidia-bug-report.log.gz)

matte-schwartz, 2025-03-26:
I installed plasma-x11 to take a look at nvidia-settings, and this actually might explain things:

[image]

Is the driver possibly limited to only UHBR10 on DP2.1 right now? 4 lanes x 10Gbps would make sense if that's the case which would also explain why the 7680x2160@120Hz mode **does** work, since even Displayport 1.4 can handle that with DSC. 7580x2160p@240Hz requires at least UHBR13.5. 

Is there an easy way to check this information on Windows to compare it against Linux via NVIDIA tooling? When I checked with GPU-Z in my 7680x2160@240Hz mode I got but that Link Rate (current) number doesn't really make sense:

[image]

matte-schwartz, 2025-04-10:
I ended up getting a second DisplayPort 2.1 compatible monitor, the [LG 45gx950a-b](https://www.lg.com/us/monitors/lg-45gx950a-b-gaming-monitor), which has a maximum resolution/refresh rate of 5120x2160p@165Hz. On this monitor I'm able to select that highest modeset without any issues.

[nvidia-bug-report.log.gz](https://github.com/user-attachments/files/19678820/nvidia-bug-report.log.gz)

Sewer56, 2025-04-30:
I bought and installed a 5080 earlier today and can confirm I have ran into the same issue myself. Same display, different Blackwell card.

Not much I can otherwise add, but it does feel a bit disappointing after buying the GPU *specifically for DP2.1 support*.

Sewer56, 2025-04-30:
Potentially worth noting; I use the same display as the person who opened the issue.
And (almost) the same cable, just the DP54 certified variant of the cable, rather than the DP80 one; as I needed 3 meters.

Mattheish, 2025-04-30:
I have exactly the same issue, with same monitor, GPU and cable. If necessary I can also provide some logs.

getcontext, 2025-05-07:
Same here. Tried xrandr to force refresh rate but no luck. Also I can have 8k only on nvidia-open drivers. When I install standard then 5k max resolution but with 240hz.

Mattheish, 2025-06-07:
Problem still exists for me under Driver Version: 575.57.08

Rivelia, 2025-06-26:
Problem still persists for me as well.

Sewer56, 2025-08-06:
We should have bought AMD 🤣

bugfixin, 2025-09-09:
Same issue here - I have a thought that it may be related to DSC (as even 80gbps may not be able to drive the 2x4k@240hz the monitor can handle). I've tried a wide variety of cables which all work in windows and are labeled as "DP2.1 80gbps" cables, including the one that came with the monitor (A samsung LS57CG952NNXZA, AKA G95NC). I can confirm I can get 120HZ refresh rates when I disable VRR in the monitor OSD, but no more. This behavior persists in X and Wayland, with driver version 580.82.07 and on kernels as recent as 6.16.3

bugfixin, 2025-09-09:
I've also tried this with monitor firmware versions 1005.3 and 1007.0.

The monitor OSD has options for "Adaptive sync" [on/off] and when that option is set to "ON", The OSD displays an option for refresh rate [120/240].
This changes the maximum selectable refresh rate in linux, but no combinations allow for 240hz.

| Adaptive Sync in Monitor OSD | Refresh Rate selection in Monitor OSD | Maximum mode selectable in linux | Adaptive Sync Selectable in linux |
| --- | --- | --- | --- |
| ON | 240 | _7840x2160@60hz_ | YES |
| ON | 120 | 7840x2160@120hz | YES | 
| OFF | N/A | _7840x2160@120hz_ | NO | 

Italics where refresh rate maximums are lower than expected.

I am including the edid-decodes for the different OSD settings (Adaptive sync OFF, Adaptive Sync ON+refresh120, and Adaptive sync ON+refresh240)
[adaptivesyncoff.log](https://github.com/user-attachments/files/22242387/adaptivesyncoff.log)
[adaptivesyncon-120.log](https://github.com/user-attachments/files/22242386/adaptivesyncon-120.log)
[adaptivesyncon-240.log](https://github.com/user-attachments/files/22242385/adaptivesyncon-240.log)

getcontext, 2025-10-18:
If xorg gets refreshed by nvidia-settings, those settings are getting gone. They have to be added everytime xorg is refreshed.

Sewer56, 2025-10-20:
@2eronia

It's not all sunshine and rainbow though.

If VRR is enabled, the (G95NC) monitor will flicker as it transitions from idle to not idle- even if just a text cursor flickers.
So you'd only really want VRR enabled for non-static content, like games.

Binary-Eater, 2025-10-22:
Just to provide an update, we have implemented the fix for the issue. It should be put in a future driver release (fixes both X11 and Wayland).

Sewer56, 2025-11-04:
Actually, hmm, doesn't seem Nvidia released aarch64 driver for 580.105.08 .
That's quite problematic for making a PR- ; well, at least I can test locally for now.

Sewer56, 2025-11-04:
@matte-schwartz I cannot tell you at the moment, after rebuilding my system I got

`Failed to allocate NvKmsKapiDevice`

As if I was missing the kernel module.
There is no kernel module at https://github.com/NVIDIA/open-gpu-kernel-modules/releases for `580.105.08`, just like there's no release for aarch64 driver for `580.105.08`, only x64.

So I'm kinda stuck as far as finding out, unfortunately.

They either didn't make a full release, or just have not uploaded everything yet.

Sewer56, 2025-11-05:
@Binary-Eater @matte-schwartz no ball on my end.
I noticed the module went up since last comment, and patched my system package to use it.
I can confirm in `nvidia-smi` I am in the latest driver, and `7680x2160p@240hz` ***is*** reported and available from the display modes.

But unfortunately, switching to it on wayland (hyprland) just locks up the system.
Consequently, after rebooting the system is now softlocked on a black screen.

Thankfully I'm on NixOS so I can roll back- but for those not as fortunate, you can save yourself a chroot by setting the monitor back to 120Hz mode, so it wouldn't try switching on 240Hz.

Binary-Eater, 2025-11-05:
@Sewer56 would you mind sharing nvidia-bug-report.sh dumps in your recovered state after you repro-ed the issue in a previous boot? We can get your previous boot kernel logs from there and understand what is happening through systemd-journald.

Binary-Eater, 2025-11-05:
For the record, I expect the fix in-place to behave the same as the workaround suggested for the X11 users, so this might be a separate issue now that we have ungated the mode that would need to be addressed.

Binary-Eater, 2025-11-05:
@Sewer56 it should be in your path on NixOS if using the nixpkgs approach.

Binary-Eater, 2025-11-05:
https://github.com/NixOS/nixpkgs/issues/167180

Sewer56, 2025-11-05:
```
❯ sudo nvidia-bug-report.sh                                                                                                           00:48
[sudo] password for sewer:

nvidia-bug-report.sh will now collect information about your
system and create the file 'nvidia-bug-report.log.gz' in the current
directory.  It may take several seconds to run.  In some
cases, it may hang trying to capture data generated dynamically
by the Linux kernel and/or the NVIDIA kernel module.  While
the bug report log file will be incomplete if this happens, it
may still contain enough data to diagnose your problem.

If nvidia-bug-report.sh hangs, consider running with the --safe-mode
and --extra-system-data command line arguments.

Please include the 'nvidia-bug-report.log.gz' log file when reporting
your bug via the NVIDIA Linux forum (see forums.developer.nvidia.com)
or by sending email to 'linux-bugs@nvidia.com'.

By delivering 'nvidia-bug-report.log.gz' to NVIDIA, you acknowledge
and agree that personal information may inadvertently be included in
the output.  Notwithstanding the foregoing, NVIDIA will use the
output only for the purpose of investigating your reported issue.

Running nvidia-bug-report.sh...
Detected driver type: Running RM Driver on GPUs
 complete.

Summary of Skipped Sections:

Skipped Component                   | Details
================================================================================
ldd output                          | glxinfo not found
--------------------------------------------------------------------------------
lsusb output                        | lsusb not found
--------------------------------------------------------------------------------
dmidecode output                    | dmidecode not found
--------------------------------------------------------------------------------
xrandr output                       | xrandr not found
--------------------------------------------------------------------------------
ibstat output                       | ibstat not found
--------------------------------------------------------------------------------
acpidump output                     | acpidump not found
--------------------------------------------------------------------------------
mst output                          | mst not found
--------------------------------------------------------------------------------
nvlsm-bug-report.sh output          | nvlsm-bug-report.sh not found
--------------------------------------------------------------------------------

Summary of Errors:

Error Component                     | Details                                                      | Resolution
=========================================================================================================================

~
❯ 
```

Yup, seems so.
I did a switch into 240Hz mode with Adaptive-Sync ON, so it crashes.
Then I did Ctrl+Alt+Del to try toggling a reboot. (But it doesn't reboot within 1 minute).
So I held power and switched monitor back before rebooting.

No output from the tool though.

Sewer56, 2025-11-05:
[nvidia-bug-report.log.gz](https://github.com/user-attachments/files/23347968/nvidia-bug-report.log.gz)

Only a gzip.
[Ah, checks out, the program blurb mentions it. Sorry, I'm real sleepy >.<, really late]

Sewer56, 2025-11-05:
Actually, I think I might have figured something useful.

Normally it locks up if I turn on Adaptive Sync + set res to 7680x2160@240Hz.
That's true for both switching Adaptive Sync on at runtime, and booting into that mode when Adaptive Sync is enabled.

However, if I switch into a TTY, then turn on Adaptive Sync on the display, and then switch back to my graphical session- there is no lockup. However, the only available modes are the pre-patch modes. i.e. same as last driver update.

vonswoopington, 2025-12-01:
> Maybe we could compare? I've been on Hyprland (Wayland) personally. With some usage I also noticed that in the 7680x2160@240Hz mode, rebooting often/always boots to a black screen once I'm back into the OS.
> 

I’m running KDE Plasma on Wayland, and I’m seeing the same black-screen issue. I also run into another problem when I’m in PBP mode. CachyOS won’t recover from sleep, and instead of waking normally I get a black screen that turns into moving static, kind of like an old Channel 3 screen with no input. About half the time I can switch to a TTY with Ctrl+Alt+F2 to reboot, but the other half I have to hard power cycle.

josuazurbruegg, 2025-12-01:
> > Fix works for me! Using Fedora with Cosmic Environment, I can now select 7680x2160@240hz and everything works beautifully, including Adaptive Sync. Thank you very much!
> 
> Does it work if you enable adaptive sync directly from the monitor's settings through its OSD?

Sorry, I don't seem to get any notifications from GitHub, even though I have subscribed to the issue.

Yes, I enabled Adaptive Sync directly through the monitor's settings/OSD menu. I think when switching from 120hz to 240hz output (using monitor OSD), the display goes dark as well, but after a reboot (and maybe disconnecting the monitor from power for a moment, I'll have to retry and report), everything works, and stays that way, even after rebooting again.

I have a three monitor setup with two vertical 25" displays on each side, this is very wonky with the big center display @240hz. But if I disable the two side displays, 240hz works fine for me.

I'm using Cosmic (Wayland compositor), an RTX 5090 (580.105.08) and the monitors firmware version is 1008.2.

Sewer56, 2025-12-03:
Yeah, makes sense to me.
Only thing I'd want to hear is confirmations from others that things are working okay; but so far, it's good on my end.

Binary-Eater, 2025-12-03:
Yeah, no worries. It would be easier to track either way in a separate issue.

vonswoopington, 2025-12-03:
> Yeah, makes sense to me. Only thing I'd want to hear is confirmations from others that things are working okay; but so far, it's good on my end.

That's awesome to hear! I'm willing to test now but I can't seem to figure out how to update to the latest beta in CachyOS.

vonswoopington, 2025-12-03:
I was able to get `590.44.01` beta installed and I can confirm that I can enable "Adaptive Sync" through the OSD now without going straight to a black screen.  I'm still getting a black screen at boot but a power cycle of the monitor seems to fix it. Thanks!