NVIDIA/open-gpu-kernel-modules #818: GPU cant go to D3cold even when no process is running

tr1xem, 2025-03-24:
### NVIDIA Open GPU Kernel Modules Version

 570.133.07

### Please confirm this issue does not happen with the proprietary driver (of the same version). This issue tracker is only for bugs specific to the open kernel driver.

- [x] I confirm that this does not happen with the proprietary driver package.

### Operating System and Version

CachyOS

### Kernel Release

Linux cachyos 6.13.7-3-cachyos #1 SMP PREEMPT_DYNAMIC Wed, 19 Mar 2025 19:07:38 +0000 x86_64 GNU/Linux

### Please confirm you are running a stable release kernel (e.g. not a -rc). We do not accept bug reports for unreleased kernels.

- [x] I am running on a stable kernel release.

### Hardware: GPU

GPU 0: NVIDIA GeForce RTX 4050 Laptop GPU (UUID: GPU-230091c2-98fc-0529-86de-0f02014e5b36)

### Describe the bug

Im using rtd3 power management earlier on 0x03 that means the gpu should go to d3cold when not in use earlier (before 570) it was working fine gpu was sleeping good but now gpu is always no matter what i do its always D0

### To Reproduce

Just update and monitor

### Bug Incidence

Always

### nvidia-bug-report.log.gz

[nvidia-bug-report.log.gz](https://github.com/user-attachments/files/19429774/nvidia-bug-report.log.gz)

### More Info

I cant test propitory as my system wont even boot using them

imaGuru, 2025-08-25:
@tr1xem Hi. Can you check out my comment here https://github.com/NVIDIA/open-gpu-kernel-modules/issues/905#issuecomment-3196986429 and see if your issue is related to mine (GPU oscillates between d0 and d3cold)?

tr1xem, 2025-08-25:
> [@tr1xem](https://github.com/tr1xem) Hi. Can you check out my comment here [#905 (comment)](https://github.com/NVIDIA/open-gpu-kernel-modules/issues/905#issuecomment-3196986429) and see if your issue is related to mine (GPU oscillates between d0 and d3cold)?

it got fixed in 575

other than that tlp was issue because of it gpu cant sleep i removed tlp

Now gpu sleeps it just turns on for a bit while opening a xwayland app (hyprland mis config from myside as my pervious config is broken due to update) but yep above 575 everything works good for me 

```
options nvidia NVreg_UsePageAttributeTable=1 \
    NVreg_InitializeSystemMemoryAllocations=0 \
    NVreg_DynamicPowerManagement=0x02 \
    NVreg_RegistryDwords=RMIntrLockingMode=1 \
    NVreg_EnableS0ixPowerManagement=1
    
```
my config

lonelyfloat, 2025-10-05:
@imaGuru Unfortunately this solution didn't work for me. However, I found out that for my issue, the GPU can go in and out of d3cold when `nvidia_drm.modeset=0` is true. However, when modeset is set to 0 my external monitor doesn't work because the HDMI port is directly connected to the NVIDIA GPU.

tr1xem, 2025-10-07:
@lonelyfloat thats weird see if ur display is conected ur gpu cant go to d3cold state anyways