NVIDIA/open-gpu-kernel-modules #822: HDMI output cannot exceed 120Hz at 1920x1080

MAGG4444, 2025-04-05:
### NVIDIA Open GPU Kernel Modules Version

570.133.07

### Please confirm this issue does not happen with the proprietary driver (of the same version). This issue tracker is only for bugs specific to the open kernel driver.

- [ ] I confirm that this does not happen with the proprietary driver package.

### Operating System and Version

Arch Linux, NixOS 25.05 and Ubuntu 24.04

### Kernel Release

6.12.21

### Please confirm you are running a stable release kernel (e.g. not a -rc). We do not accept bug reports for unreleased kernels.

- [x] I am running on a stable kernel release.

### Hardware: GPU

NVIDIA GeForce RTX 3080 Laptop GPU & NVIDIA GeForce RTX 4080 Laptop GPU

### Describe the bug

When using an external 1080p 180Hz monitor via HDMI, setting the refresh rate above 120Hz causes the display to go black with no signal. The monitor reports "no input" or "disconnected." This happens:

- On both X11 and Wayland
- Regardless of desktop environment
- On both proprietary `nvidia` and open `nvidia-open` drivers
- Across different distributions (tested on Arch Linux, NixOS, and Ubuntu)

Even after a system reboot, the issue persists until the refresh rate is manually reset to 120Hz or lower. There are no error messages in logs; video output simply stops.

### To Reproduce

- Connect a 1920x1080 monitor via HDMI and set refresh rates higher than 120Hz.
- Install and use either the `nvidia` or `nvidia-open` driver.

### Bug Incidence

Always

### nvidia-bug-report.log.gz

[nvidia-bug-report.log.gz](https://github.com/user-attachments/files/19615198/nvidia-bug-report.log.gz)

### More Info

- DisplayPort works correctly at 180Hz
- Using `nouveau` allows 180Hz over HDMI
- Issue confirmed on systems with RTX3080 and RTX4080
- HDMI cables and ports have been tested and are not the cause
- Multiple users have the same issue: https://bbs.archlinux.org/viewtopic.php?id=302969

NgoHuy, 2025-04-10:
I confirm it happened on my widescreen 21:9 when I set resolution bigger than 1920x1080 causes black screen.

Binary-Eater, 2025-04-27:
If you have ssh access to the system, would you mind providing the bug report from `nvidia-bug-report.sh` after entering the black screen state. Also in that state, do you mind sharing the output of the following command `nvidia-settings -t -q flatpanelsignal`?

abchauhan-nv, 2025-05-03:
Hi @MAGG4444 ,

Can you please share the HDMI monitor model?
Would it be possible to capture the EDID in the binary format. Please attach it to this thread. Instructions to capture it is available here - [managing-a-display-edid-on-linux](https://nvidia.custhelp.com/app/answers/detail/a_id/3571/~/managing-a-display-edid-on-linux)

Thank you

MAGG4444, 2025-05-07:
@Binary-Eater @abchauhan-nv, thank you for following up!

I tried `nvidia-settings -t -q flatpanelsignal` but it seems like it is not a recognized attribute name on my device.

My X11 environment is running on Linux Mint, here the information collected while it's black screen:

```
root@SAKURA:~# uname --kernel-version
#52-Ubuntu SMP PREEMPT_DYNAMIC Thu Dec  5 13:09:44 UTC 2024

root@MAGG:~# nvidia-settings -t -q flatpanelsignal
ERROR: Error parsing query 'flatpanelsignal' (Unrecognized attribute name).

root@MAGG:~# nvidia-settings -t -q RefreshRate
180.00 Hz

root@MAGG:~# nvidia-smi
Tue May  6 20:19:12 2025       
+-----------------------------------------------------------------------------------------+
| NVIDIA-SMI 570.133.07             Driver Version: 570.133.07     CUDA Version: 12.8     |
|-----------------------------------------+------------------------+----------------------+
| GPU  Name                 Persistence-M | Bus-Id          Disp.A | Volatile Uncorr. ECC |
| Fan  Temp   Perf          Pwr:Usage/Cap |           Memory-Usage | GPU-Util  Compute M. |
|                                         |                        |               MIG M. |
|=========================================+========================+======================|
|   0  NVIDIA GeForce RTX 3080 ...    Off |   00000000:01:00.0  On |                  N/A |
| N/A   50C    P8             16W /  115W |     313MiB /  16384MiB |      0%      Default |
|                                         |                        |                  N/A |
+-----------------------------------------+------------------------+----------------------+
                                                                                         
+-----------------------------------------------------------------------------------------+
| Processes:                                                                              |
|  GPU   GI   CI              PID   Type   Process name                        GPU Memory |
|        ID   ID                                                               Usage      |
|=========================================================================================|
|    0   N/A  N/A           10946      G   /usr/lib/xorg/Xorg                      153MiB |
|    0   N/A  N/A           11296      G   cinnamon                                 52MiB |
|    0   N/A  N/A           11574      G   ...ersion=20250506-050110.590000         63MiB |
+-----------------------------------------------------------------------------------------+
```

And the bug report here: [nvidia-bug-report.log.gz](https://github.com/user-attachments/files/20074087/nvidia-bug-report.log.gz)

Let me know if you need anything else!

abchauhan-nv, 2025-05-13:
Thank you, @MAGG4444 . Would you mind checking if disabling HDMI deep color - ,i.e. setting nvidia_modeset.hdmi_deepcolor=0 in the kernel command line parameters help avoid the issue?

Binary-Eater, 2026-03-15:
@fshp I took a look and this monitor is indeed a HDMI 2.1 FRL monitor. While the fix we are targeting for the 1080p@180Hz issue is in progress, that same change will likely not resolve the issue you are seeing.

https://www.samsung.com/us/monitors/gaming/34-inch-odyssey-oled-g8-g85sd-ultra-wqhd-175hz-03ms-g-sync-sku-ls34dg856snxza/

Do you mind opening a separate issue to track?

Binary-Eater, 2026-10-23:
This should be fixed with our current driver releases.