NVIDIA/open-gpu-kernel-modules #834: Sleepmode fails sometimes : nvidia-sleep.sh: line 45: echo: write error: Input/output error

tobiasBora, 2025-04-22:
### NVIDIA Open GPU Kernel Modules Version

570.133.07-6.12.23

### Please confirm this issue does not happen with the proprietary driver (of the same version). This issue tracker is only for bugs specific to the open kernel driver.

- [x] I confirm that this does not happen with the proprietary driver package.

### Operating System and Version

NixOs unstable (commit 2631b0b7abcea6e640ce31cd78ea58910d31e650)

### Kernel Release

6.12.23

### Please confirm you are running a stable release kernel (e.g. not a -rc). We do not accept bug reports for unreleased kernels.

- [x] I am running on a stable kernel release.

### Hardware: GPU

GPU 0: NVIDIA RTX 2000 Ada Generation Laptop GPU

### Describe the bug

From time to time, I am unable to go to sleep mode, the computer would directly wake up. Note that this is not deterministic, as it may work from time to time. When this happens, I cannot anymore run `nvidia-smi` (hang), and I need to reboot to get again access to the full system. Here are the logs I obtain when I do this operation:

```
avril 21 22:46:37 bestosprec systemd-logind[2108]: The system will suspend now!
avril 21 22:46:37 bestosprec ModemManager[3337]: <msg> [sleep-monitor-systemd] system is about to suspend
avril 21 22:46:37 bestosprec dbus-daemon[2016]: [system] Activating via systemd: service name='org.freedesktop.nm_dispatcher' unit='dbus-org.freedesktop.nm-dispatcher.service' requested by ':1.8' (uid=0 pid=2148 comm="/nix/store/4bywivanfabyaz9ffw26fwbasp3ns52v-networ" label="kernel")
avril 21 22:46:37 bestosprec systemd[1]: Starting Network Manager Script Dispatcher Service...
avril 21 22:46:37 bestosprec dbus-daemon[2016]: [system] Successfully activated service 'org.freedesktop.nm_dispatcher'
avril 21 22:46:37 bestosprec systemd[1]: Started Network Manager Script Dispatcher Service.
avril 21 22:46:37 bestosprec kernel: wlp0s20f3: deauthenticating from f8:ab:05:09:2f:e0 by local choice (Reason: 3=DEAUTH_LEAVING)
avril 21 22:46:37 bestosprec wpa_supplicant[2299]: nl80211: send_event_marker failed: Source based routing not supported
avril 21 22:46:37 bestosprec wpa_supplicant[2299]: wlp0s20f3: CTRL-EVENT-DISCONNECTED bssid=f8:ab:05:09:2f:e0 reason=3 locally_generated=1
avril 21 22:46:37 bestosprec wpa_supplicant[2299]: wlp0s20f3: CTRL-EVENT-DSCP-POLICY clear_all
avril 21 22:46:37 bestosprec avahi-daemon[2013]: Withdrawing address record for 2001:861:3381:7880:2faf:10f3:b12e:df03 on wlp0s20f3.
avril 21 22:46:37 bestosprec avahi-daemon[2013]: Withdrawing address record for 2001:861:3381:7880:6d17:f37a:2a7:140d on wlp0s20f3.
avril 21 22:46:37 bestosprec avahi-daemon[2013]: Leaving mDNS multicast group on interface wlp0s20f3.IPv6 with address 2001:861:3381:7880:6d17:f37a:2a7:140d.
avril 21 22:46:37 bestosprec avahi-daemon[2013]: Joining mDNS multicast group on interface wlp0s20f3.IPv6 with address fe80::50ca:5a00:8aa6:da9c.
avril 21 22:46:37 bestosprec avahi-daemon[2013]: Registering new address record for fe80::50ca:5a00:8aa6:da9c on wlp0s20f3.*.
avril 21 22:46:37 bestosprec avahi-daemon[2013]: Withdrawing address record for fe80::50ca:5a00:8aa6:da9c on wlp0s20f3.
avril 21 22:46:37 bestosprec avahi-daemon[2013]: Leaving mDNS multicast group on interface wlp0s20f3.IPv6 with address fe80::50ca:5a00:8aa6:da9c.
avril 21 22:46:37 bestosprec avahi-daemon[2013]: Interface wlp0s20f3.IPv6 no longer relevant for mDNS.
avril 21 22:46:37 bestosprec avahi-daemon[2013]: Interface wlp0s20f3.IPv4 no longer relevant for mDNS.
avril 21 22:46:37 bestosprec avahi-daemon[2013]: Leaving mDNS multicast group on interface wlp0s20f3.IPv4 with address 192.168.1.12.
avril 21 22:46:37 bestosprec avahi-daemon[2013]: Withdrawing address record for 192.168.1.12 on wlp0s20f3.
avril 21 22:46:37 bestosprec avahi-daemon[2013]: Joining mDNS multicast group on interface wlp0s20f3.IPv4 with address 192.168.1.12.
avril 21 22:46:37 bestosprec avahi-daemon[2013]: New relevant interface wlp0s20f3.IPv4 for mDNS.
avril 21 22:46:37 bestosprec avahi-daemon[2013]: Registering new address record for 192.168.1.12 on wlp0s20f3.IPv4.
avril 21 22:46:37 bestosprec avahi-daemon[2013]: Withdrawing address record for 192.168.1.12 on wlp0s20f3.
avril 21 22:46:37 bestosprec avahi-daemon[2013]: Leaving mDNS multicast group on interface wlp0s20f3.IPv4 with address 192.168.1.12.
avril 21 22:46:37 bestosprec avahi-daemon[2013]: Interface wlp0s20f3.IPv4 no longer relevant for mDNS.
avril 21 22:46:37 bestosprec systemd[1]: Stopped target Host and Network Name Lookups.
avril 21 22:46:37 bestosprec systemd[1]: Stopping Host and Network Name Lookups...
avril 21 22:46:37 bestosprec systemd[1]: Stopped target User and Group Name Lookups.
avril 21 22:46:37 bestosprec systemd[1]: Stopping User and Group Name Lookups...
avril 21 22:46:37 bestosprec systemd[1]: Stopping Name Service Cache Daemon (nsncd)...
avril 21 22:46:37 bestosprec systemd[1]: nscd.service: Deactivated successfully.
avril 21 22:46:37 bestosprec systemd[1]: Stopped Name Service Cache Daemon (nsncd).
avril 21 22:46:37 bestosprec systemd[1]: nscd.service: Consumed 401ms CPU time, 4.8M memory peak, 560K memory swap peak, 88K read from disk, 552K written to disk, 183.2K incoming IP traffic, 92.9K outgoing IP traffic.
avril 21 22:46:37 bestosprec avahi-daemon[2013]: Got SIGHUP, reloading.
avril 21 22:46:37 bestosprec avahi-daemon[2013]: No service file found in /etc/avahi/services.
avril 21 22:46:37 bestosprec systemd[1]: Starting Name Service Cache Daemon (nsncd)...
avril 21 22:46:37 bestosprec systemd[1]: Stopping Network Time Synchronization...
avril 21 22:46:37 bestosprec nsncd[305152]: Apr 21 20:46:37.532 INFO started, config: Config { ignored_request_types: {}, worker_count: 8, handoff_timeout: 10s }, path: "/var/run/nscd/socket"
avril 21 22:46:37 bestosprec systemd[1]: Started Name Service Cache Daemon (nsncd).
[... 294 lines cut ...]
avril 21 22:46:43 bestosprec avahi-daemon[2013]: Registering new address record for 2001:861:3381:7880:4290:f0f:d12c:2364 on wlp0s20f3.*.
avril 21 22:46:49 bestosprec kscreenlocker_greet[305094]: pam_kwallet5(kde:auth): pam_kwallet5: pam_sm_authenticate
avril 21 22:46:49 bestosprec kscreenlocker_greet[305094]: pam_kwallet5(kde:auth): pam_kwallet5: we were already executed
avril 21 22:46:49 bestosprec kscreenlocker_greet[305094]: pam_kwallet5(kde:setcred): pam_kwallet5: pam_sm_setcred
avril 21 22:46:50 bestosprec kscreenlocker_greet[305094]: qt.qpa.wayland: Could not create EGL surface (EGL error 0x3000)
avril 21 22:46:50 bestosprec kscreenlocker_greet[305094]: Failed to write to the pipe: Bad file descriptor.
avril 21 22:46:52 bestosprec systemd[1]: NetworkManager-dispatcher.service: Deactivated successfully.
```

PS: not 100% sure if this bug is not reproducible on the closed-source version since it does not fire deterministically.

### To Reproduce

- In KDE plasma, either close the lid or enter sleep mode by going into KDE's menu
- Observe that it logs you out, but does not enter sleep mode

(this is not always reproducible)

### Bug Incidence

Once

### nvidia-bug-report.log.gz

[nvidia-bug-report.log.gz](https://github.com/user-attachments/files/19845877/nvidia-bug-report.log.gz)

igorgue, 2025-05-03:
Same thing happens to me, it starts to fail after a few suspend cycles, and there's no way to get it back to work unless you reboot.

amrit1711, 2025-05-11:
Hi,
I have filed a bug 5274139 internally for tracking purpose.
Requesting @ninetailedtori @igorgue to share nvidia bug report and reliable repro steps if any.
Are you seeing this issue with close source or open kernel module driver ?

ninetailedtori, 2025-05-11:
I'm noticing this on open. I have not tested closed source yet, and not much intention of given the support for the WMs and comps I use. I'll send a few crash logs from a few of my boots once I have time for the open source kernel modules and driver 😄

fp4code, 2025-05-16:
Same random bug with closed source 575.51.02 on RTX A3000 Laptop GPU with 6.14.0-15-generic Ubuntu 25.04.

pinkycollie, 2025-05-16:
Fixing this means either freeing VRAM, tuning your workloads, updating drivers, or hardware upgrades.

tobiasBora, 2025-06-09:
I tried with the closed source, and I also sometimes see sleep issues, but the problem is not exactly the same as the above reported one, since the system would not even wake up (just black screen with fans at 100%… very dangerous for the hardware when this occurs when the laptop is in its case). Here are a few errors (only red lines in log):
```
juin 09 12:00:39 bestosprec kernel: NVRM: failed to allocate vmap() page descriptor table!
juin 09 12:00:39 bestosprec kernel: nvidia 0000:01:00.0: PM: pci_pm_suspend(): nv_pmops_suspend [nvidia] returns -5
juin 09 12:00:39 bestosprec kernel: nvidia 0000:01:00.0: PM: dpm_run_callback(): pci_pm_suspend returns -5
juin 09 12:00:39 bestosprec kernel: nvidia 0000:01:00.0: PM: failed to suspend async: error -5
juin 09 12:00:39 bestosprec kernel: PM: Some devices failed to suspend, or early wake event detected
juin 09 12:00:39 bestosprec kernel: spd5118 0-0050: PM: dpm_run_callback(): spd5118_resume [spd5118] returns -6
juin 09 12:00:39 bestosprec kernel: spd5118 0-0050: PM: failed to resume async: error -6
…
[above error repeats once more]
…
juin 09 12:01:41 bestosprec systemd-sleep[167425]: Failed to put system to sleep. System resumed again: Input/output error
…
juin 09 12:01:41 bestosprec systemd[1]: Failed to start System Suspend.
…
[in yellow]
juin 09 12:02:53 bestosprec kernel: thunderbolt 0000:00:0d.3: 0: timeout reading config space 1 from 0xa
juin 09 12:02:53 bestosprec kernel: thunderbolt 0000:00:0d.3: 0: timeout reading config space 1 from 0xa
juin 09 12:02:53 bestosprec kernel: thunderbolt 0000:00:0d.3: 0: timeout reading config space 1 from 0xa
juin 09 12:02:53 bestosprec kernel: ------------[ cut here ]------------
juin 09 12:02:53 bestosprec kernel: wrong offset (expected 39, got a
                                    )
juin 09 12:02:53 bestosprec kernel: WARNING: CPU: 4 PID: 135737 at drivers/thunderbolt/ctl.c:222 check_config_address+0x8f/0xb0 [thunderbolt]
juin 09 12:02:53 bestosprec kernel: Modules linked in: ccm sd_mod scsi_mod scsi_common qrtr nft_chain_nat xt_MASQUERADE xfrm_user xfrm_algo xt_addrtype rfcomm snd_seq_dummy snd_hrtimer snd_seq snd_seq_device a>
juin 09 12:02:53 bestosprec kernel:  soundwire_bus mac80211 snd_soc_core loop xt_nat nf_nat snd_compress ac97_bus nf_conntrack snd_pcm_dmaengine intel_uncore_frequency intel_uncore_frequency_common nf_defrag_i>
juin 09 12:02:53 bestosprec kernel:  snd_timer intel_gtt ttm llc dell_smbios intel_rapl_common tpm_crb btbcm videobuf2_common intel_ish_ipc cryptd dcdbas i2c_algo_bit vfat fat dell_wmi_ddv dell_wmi_sysman firm>
juin 09 12:02:53 bestosprec kernel:  efi_pstore nfnetlink efivarfs tpm libaescfb ecdh_generic ecc rng_core dmi_sysfs ip_tables x_tables autofs4 sdhci_pci cqhci sdhci nvme atkbd libps2 led_class xhci_pci vivald>
juin 09 12:02:53 bestosprec kernel: CPU: 4 UID: 0 PID: 135737 Comm: kworker/u89:25 Tainted: P           O       6.12.23 #1-NixOS
juin 09 12:02:53 bestosprec kernel: Tainted: [P]=PROPRIETARY_MODULE, [O]=OOT_MODULE
juin 09 12:02:53 bestosprec kernel: Hardware name: Dell Inc. Precision 3591/0KTW71, BIOS 1.13.0 02/07/2025
juin 09 12:02:53 bestosprec kernel: Workqueue: async async_run_entry_fn
juin 09 12:02:53 bestosprec kernel: RIP: 0010:check_config_address+0x8f/0xb0 [thunderbolt]
juin 09 12:02:53 bestosprec kernel: Code: ff c3 cc cc cc cc 90 48 c7 c7 28 9a b4 c0 e8 18 d4 be c8 90 0f 0b 90 90 eb da 90 48 c7 c7 50 9a b4 c0 89 c6 e8 02 d4 be c8 90 <0f> 0b 90 90 eb c4 90 89 fa 89 ce 48 c7 >
juin 09 12:02:53 bestosprec kernel: RSP: 0018:ffffa3a3c2c6fa30 EFLAGS: 00010246
juin 09 12:02:53 bestosprec kernel: RAX: 0000000000000000 RBX: 0000000000000042 RCX: 0000000000000000
juin 09 12:02:53 bestosprec kernel: RDX: 0000000000000000 RSI: 0000000000000000 RDI: 0000000000000000
juin 09 12:02:53 bestosprec kernel: RBP: 0000000000000000 R08: 0000000000000000 R09: 0000000000000000
juin 09 12:02:53 bestosprec kernel: R10: 0000000000000000 R11: 0000000000000000 R12: ffffa3a3c2c6fa64
juin 09 12:02:53 bestosprec kernel: R13: 0000000000000000 R14: ffff8cb08fe07f40 R15: ffffa3a3c2c6fbc8
juin 09 12:02:53 bestosprec kernel: FS:  0000000000000000(0000) GS:ffff8cb08f400000(0000) knlGS:0000000000000000
juin 09 12:02:53 bestosprec kernel: CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
juin 09 12:02:53 bestosprec kernel: CR2: 00007f9f08199488 CR3: 0000000887422001 CR4: 0000000000f72ef0
juin 09 12:02:53 bestosprec kernel: DR0: 0000000000000000 DR1: 0000000000000000 DR2: 0000000000000000
juin 09 12:02:53 bestosprec kernel: DR3: 0000000000000000 DR6: 00000000ffff07f0 DR7: 0000000000000400
juin 09 12:02:53 bestosprec kernel: PKRU: 55555554
juin 09 12:02:53 bestosprec kernel: Call Trace:
juin 09 12:02:53 bestosprec kernel:  <TASK>
juin 09 12:02:53 bestosprec kernel:  tb_cfg_read_raw+0x24c/0x2b0 [thunderbolt]
juin 09 12:02:53 bestosprec kernel:  tb_cfg_read+0x59/0x100 [thunderbolt]
juin 09 12:02:53 bestosprec kernel:  tb_port_find_cap+0xcf/0x210 [thunderbolt]
juin 09 12:02:53 bestosprec kernel:  tb_switch_tmu_init+0xbd/0x4d0 [thunderbolt]
juin 09 12:02:53 bestosprec kernel:  tb_switch_resume+0xe1/0x330 [thunderbolt]
juin 09 12:02:53 bestosprec kernel:  tb_runtime_resume+0x32/0x130 [thunderbolt]
juin 09 12:02:53 bestosprec kernel:  ? __pfx_pci_pm_runtime_resume+0x10/0x10
juin 09 12:02:53 bestosprec kernel:  __rpm_callback+0x41/0x170
juin 09 12:02:53 bestosprec kernel:  ? __pfx_pci_pm_runtime_resume+0x10/0x10
juin 09 12:02:53 bestosprec kernel:  rpm_callback+0x59/0x70
[... cut ...]
juin 09 12:02:53 bestosprec kernel: nvidia 0000:01:00.0: PM: failed to suspend async: error -5
juin 09 12:02:53 bestosprec kernel: PM: Some devices failed to suspend, or early wake event detected
juin 09 12:02:53 bestosprec kernel: spd5118 0-0050: PM: dpm_run_callback(): spd5118_resume [spd5118] returns -6
juin 09 12:02:53 bestosprec kernel: spd5118 0-0050: PM: failed to resume async: error -6

[… error involving nvidia 0000:01:00.0: PM: pci_pm_suspend(): nv_pmops_suspend [nvidia] returns -5 repeats a few more times]
```

aaronp24, 2025-06-13:
Thanks for testing this. If you didn't see the `vmap` error or the one about `kgspCreateRadix3` then you didn't  hit either problem that that patch is supposed to fix.

> After a reboot, I tried to go to sleep mode, and I see the usual message a few seconds on the black screen when turning it on again:
> 
> ```
> spd5118 0-0050: Failed to write b = 0: -6
> spd5118 0-0050: PM: dpm_run_callback(): spd5118_resume [spd5118] returns -6
> spd5118 0-0050: PM: failed to resume async: error -6
> ```

Apparently spd5118 is a "SPD5118 (JEDEC JESD300) compliant temperature sensor chip" so I don't think that has anything to do with the GPU.

> Here are some random possibly related messages in log (none of them are error):
> 
> ```
> juin 13 15:17:24 bestosprec kernel: NVRM: loading NVIDIA UNIX Open Kernel Module for x86_64  570.153.02  Release Build  (nixbld@)
> …
> juin 13 15:17:27 bestosprec kernel: [drm] Initialized nvidia-drm 0.0.0 for 0000:01:00.0 on minor 0
> juin 13 15:17:27 bestosprec kernel: nvidia 0000:01:00.0: [drm] No compatible format found
> juin 13 15:17:27 bestosprec kernel: nvidia 0000:01:00.0: [drm] Cannot find any crtc or sizes
> ```

These are printed by the core DRM fbdev helper when it doesn't find any connected display devices. It's normal if you don't have any monitors attached to the GPU.

> If I get again a crash, I'll let you know.

Sounds good, thanks!

tobiasBora, 2025-06-14:
> Thanks for testing this. If you didn't see the vmap error or the one about kgspCreateRadix3 then you didn't hit either problem that that patch is supposed to fix.

Ok thanks. So I guess your patch does not suppress the error from the log, just mitigates its effect. Any idea why this error spawn from the start? It may also help to have a more reproducible bug.

> Apparently spd5118 is a "SPD5118 (JEDEC JESD300) compliant temperature sensor chip" so I don't think that has anything to do with the GPU.

Oh, interesting, thanks!

ninetailedtori, 2025-06-21:
Unsure if any of my other reports count under this, my journals got cleared for disk space. See https://github.com/NVIDIA/open-gpu-kernel-modules/issues/472, https://github.com/NVIDIA/open-gpu-kernel-modules/issues/815 if either work :)

KiLLeRRaT, 2025-08-11:
I'm not sure if it will help anyone, or if it applies to only the proprietry drivers and what not, but I've come across this tweak to `/usr/lib/systemd/system-sleep/nvidia` that seems to have helped my problem:

```
--- /tmp/nvidia-orig	2025-08-12 08:41:46.945362306 +1200
+++ /tmp/nvidia-edited	2025-08-12 08:41:28.711997186 +1200
@@ -29,6 +29,11 @@
 fi
 
 case "$1" in
+	# FROM: https://forum.artixlinux.org/index.php?PHPSESSID=cl2ls1ebedmfjfjvif5lg7evmu&topic=7425.msg45423#msg45423
+	# Added the pre only.
+	pre)
+		/usr/bin/nvidia-sleep.sh "suspend"
+		;;
 	post)
 		/usr/bin/nvidia-sleep.sh "resume"
 		;;

```

From: https://forum.artixlinux.org/index.php?PHPSESSID=cl2ls1ebedmfjfjvif5lg7evmu&topic=7425.msg45423#msg45423

I will update this post if I discover it doesn't work every time.  Last night I would suspend, and the system keeps waking.  Without closing any applications or doing anything, I only edited the above file, and there after suspend worked solid without any problems.

LFd3v, 2025-09-02:
I am glad to report that after upgrading to version 580.76.05 (but kernel was also updated tho) it has been almost 7 days and there was no problems with suspending/resuming the computer (no other manual changes since the problem started months ago, and I have also checked and `/usr/lib/systemd/system-sleep/nvidia` is still the same, i.e. no "pre" section). I did not have to close any apps, and I have been using llama.cpp/ollama almost daily, and there were no issues suspending.

```
$ pacman -Q | grep -e nvidia -e cuda -e linux-zen  
cuda 12.9.1-2
lib32-nvidia-utils 580.76.05-1
lib32-opencl-nvidia 580.76.05-1
linux-firmware-nvidia 20250808-1
linux-zen 6.16.3.zen1-1
linux-zen-headers 6.16.3.zen1-1
nvidia-cg-toolkit 3.1-8
nvidia-open-dkms 580.76.05-4
nvidia-prime 1.0-5
nvidia-settings 580.76.05-1
nvidia-utils 580.76.05-4
opencl-nvidia 580.76.05-4

$ uptime  
 08:33:50 up 6 days, 17:02,  1 user,  load average: 0,12, 0,31, 0,38
```