NVIDIA/open-gpu-kernel-modules #835: Intermittent Framebuffer Corruption and Window Stalls on NVIDIA GBM Stack

mysteryx93, 2025-04-23:
### NVIDIA Open GPU Kernel Modules Version

570.144

### Please confirm this issue does not happen with the proprietary driver (of the same version). This issue tracker is only for bugs specific to the open kernel driver.

- [x] I confirm that this does not happen with the proprietary driver package.

### Operating System and Version

Garuda Linux Broadwing

### Kernel Release

6.14.3-zen1-1.1-zen

### Please confirm you are running a stable release kernel (e.g. not a -rc). We do not accept bug reports for unreleased kernels.

- [x] I am running on a stable kernel release.

### Hardware: GPU

GeoForce GTX 2060 Mobile

### Describe the bug

Running on Wayland with the proprietary NVIDIA drivers, I experience occasional graphical glitches that affect window rendering and responsiveness. It causes momentary partial freezes or graphical corruption (screen tearing, flickering regions, UI elements not redrawing), and soon after, full system freeze.

These symptoms do not occur on X11, or present differently (e.g. HDMI issues instead), suggesting driver instability specific to Wayland compositing.

### To Reproduce

Boot into Wayland session with proprietary NVIDIA drivers.
Use a multi-window setup or compositing-heavy applications (browser + video playback + IDE, etc.)
Use for a few hours, sometimes sooner
Observe sudden visual glitches:
- Frozen windows
- Artifacts in title bars
- Areas of desktop not redrawing
- Partial responsiveness, followed by recovery
- Full system freeze

### Bug Incidence

Sometimes

### nvidia-bug-report.log.gz

This report is shortly after visual glitches; but before a full system freeze.
[nvidia-bug-report.log.gz](https://github.com/user-attachments/files/19858999/nvidia-bug-report.log.gz)

### More Info

Suspected Root Cause / Theory
- Likely buffer flip or damage event mishandling, possibly in NVIDIA’s GBM buffer management
- Compositor (KWin) attempts redraw → NVIDIA driver fails to update part of the scanout plane
- Wayland surface commit cycle broken or delayed
- Session enters a semi-stalled state with cursor updated by hardware plane, but window buffers stale or unflushed
- May relate to power-state transitions, VRR, or OpenGL context leakage

Suggested Areas to Investigate
- GBM/Wayland buffer lifecycle mismatch (flush/fence delay?)
- nvidia-drm scanout path vs KWin’s presentation stack
- HDMI signal management under dynamic buffer transitions
- Resource leak in DRM plane allocator
- Timing misalignment between compositor + hardware event queue

Binary-Eater, 2025-04-27:
@mysteryx93 thanks for this very detailed analysis. I think it would better to provide this on the [NVIDIA/egl-wayland](https://github.com/NVIDIA/egl-wayland) issue tracker.

mysteryx93, 2025-04-27:
ok posted here
https://github.com/NVIDIA/egl-wayland/issues/171