NVIDIA/open-gpu-kernel-modules #840: add kernel-6.15 support

jeamieofqidan, 2025-04-26:
### NVIDIA Open GPU Kernel Modules Version

575.51.02

### Operating System and Version

Fedora 43(rawhide)

### Kernel Release

6.15.0-0.rc3.20250424gita79be02bba5c.31.fc43.x86_64.img

### Please confirm you are running a stable release kernel (e.g. not a -rc). We do not accept bug reports for unreleased kernels.

- [ ] I am running on a stable kernel release.

### Build Command

make modules

### Terminal output/Build Log

Although I know that kernel-6.15 is only a pre-release version, it has already launched its third version, and the latest official driver launched on April 17 still does not support kernel-6.15. So I would like to ask for a solution here.

ptr1337, 2025-04-26:
Here are some patches, which we made:
570: https://github.com/CachyOS/CachyOS-PKGBUILDS/tree/master/nvidia/nvidia-utils
575: https://github.com/CachyOS/CachyOS-PKGBUILDS/tree/feat/nvidia-575/nvidia/nvidia-utils

jeamieofqidan, 2025-04-27:
Some patches failed to apply and the compilation still cannot be completed. Can you give me a complete (patched) driver package?

jeamieofqidan, 2025-04-27:
Here are build log:
  CC [M]  nvidia-uvm/uvm_pmm_sysmem.o
  CC [M]  nvidia-uvm/uvm_pmm_gpu.o
nvidia-uvm/uvm_pmm_gpu.c:2580:2: 错误：‘#else’ 没有匹配的 ‘#if’
 2580 | #else
      |  ^~~~
nvidia-uvm/uvm_pmm_gpu.c: 在函数‘hmm_block_atomic_fault_locked’中:
nvidia-uvm/uvm_pmm_gpu.c:2631:5: 错误：隐式声明函数‘hmm_range_fault_begin’ [-Wimplicit-function-declaration]
 2631 |     hmm_range_fault_begin(va_block);
      |     ^~~~~~~~~~~~~~~~~~~~~
nvidia-uvm/uvm_pmm_gpu.c:2636:12: 错误：隐式声明函数‘make_device_exclusive’ [-Wimplicit-function-declaration]
 2636 |     page = make_device_exclusive(service_context->block_context->mm, start,
      |            ^~~~~~~~~~~~~~~~~~~~~
nvidia-uvm/uvm_pmm_gpu.c:2636:10: 错误：assignment to ‘struct page *’ from ‘int’ makes pointer from integer without a cast [-Wint-conversion]
 2636 |     page = make_device_exclusive(service_context->block_context->mm, start,
      |          ^
nvidia-uvm/uvm_pmm_gpu.c:2659:9: 错误：隐式声明函数‘hmm_range_fault_retry’ [-Wimplicit-function-declaration]
 2659 |     if (hmm_range_fault_retry(va_block)) {
      |         ^~~~~~~~~~~~~~~~~~~~~
包含于文件 nvidia-uvm/uvm_pmm_gpu.c:163:
nvidia-uvm/uvm_pmm_gpu.c:2679:24: 错误：implicit declaration of function ‘hmm_va_block_cpu_page_is_same’; did you mean ‘uvm_va_block_cpu_page_index’? [-Wimplicit-function-declaration]
 2679 |             UVM_ASSERT(hmm_va_block_cpu_page_is_same(va_block, page_index, page));
      |                        ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
nvidia-uvm/uvm_common.h:137:47: 附注：in definition of macro ‘UVM_IGNORE_EXPR’
  137 | #define UVM_IGNORE_EXPR(expr) ((void)sizeof(!(expr)))
      |                                               ^~~~
nvidia-uvm/uvm_common.h:176:45: 附注：在宏 ‘UVM_ASSERT_MSG_IGNORE’ 展开时
  176 |     #define UVM_ASSERT(expr)                UVM_ASSERT_MSG_IGNORE(expr, "\n")
      |                                             ^~~~~~~~~~~~~~~~~~~~~
nvidia-uvm/uvm_pmm_gpu.c:2679:13: 附注：在宏 ‘UVM_ASSERT’ 展开时
 2679 |             UVM_ASSERT(hmm_va_block_cpu_page_is_same(va_block, page_index, page));
      |             ^~~~~~~~~~
nvidia-uvm/uvm_pmm_gpu.c:2684:27: 错误：implicit declaration of function ‘hmm_va_block_cpu_page_populate’; did you mean ‘uvm_va_block_cpu_page_index’? [-Wimplicit-function-declaration]
 2684 |             NV_STATUS s = hmm_va_block_cpu_page_populate(va_block, page_index, page);
      |                           ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                           uvm_va_block_cpu_page_index
nvidia-uvm/uvm_pmm_gpu.c:2690:9: 错误：implicit declaration of function ‘cpu_mapping_clear’; did you mean ‘cpumask_clear’? [-Wimplicit-function-declaration]
 2690 |         cpu_mapping_clear(va_block, page_index);
      |         ^~~~~~~~~~~~~~~~~
      |         cpumask_clear
nvidia-uvm/uvm_pmm_gpu.c:2703:5: 错误：隐式声明函数‘hmm_release_atomic_pages’ [-Wimplicit-function-declaration]
 2703 |     hmm_release_atomic_pages(va_block, service_context);
      |     ^~~~~~~~~~~~~~~~~~~~~~~~
nvidia-uvm/uvm_pmm_gpu.c: 在文件作用域：
nvidia-uvm/uvm_pmm_gpu.c:2708:2: 错误：‘#endif’ 没有匹配的 ‘#if’
 2708 | #endif
      |  ^~~~~
make[5]: *** [/usr/src/kernels/6.14.0-0.rc7.20250321gitb3ee1e460951.60.fc43.x86_64/scripts/Makefile.build:207：nvidia-uvm/uvm_pmm_gpu.o] 错误 1
make[4]: *** [/usr/src/kernels/6.14.0-0.rc7.20250321gitb3ee1e460951.60.fc43.x86_64/Makefile:2040：.] 错误 2
make[3]: *** [/usr/src/kernels/6.14.0-0.rc7.20250321gitb3ee1e460951.60.fc43.x86_64/Makefile:263：__sub-make] 错误 2
make[3]: 离开目录“/home/lamarujian/open-gpu-kernel-modules-main/kernel-open”
make[2]: *** [Makefile:263：__sub-make] 错误 2
make[2]: 离开目录“/usr/src/kernels/6.14.0-0.rc7.20250321gitb3ee1e460951.60.fc43.x86_64”
make[1]: *** [Makefile:140：modules] 错误 2
make[1]: 离开目录“/home/lamarujian/open-gpu-kernel-modules-main/kernel-open”
make: *** [Makefile:59：modules] 错误 2

jeamieofqidan, 2025-04-29:
The open source kernel was compiled successfully, and the patched files were integrated into the official driver installation package

es20490446e, 2025-05-01:
[drivers-gnu17.patch](https://pastebin.com/raw/KFXSB6qS)

jeamieofqidan, 2025-05-12:
kernel-6.15 is near to It's official release, so it is time to release an open source kernel driver compatible with kernel-6.15.

ptr1337, 2025-05-24:
> > kernel-6.15 is near to It's official release, so it is time to release an open source kernel driver compatible with kernel-6.15.
> 
> I think there should be a driver release on 19th may, because the 5060 will be released there, but not sure if this applies to linux. We will see.

latest 570 appears to **mostly** support it. There is for the closed source driver a patch needed and can be found here:
https://github.com/CachyOS/kernel-patches/blob/master/6.15/misc/nvidia/0003-Workaround-nv_vm_flags_-calling-GPL-only-code.patch

In case someone needs it

jeamieofqidan, 2025-05-29:
I never thought that one day, I, a non-professional Linux enthusiast, would argue with programmers on GitHub over an issue related to a program patch.

closed 2025-05-29 by jeamieofqidan