NVIDIA/open-gpu-kernel-modules #848: Driver fails to allocate memory for decoder

x5f3759df, 2025-05-05:
### NVIDIA Open GPU Kernel Modules Version

575.51.03

### Please confirm this issue does not happen with the proprietary driver (of the same version). This issue tracker is only for bugs specific to the open kernel driver.

- [ ] I confirm that this does not happen with the proprietary driver package.

### Operating System and Version

Fedora Linux 42 (KDE Plasma Desktop Edition)

### Kernel Release

Linux fedora 6.14.4-300.fc42.x86_64

### Please confirm you are running a stable release kernel (e.g. not a -rc). We do not accept bug reports for unreleased kernels.

- [x] I am running on a stable kernel release.

### Hardware: GPU

GPU 0: NVIDIA GeForce RTX 5070 (UUID: GPU-0c71241b-6112-2e53-8d60-a2c6e634e234)

### Describe the bug

Driver fails to allocate any memory for hw decoding. Even is memory is free. Logs from dmesg says that.

```[мая 5 10:32] NVRM: nvCheckOkFailedNoLog: Check failed: Out of memory [NV_ERR_NO_MEMORY] (0x00000051) returned from _memdescAllocInternal(pMemDesc) @ mem_desc.c:1353
[  +0,000004] NVRM: sysmemConstruct_IMPL: *** Cannot allocate sysmem through fb heap
[  +0,000017] NVRM: nvAssertOkFailedNoLog: Assertion failed: Out of memory [NV_ERR_NO_MEMORY] (0x00000051) returned from pRmApi->Alloc(pRmApi, device->session->handle, isSystemMemory ? device->handle : device->subhandle, &physHandle, isSystemMemory ? NV01_MEMORY_SYSTEM : NV01_MEMORY_LOCAL_USER, &memAllocParams, sizeof(memAllocParams)) @ nv_gpu_ops.c:4647
[мая 5 10:37] NVRM: nvCheckOkFailedNoLog: Check failed: Out of memory [NV_ERR_NO_MEMORY] (0x00000051) returned from _memdescAllocInternal(pMemDesc) @ mem_desc.c:1353
[  +0,000005] NVRM: sysmemConstruct_IMPL: *** Cannot allocate sysmem through fb heap
[  +0,000025] NVRM: nvAssertOkFailedNoLog: Assertion failed: Out of memory [NV_ERR_NO_MEMORY] (0x00000051) returned from pRmApi->Alloc(pRmApi, device->session->handle, isSystemMemory ? device->handle : device->subhandle, &physHandle, isSystemMemory ? NV01_MEMORY_SYSTEM : NV01_MEMORY_LOCAL_USER, &memAllocParams, sizeof(memAllocParams)) @ nv_gpu_ops.c:4647
[мая 5 10:43] NVRM: nvCheckOkFailedNoLog: Check failed: Out of memory [NV_ERR_NO_MEMORY] (0x00000051) returned from _memdescAllocInternal(pMemDesc) @ mem_desc.c:1353
[  +0,000005] NVRM: sysmemConstruct_IMPL: *** Cannot allocate sysmem through fb heap
[  +0,000010] NVRM: nvAssertOkFailedNoLog: Assertion failed: Out of memory [NV_ERR_NO_MEMORY] (0x00000051) returned from pRmApi->Alloc(pRmApi, device->session->handle, isSystemMemory ? device->handle : device->subhandle, &physHandle, isSystemMemory ? NV01_MEMORY_SYSTEM : NV01_MEMORY_LOCAL_USER, &memAllocParams, sizeof(memAllocParams)) @ nv_gpu_ops.c:4647
[  +6,016231] NVRM: nvCheckOkFailedNoLog: Check failed: Out of memory [NV_ERR_NO_MEMORY] (0x00000051) returned from _memdescAllocInternal(pMemDesc) @ mem_desc.c:1353
[  +0,000012] NVRM: sysmemConstruct_IMPL: *** Cannot allocate sysmem through fb heap
[  +0,000014] NVRM: nvAssertOkFailedNoLog: Assertion failed: Out of memory [NV_ERR_NO_MEMORY] (0x00000051) returned from pRmApi->Alloc(pRmApi, device->session->handle, isSystemMemory ? device->handle : device->subhandle, &physHandle, isSystemMemory ? NV01_MEMORY_SYSTEM : NV01_MEMORY_LOCAL_USER, &memAllocParams, sizeof(memAllocParams)) @ nv_gpu_ops.c:4647
[  +4,283316] NVRM: nvCheckOkFailedNoLog: Check failed: Out of memory [NV_ERR_NO_MEMORY] (0x00000051) returned from _memdescAllocInternal(pMemDesc) @ mem_desc.c:1353
[  +0,000004] NVRM: sysmemConstruct_IMPL: *** Cannot allocate sysmem through fb heap
[  +0,000009] NVRM: nvAssertOkFailedNoLog: Assertion failed: Out of memory [NV_ERR_NO_MEMORY] (0x00000051) returned from pRmApi->Alloc(pRmApi, device->session->handle, isSystemMemory ? device->handle : device->subhandle, &physHandle, isSystemMemory ? NV01_MEMORY_SYSTEM : NV01_MEMORY_LOCAL_USER, &memAllocParams, sizeof(memAllocParams)) @ nv_gpu_ops.c:4647
[мая 5 10:45] NVRM: nvCheckOkFailedNoLog: Check failed: Out of memory [NV_ERR_NO_MEMORY] (0x00000051) returned from _memdescAllocInternal(pMemDesc) @ mem_desc.c:1353
[  +0,000004] NVRM: sysmemConstruct_IMPL: *** Cannot allocate sysmem through fb heap
[  +0,000012] NVRM: nvAssertOkFailedNoLog: Assertion failed: Out of memory [NV_ERR_NO_MEMORY] (0x00000051) returned from pRmApi->Alloc(pRmApi, device->session->handle, isSystemMemory ? device->handle : device->subhandle, &physHandle, isSystemMemory ? NV01_MEMORY_SYSTEM : NV01_MEMORY_LOCAL_USER, &memAllocParams, sizeof(memAllocParams)) @ nv_gpu_ops.c:4647
[ +26,625894] NVRM: nvCheckOkFailedNoLog: Check failed: Out of memory [NV_ERR_NO_MEMORY] (0x00000051) returned from _memdescAllocInternal(pMemDesc) @ mem_desc.c:1353
[  +0,000003] NVRM: sysmemConstruct_IMPL: *** Cannot allocate sysmem through fb heap
[  +0,000012] NVRM: nvAssertOkFailedNoLog: Assertion failed: Out of memory [NV_ERR_NO_MEMORY] (0x00000051) returned from pRmApi->Alloc(pRmApi, device->session->handle, isSystemMemory ? device->handle : device->subhandle, &physHandle, isSystemMemory ? NV01_MEMORY_SYSTEM : NV01_MEMORY_LOCAL_USER, &memAllocParams, sizeof(memAllocParams)) @ nv_gpu_ops.c:4647
[мая 5 10:58] NVRM: nvCheckOkFailedNoLog: Check failed: Out of memory [NV_ERR_NO_MEMORY] (0x00000051) returned from _memdescAllocInternal(pMemDesc) @ mem_desc.c:1353
[  +0,000005] NVRM: sysmemConstruct_IMPL: *** Cannot allocate sysmem through fb heap
[  +0,000007] NVRM: nvAssertOkFailedNoLog: Assertion failed: Out of memory [NV_ERR_NO_MEMORY] (0x00000051) returned from pRmApi->Alloc(pRmApi, device->session->handle, isSystemMemory ? device->handle : device->subhandle, &physHandle, isSystemMemory ? NV01_MEMORY_SYSTEM : NV01_MEMORY_LOCAL_USER, &memAllocParams, sizeof(memAllocParams)) @ nv_gpu_ops.c:4647
[  +0,073746] NVRM: nvCheckOkFailedNoLog: Check failed: Out of memory [NV_ERR_NO_MEMORY] (0x00000051) returned from _memdescAllocInternal(pMemDesc) @ mem_desc.c:1353
[  +0,000004] NVRM: sysmemConstruct_IMPL: *** Cannot allocate sysmem through fb heap
[  +0,000007] NVRM: nvAssertOkFailedNoLog: Assertion failed: Out of memory [NV_ERR_NO_MEMORY] (0x00000051) returned from pRmApi->Alloc(pRmApi, device->session->handle, isSystemMemory ? device->handle : device->subhandle, &physHandle, isSystemMemory ? NV01_MEMORY_SYSTEM : NV01_MEMORY_LOCAL_USER, &memAllocParams, sizeof(memAllocParams)) @ nv_gpu_ops.c:4647
[  +0,106133] NVRM: GPU at PCI:0000:01:00: GPU-0c71241b-6112-2e53-8d60-a2c6e634e234
[  +0,000006] NVRM: GPU Board Serial Number: 0
[  +0,000001] NVRM: Xid (PCI:0000:01:00): 32, pid=22827, name=vo, Channel ID 0000003d intr0 00000000 intr1 80000000
[  +0,016731] NVRM: Xid (PCI:0000:01:00): 32, pid=22827, name=vo, Channel ID 0000003d intr0 00000000 intr1 80000000
```

For example - mpv

```
[vaapi] libva: /usr/lib64/dri/nvidia_drv_video.so init failed
[ffmpeg] av_log callback called with bad parameters (NULL AVClass).
[ffmpeg] This is a bug in one of FFmpeg libraries used.
[ffmpeg] Unable to submit command buffer: VK_ERROR_DEVICE_LOST
[ffmpeg/video] h264: get_buffer() failed
[ffmpeg/video] h264: decode_slice_header error
[ffmpeg/video] h264: no frame!
Error while decoding frame (hardware decoding)!
[ffmpeg] av_log callback called with bad parameters (NULL AVClass).
[ffmpeg] This is a bug in one of FFmpeg libraries used.
```

BUT next try it will be okay. Issue is not consistent. Sometimes it may happen right after suspend wake up.

### To Reproduce

Try to use player with hardware decoding support (mpv, firefox, etc).

### Bug Incidence

Sometimes

### nvidia-bug-report.log.gz

[nvidia-bug-report.log.gz](https://github.com/user-attachments/files/20032620/nvidia-bug-report.log.gz)

LesterLyu, 2025-05-16:
Got the same random issue on 5060Ti on 575.51.02 when triggering ffmpeg decoding.

```
[52595.758098] NVRM: nvCheckOkFailedNoLog: Check failed: Out of memory [NV_ERR_NO_MEMORY] (0x00000051) returned from _memdescAllocInternal(pMemDesc) @ mem_desc.c:1353
[52595.759943] NVRM: sysmemConstruct_IMPL: *** Cannot allocate sysmem through fb heap
[52595.760899] NVRM: nvAssertOkFailedNoLog: Assertion failed: Out of memory [NV_ERR_NO_MEMORY] (0x00000051) returned from pRmApi->Alloc(pRmApi, device->session->handle, isSystemMemory ? device->handle : device->subhandle, &physHandle, isSystemMemory ? NV01_MEMORY_SYSTEM : NV01_MEMORY_LOCAL_USER, &memAllocParams, sizeof(memAllocParams)) @ nv_gpu_ops.c:4647
[52611.858258] NVRM: nvCheckOkFailedNoLog: Check failed: Out of memory [NV_ERR_NO_MEMORY] (0x00000051) returned from _memdescAllocInternal(pMemDesc) @ mem_desc.c:1353
[52611.860142] NVRM: sysmemConstruct_IMPL: *** Cannot allocate sysmem through fb heap
[52611.861108] NVRM: nvAssertOkFailedNoLog: Assertion failed: Out of memory [NV_ERR_NO_MEMORY] (0x00000051) returned from pRmApi->Alloc(pRmApi, device->session->handle, isSystemMemory ? device->handle : device->subhandle, &physHandle, isSystemMemory ? NV01_MEMORY_SYSTEM : NV01_MEMORY_LOCAL_USER, &memAllocParams, sizeof(memAllocParams)) @ nv_gpu_ops.c:4647
[52620.465122] NVRM: nvCheckOkFailedNoLog: Check failed: Out of memory [NV_ERR_NO_MEMORY] (0x00000051) returned from _memdescAllocInternal(pMemDesc) @ mem_desc.c:1353
[52620.466954] NVRM: sysmemConstruct_IMPL: *** Cannot allocate sysmem through fb heap
[52620.468136] NVRM: nvAssertOkFailedNoLog: Assertion failed: Out of memory [NV_ERR_NO_MEMORY] (0x00000051) returned from pRmApi->Alloc(pRmApi, device->session->handle, isSystemMemory ? device->handle : device->subhandle, &physHandle, isSystemMemory ? NV01_MEMORY_SYSTEM : NV01_MEMORY_LOCAL_USER, &memAllocParams, sizeof(memAllocParams)) @ nv_gpu_ops.c:4647
```

pinkycollie, 2025-05-16:
Fixing this means either freeing VRAM, tuning your workloads, updating drivers, or hardware upgrades.

x5f3759df, 2025-05-25:
My VRAM was free for that moment. At least more than 8+ gigs I guess.

LesterLyu, 2025-05-25:
I’ve found a setup that greatly improves stability:
- **Boot in UEFI mode** with CSM disabled, and enable **Resizable BAR**.
- Add the kernel parameters `cma=2048M nvidia_drm.modeset=1` to GRUB.
- If you use ZFS, cap ARC to a reasonable size for your RAM—for example, `zfs.zfs_arc_max=8589934592` (8 GiB).
- Before you activate the GPU, keep at least **6 GiB of free system RAM**. The driver often fails when caches and buffers consume most of the memory, suggesting it has low allocation priority. When memory is tight, running `echo 3 > /proc/sys/vm/drop_caches` frequently resolves the issue.
- And this is not a VRAM issue.

Cloudwalk9, 2025-05-28:
The driver itself seems to be running out of *kernel* memory. Definitely not a VRAM issue.

x5f3759df, 2025-05-29:
Looks like not a VRAM issue indeed. Not good news, because I have 32 gigs of RAM - this is strange

gouthamravee, 2025-06-08:
I'm seeing the same problem with a 1660 on debian. I'm only using the GPU for Plex and Immich transcoding, nothing fancy. Logs attached below. 

This was happening on `575.51.02`, I don't remember what the previous version I installed was so for now I rolled back to `570.153.02`. 

```
[1580597.411153] nvidia 0000:01:00.0: firmware: direct-loading firmware nvidia/575.51.02/gsp_tu10x.bin
[1580597.598572] NVRM: nvCheckOkFailedNoLog: Check failed: Out of memory [NV_ERR_NO_MEMORY] (0x00000051) returned from _memdescAllocInternal(pMemDesc) @ mem_desc.c:1353
[1580597.598590] NVRM: nvAssertOkFailedNoLog: Assertion failed: Out of memory [NV_ERR_NO_MEMORY] (0x00000051) returned from status @ kernel_gsp.c:4615
[1580597.598626] NVRM: nvCheckOkFailedNoLog: Check failed: Out of memory [NV_ERR_NO_MEMORY] (0x00000051) returned from kgspCreateRadix3(pGpu, pKernelGsp, &pKernelGsp->pGspUCodeRadix3Descriptor, NULL, pGspFw->pImageData, pGspFw->imageSize) @ kernel_gsp.c:4502
[1580597.598631] NVRM: kgspInitRm_IMPL: Error preparing GSP-RM image
[1580597.598642] NVRM: RmInitAdapter: Cannot initialize GSP firmware RM
[1580597.608423] NVRM: GPU 0000:01:00.0: RmInitAdapter failed! (0x62:0x51:1941)
[1580597.616708] NVRM: GPU 0000:01:00.0: rm_init_adapter failed, device minor number 0
[1580597.671513] nvidia 0000:01:00.0: firmware: direct-loading firmware nvidia/575.51.02/gsp_tu10x.bin
[1580597.797190] NVRM: nvCheckOkFailedNoLog: Check failed: Out of memory [NV_ERR_NO_MEMORY] (0x00000051) returned from _memdescAllocInternal(pMemDesc) @ mem_desc.c:1353
[1580597.797217] NVRM: nvAssertOkFailedNoLog: Assertion failed: Out of memory [NV_ERR_NO_MEMORY] (0x00000051) returned from status @ kernel_gsp.c:4615
[1580597.797310] NVRM: nvCheckOkFailedNoLog: Check failed: Out of memory [NV_ERR_NO_MEMORY] (0x00000051) returned from kgspCreateRadix3(pGpu, pKernelGsp, &pKernelGsp->pGspUCodeRadix3Descriptor, NULL, pGspFw->pImageData, pGspFw->imageSize) @ kernel_gsp.c:4502
[1580597.797317] NVRM: kgspInitRm_IMPL: Error preparing GSP-RM image
[1580597.797334] NVRM: RmInitAdapter: Cannot initialize GSP firmware RM
[1580597.804417] NVRM: GPU 0000:01:00.0: RmInitAdapter failed! (0x62:0x51:1941)
[1580597.817598] NVRM: GPU 0000:01:00.0: rm_init_adapter failed, device minor number 0
[1654968.764693] NVRM: nvAssertFailedNoLog: Assertion failed: rmapiLockIsOwner() && rmGpuLockIsOwner() @ conf_compute_api.c:77
[1657982.640877] nvidia 0000:01:00.0: firmware: direct-loading firmware nvidia/575.51.02/gsp_tu10x.bin
[1657982.761434] NVRM: nvCheckOkFailedNoLog: Check failed: Out of memory [NV_ERR_NO_MEMORY] (0x00000051) returned from _memdescAllocInternal(pMemDesc) @ mem_desc.c:1353
[1657982.761446] NVRM: nvAssertOkFailedNoLog: Assertion failed: Out of memory [NV_ERR_NO_MEMORY] (0x00000051) returned from status @ kernel_gsp.c:4615
[1657982.761463] NVRM: nvCheckOkFailedNoLog: Check failed: Out of memory [NV_ERR_NO_MEMORY] (0x00000051) returned from kgspCreateRadix3(pGpu, pKernelGsp, &pKernelGsp->pGspUCodeRadix3Descriptor, NULL, pGspFw->pImageData, pGspFw->imageSize) @ kernel_gsp.c:4502
[1657982.761467] NVRM: kgspInitRm_IMPL: Error preparing GSP-RM image
[1657982.761476] NVRM: RmInitAdapter: Cannot initialize GSP firmware RM
[1657982.768674] NVRM: GPU 0000:01:00.0: RmInitAdapter failed! (0x62:0x51:1941)
[1657982.775064] NVRM: GPU 0000:01:00.0: rm_init_adapter failed, device minor number 0
[1658531.842972] [drm] [nvidia-drm] [GPU ID 0x00000100] Unloading driver
[1658531.939452] nvidia-modeset: Unloading
[1658532.029494] nvidia-nvlink: Unregistered Nvlink Core, major device number 241
[1658834.467958] nvidia-nvlink: Nvlink Core is being initialized, major device number 241
[1658834.477610] nvidia 0000:01:00.0: vgaarb: changed VGA decodes: olddecodes=none,decodes=none:owns=none
[1658834.521419] NVRM: loading NVIDIA UNIX Open Kernel Module for x86_64  570.153.02  Release Build  (dvs-builder@U22-A23-20-3)  Tue May 13 16:34:58 UTC 2025
```

Cloudwalk9, 2025-06-27:
The driver has a kernel memory page leak. nvidia_pte_cache in slabtop looked pretty suspect.

Mario156090, 2025-07-06:
Same problem here. :(

PsyonicOne, 2025-10-04:
I seem to be still getting this issue on driver 580.82.07
I get 100s of this:

`NVRM: gpuHandleSanityCheckRegReadError_GM107: Possible bad register read: addr: 0xbb0080,  regvalue: 0xbad1e5a0,  error code: Unknown SYS_PRI_ERROR_CODE`

Then a few of these:

`NVRM: nvAssertFailedNoLog: Assertion failed: (status == NV_OK) || (status == NV_ERR_GPU_IN_FULLCHIP_RESET) @ mem.c:180`

Then it crashes with a few of:

`NVRM: nvAssertOkFailedNoLog: Assertion failed: Out of memory [NV_ERR_NO_MEMORY] (0x00000051) returned from pRmApi->AllocWithSecInfo(pRmApi, hClient, hParent, &pChannelGpfifoParams->hPhysChannelGroup, KEPLER_CHANNEL_GROUP_A, NV_PTR_TO_NvP64(&tsgParams), sizeof(tsgParams), RMAPI_ALLOC_FLAGS_SKIP_RPC, NvP64_NULL, &pRmApi->defaultSecInfo) @ kernel_channel.c:381`