NVIDIA/open-gpu-kernel-modules #851: RTX 5090 FE doesn't support 480Hz refresh rate for PG27AQDP with DP 1.4 (DSC)

silvanshade, 2025-05-06:
### NVIDIA Open GPU Kernel Modules Version

570.144, 575.51.02 Beta

### Please confirm this issue does not happen with the proprietary driver (of the same version). This issue tracker is only for bugs specific to the open kernel driver.

- [x] I confirm that this does not happen with the proprietary driver package.

### Operating System and Version

CachyOS April 2025

### Kernel Release

6.14.5

### Please confirm you are running a stable release kernel (e.g. not a -rc). We do not accept bug reports for unreleased kernels.

- [x] I am running on a stable kernel release.

### Hardware: GPU

GPU 0: NVIDIA GeForce RTX 5090 (UUID: GPU-50649b99-868e-32a7-99da-2d69ca3ae9b7)

### Describe the bug

I am using an RTX 5090 FE with the Asus PG27AQDP OLED monitor which has both Displayport 1.4 and HDMI 2.1 inputs.

Desktop environment is KDE Plasma `6.3.4` Wayland although this behavior is the same with Plasma X11.

When using the DP 1.4 inputs, the RTX 5090 will not allow to set the display to 480Hz refresh rate. The maximum that can be selected is 360Hz.

The nvidia settings app run under Plasma X11 shows that the connection link is `4 lanes @ 8.10 Gbps` which should be enough bandwidth with DSC to support 480Hz:

[image]

This is the same connection link configuration that runs 2560x1440 @ 480Hz 10-bit HDR under Windows 11:

[image]

[image]

This appears to be a bug with DisplayPort under Linux specifically.

Alternatively, the PG27AQDP *does* work correctly at 480Hz under Linux if using the HDMI 2.1 port.

I also verified that this is not a problem with the EDID as reported over DisplayPort because I dumped it from the nvidia settings app and loaded it into AW EDID editor and you can clearly see "display range limits and additional timings" block shows the monitor supporting up to 480Hz:

[image]

[image]

The EDID reported over DP by nvidia-settings under linux and the CRU tool under Windows is in fact identical bitwise.

### To Reproduce

Connect Asus PG27AQDP to RTX 5090 FE DisplayPort and try setting monitor to 480Hz under Plasma Wayland or Plasma X11.

### Bug Incidence

Always

### nvidia-bug-report.log.gz

[nvidia-bug-report.log.gz](https://github.com/user-attachments/files/20072411/nvidia-bug-report.log.gz)

amrit1711, 2025-05-23:
Thank you for reporting issue, I have filed a bug 5301403 internally for tracking purpose.
Shall check for similar display internally and attempt for local repro.

Binary-Eater, 2025-10-30:
@silvanshade I suspect the fix involved for #816 will resolve this issue. The fix has not made it to a driver release yet.

amrit1711, 2025-11-26:
Hi @silvanshade 
Please confirm if you get a chance to test with driver 580.105.08