NVIDIA/open-gpu-kernel-modules #854: Fails to build on Fedora 42

Naros, 2025-05-10:
### NVIDIA Open GPU Kernel Modules Version

e00332b0

### Operating System and Version

Fedora 42

### Kernel Release

6.14.5-300.fc42.x86_64

### Please confirm you are running a stable release kernel (e.g. not a -rc). We do not accept bug reports for unreleased kernels.

- [x] I am running on a stable kernel release.

### Build Command

`make modules -j$(nproc)`

### Terminal output/Build Log

[build.log](https://github.com/user-attachments/files/20140973/build.log)

zcobol, 2025-05-12:
The easiest way to install from source, until an official fix, is as explained in https://github.com/NVIDIA/open-gpu-kernel-modules/issues/827

Just modify two lines in `kernel-open/common/inc/conftest.h` by adding `-std=gnu17` compiler flag and it builds on `FC42` with `GCC15` also.

Or go the `rpmfusion` option.

njgrisafi, 2025-05-17:
> Or go the rpmfusion option.

Can you provide the best way to do this?

I followed the kernel open guide here https://rpmfusion.org/Howto/NVIDIA#Kernel_Open but having issues finding the repository.

x5f3759df, 2025-05-25:
You can use rawhide repo to install latest beta driver

```
sudo dnf install rpmfusion-nonfree-release-rawhide
sudo dnf --enable-repo=rpmfusion-nonfree-rawhide update *\nvidia\*
```

Naros, 2025-08-26:
Last time I tried to use rpmfusion, it did not support the 50-series cards.