NVIDIA/open-gpu-kernel-modules #856: akmods-nvidia-open not installing the open-source driver

ethanbergstrom, 2025-05-18:
### NVIDIA Open GPU Kernel Modules Version

570.144-1.fc42.x86_64

### Please confirm this issue does not happen with the proprietary driver (of the same version). This issue tracker is only for bugs specific to the open kernel driver.

- [x] I confirm that this does not happen with the proprietary driver package.

### Operating System and Version

Fedora release 42 (Adams) - Silverblue / Atomic GNOME

### Kernel Release

Linux exia 6.14.6-300.fc42.x86_64 #1 SMP PREEMPT_DYNAMIC Fri May  9 20:11:19 UTC 2025 x86_64 GNU/Linux

### Please confirm you are running a stable release kernel (e.g. not a -rc). We do not accept bug reports for unreleased kernels.

- [x] I am running on a stable kernel release.

### Hardware: GPU

NVIDIA Corporation GB203 [GeForce RTX 5070 Ti]

### Describe the bug

The package `akmod-nvidia-open` (which also installs `akmod-nvidia`) install via `rpm-ostree` is compiling and installing the proprietary driver instead of the open source one, and the proprietary drivers are not compatible with Blackwell-and-higher / 5000-series card.

I've already tried the `%_with_kmod_nvidia_open 1` trick, but that seems to not work in OSTree-based systems.

I have no issues with the proprietary driver on the same PC with a 3000-series card.

### To Reproduce

* Start with a default installation (`nouveau` loads by default)
* `rpm-ostree install akmod-nvidia-open xorg-x11-drv-nvidia`
* Reboot
* `modinfo -l nvidia`

Expected results:
* `modinfo -l nvidia` should return `Dual MIT/GPL` or similar
* Loaded driver is compatible with 5000-series card

Actual results:
* `modinfo -l nvidia` returns `NVIDIA`
* Loaded driver indicates it is not open-source, despite what was installed:
```
NVRM: The NVIDIA GPU 0000:01:00.0 (PCI ID: 10de:2c05)
NVRM: installed in this system requires use of the NVIDIA open kernel modules.
```

Note, I'm **not** using Secureboot, and haven't seen any indications in the logs of failure due to driver signing problems:
```
May 18 09:34:38 fedora kernel: secureboot: Secure boot disabled
```

### Bug Incidence

Always

### nvidia-bug-report.log.gz

[nvidia-bug-report.log.gz](https://github.com/user-attachments/files/20274801/nvidia-bug-report.log.gz)

aritger, 2025-05-19:
This sounds like a packaging issue, and not an issue with the open-gpu-kernel-modules itself.  Does the discussion here:
https://github.com/NVIDIA/open-gpu-kernel-modules/issues/700
help at all?

ethanbergstrom, 2025-05-20:
Thanks for taking a look!

Yeah, I'd already reviewed those, but unfortunately they did not help.

zcobol, 2025-05-20:
It's definitely a packaging issue. I run into the same issue on FC42 Workstation, no rpm-ostree, and after reboot there were this kind of entries in the logs:

```
[    6.712455] NVRM: The NVIDIA GPU 0000:01:00.0 (PCI ID: 10de:2d04)
               NVRM: installed in this system requires use of the NVIDIA open kernel modules.
```

To use the nvidia-open driver I had to force a rebuild:

```
zcobol@paloalto:~$ sudo akmods --kernels $(uname -r) --rebuild --force
Checking kmods exist for 6.14.6-300.fc42.x86_64            [  OK  ]
Building and installing nvidia-kmod                        [  OK  ]
Building and installing nvidia-open-kmod                   [  OK  ]
```

On restart the correct module was loaded:

```
zcobol@paloalto:~$ modinfo nvidia
filename:       /lib/modules/6.14.6-300.fc42.x86_64/extra/nvidia/nvidia.ko
import_ns:      DMA_BUF
alias:          char-major-195-*
version:        570.144
supported:      external
license:        Dual MIT/GPL
```

Codalarache, 2025-05-21:
@zcobol I don't know how you managed to make it work... I have the same fedora (42 workstation), same kernel. I installed akmod-nvidia-open, launched the same command you did, reboot but I still got the same error and license : 

NVRM: installed in this system requires use of the NVIDIA open kernel modules.

modinfo nvidia
filename:       /lib/modules/6.14.6-300.fc42.x86_64/extra/nvidia/nvidia.ko
alias:          char-major-195-*
version:        570.144
supported:      external
license:        NVIDIA
firmware:       nvidia/570.144/gsp_tu10x.bin
firmware:       nvidia/570.144/gsp_ga10x.bin

Which GPU do you have ? Mine is a Geforce 5080 running in a Razer Blade 16 2025.

Thanks in advance.

zcobol, 2025-05-22:
I'm using a `RTX 5060 Ti`. To get the nvidia open source driver to work on Fedora 42 I followed the instruction in this post:

https://discussion.fedoraproject.org/t/nvidia-driver-install-issues-on-fedora-42/149091/6

```
INSTALL NVDIA OPEN SOURCE DRIVERS

sudo dnf install https://mirrors.rpmfusion.org/free/fedora/rpmfusion-free-release-$(rpm -E %fedora).noarch.rpm https://mirrors.rpmfusion.org/nonfree/fedora/rpmfusion-nonfree-release-$(rpm -E %fedora).noarch.rpm

sudo dnf install rpmfusion-nonfree-release-tainted

sudo dnf swap akmod-nvidia akmod-nvidia-open

sudo reboot now
```

Codalarache, 2025-05-22:
Thanks for the reply. So it seems there is a bug somewhere. I tried again to follow the instructions but nothing changed.

`> sudo dnf install https://mirrors.rpmfusion.org/free/fedora/rpmfusion-free-release-$(rpm -E %fedora).noarch.rpm https://mirrors.rpmfusion.org/nonfree/fedora/rpmfusion-nonfree-release-$(rpm -E %fedora).noarch.rpm
Updating and loading repositories:
 RPM Fusion for Fedora 42 - Free tainted                              100% |  26.4 KiB/s |   8.0 KiB |  00m00s
 Fedora 42 - x86_64 - Updates                                         100% |  20.2 KiB/s |  12.8 KiB |  00m01s
 Fedora 42 - x86_64 - Updates                                         100% | 617.7 KiB/s |   3.1 MiB |  00m05s
Repositories loaded.
 https://mirrors.rpmfusion.org/free/fedora/rpmfusion-free-release-42. 100% |  54.1 KiB/s |  11.3 KiB |  00m00s
 https://mirrors.rpmfusion.org/nonfree/fedora/rpmfusion-nonfree-relea 100% |  53.9 KiB/s |  11.4 KiB |  00m00s
Package "rpmfusion-free-release-42-1.noarch" is already installed.
Package "rpmfusion-nonfree-release-42-1.noarch" is already installed.

Nothing to do.
> sudo dnf install rpmfusion-nonfree-release-tainted
Updating and loading repositories:
Repositories loaded.
Package "rpmfusion-nonfree-release-tainted-42-1.noarch" is already installed.

Nothing to do.
> sudo dnf swap akmod-nvidia akmod-nvidia-open
Updating and loading repositories:
Repositories loaded.
Package "akmod-nvidia-open-3:570.144-1.fc42.x86_64" is already installed.
No packages to remove for argument: akmod-nvidia

Nothing to do.
`

And still complaining "this system requires use of the NVIDIA open kernel modules"

elsaco, 2025-05-22:
As mentioned by @ethanbergstrom you also need to add an rpm macro to `/etc/rpm/macros.nvidia-kmod` containing
`%_with_kmod_nvidia_open 1`. If the file is not present create one.

Codalarache, 2025-05-22:
Hi, I already did this when searching for a solution

```
> cat /etc/rpm/macros.nvidia-kmod 
%_with_kmod_nvidia_open 1
> ll /etc/rpm/macros.nvidia-kmod 
-rw-r--r--. 1 root root 26 May 20 18:58 /etc/rpm/macros.nvidia-kmod
```

ethanbergstrom, 2025-05-23:
Regardless, the issue is still present on OSTree-based systems.

elsaco, 2025-05-23:
Or switch to `CachyOS` 😀

rafnogueira, 2025-08-24:
> Or switch to `CachyOS` 😀

Please don't suggest broken distros for Nvidia users. I’ve been trying to use CachyOS for over a year, and the distro doesn’t work well with NVIDIA and Wayland. Suggest Arch Linux instead. I’m using the latest NVIDIA 580 drivers and Kernel 6.16, and I’m still having games that don’t work—like Digimon Cyber Story, Age of Mythology: Retold, and others. I don’t have any of these problems on Ubuntu, Fedora, Arch, or other distros.

joaorura, 2026-08-31:
Related data point from a traditional (non-OSTree) Fedora 44 + dnf setup, in case it helps others landing on this issue with similar symptoms:

On a Dell Pro Max 16 laptop (Intel Arrow Lake-H + NVIDIA RTX PRO 1000 Blackwell, muxless Optimus), installing the **plain** `akmod-nvidia` package (no `-open` suffix) via `dnf` — **not** `akmod-nvidia-open` — worked correctly and automatically built the open kernel module. Confirmed in the akmods build log:

```
Runtime detection of kmod_nvidia_open
Using nvidia-open
```

and at runtime: `dmesg | grep NVRM` → `loading NVIDIA UNIX Open Kernel Module for x86_64 610.57.04`, and `modinfo nvidia | grep license` → `Dual MIT/GPL`. `nvidia-smi` and PRIME render offload both work fine.

So on classic (non-OSTree) Fedora, `nvidia-kmod.spec`'s build-time `lspci`-based auto-detection (`nvidia-kmod-noopen-checks` against `supported-gpus/nvidia-kmod-noopen-pciids.txt`) does correctly select the open source tree for Blackwell without needing the standalone `-open` package at all — this issue's `rpm-ostree`-specific reproduction (`%_with_kmod_nvidia_open`/detection apparently not working the same way under OSTree) may be worth separating from the "auto-detect doesn't work" case, since on traditional installs it does work as designed.

Separately, we ran into a related packaging snag while experimenting with the standalone `akmod-nvidia-open` package specifically (before realizing it wasn't needed): at the time, the plain `rpmfusion-nonfree` repo's `akmod-nvidia-open` was still on driver 595.58.03 while `xorg-x11-drv-nvidia`/`akmod-nvidia` were already on 610.57.04 — the matching 610.57.04 build of `akmod-nvidia-open` only exists in `rpmfusion-nonfree-tainted`. On top of that, `xorg-x11-drv-nvidia` has a hard `Requires: nvidia-kmod >= <version>`, which only the proprietary `akmod-nvidia`/`kmod-nvidia` packages `Provide` — `kmod-nvidia-open` does not. So removing the proprietary `kmod-nvidia-<kernel>` package while trying to keep only `-open` cascades into removing `xorg-x11-drv-nvidia` and everything depending on it, since nothing else satisfies that requirement. Filed as a separate, more precisely scoped report against rpmfusion/nvidia-kmod.