NVIDIA/open-gpu-kernel-modules #875: Not booting (KP) with NVIDIA GeForce RTX™ 3050 6GB Laptop GPU

francoism90, 2025-06-06:
### NVIDIA Open GPU Kernel Modules Version

570.153.02-3

### Please confirm this issue does not happen with the proprietary driver (of the same version). This issue tracker is only for bugs specific to the open kernel driver.

- [x] I confirm that this does not happen with the proprietary driver package.

### Operating System and Version

CachyOS (arch)

### Kernel Release

Linux 6.15.1-3-cachyos #1 SMP PREEMPT_DYNAMIC Thu, 05 Jun 2025 15:40:37 +0000 x86_64 GNU/Linux

### Please confirm you are running a stable release kernel (e.g. not a -rc). We do not accept bug reports for unreleased kernels.

- [x] I am running on a stable kernel release.

### Hardware: GPU

GPU 0: NVIDIA GeForce RTX 3050 6GB Laptop GPU

### Describe the bug

I cannot boot after upgrading to the latest open driver version, it works fine on the closed driver.

It seems to crash and fully freeze the laptop at boot, the only workaround is switching to open or downgrade.

### To Reproduce

Upgrade to 

### Bug Incidence

Once

### nvidia-bug-report.log.gz

1. Upgrade to 570.153.02-3
2. Reboot
3. Crash on boot
4. Hard-freeze

ptr1337, 2025-06-07:
Hey,

We have found a "fix" for this. The 575 driver seems to have problems with Clang built Kernels.

Since [Clang Built kernels are officially supported ](https://github.com/NVIDIA/open-gpu-kernel-modules#supported-toolchains)I would suggest that NVIDIA adds an additional CI to their automated testing?

francoism90, 2025-06-07:
@ptr1337 Yep, sorry! I'll update the issue. :)

ptr1337, 2025-06-07:
> [@ptr1337](https://github.com/ptr1337) Yep, sorry! I'll update the issue. :)

Please let me know, if it works after updating. We got some confirmations but would be good to get more :)

francoism90, 2025-06-07:
@ptr1337 I'll test, thanks for the quick fix!

I don't know if I should use the open/close one though. It seems the close one does offer more features and it actually does better power management.

But I do like the open driver, because it's better integrated into the kernel.

What do you recommend?

ptr1337, 2025-06-07:
> [@ptr1337](https://github.com/ptr1337) I'll test, thanks for the quick fix!
>
> I don't know if I should use the open/close one though. It seems the close one does offer more features and it actually [...]

The open one is fine.

francoism90, 2025-06-07:
@ptr1337 Just to confirm the open driver now works again, and I'll stick to them. It seems they are almost on par with the closed one, and even offer some extra features.

ptr1337, 2025-06-08:
Ive made a proper issue here, which also contains logs of the crash:
https://github.com/NVIDIA/open-gpu-kernel-modules/issues/877

francoism90, 2025-06-09:
Closing this issue, and continue in #877.