NVIDIA/open-gpu-kernel-modules #876: RTX 5070 TI Mobile not being detected on latest open driver

nekromantik, 2025-06-07:
### NVIDIA Open GPU Kernel Modules Version

575.57.08

### Please confirm this issue does not happen with the proprietary driver (of the same version). This issue tracker is only for bugs specific to the open kernel driver.

- [ ] I confirm that this does not happen with the proprietary driver package.

### Operating System and Version

CachyOS

### Kernel Release

6.14.4

### Please confirm you are running a stable release kernel (e.g. not a -rc). We do not accept bug reports for unreleased kernels.

- [x] I am running on a stable kernel release.

### Hardware: GPU

RTX 5070 TI

### Describe the bug

Card does not seem to be detected. Running nvidia-smi shows no devices.
Downgrading driver fixes the issue.

### To Reproduce

install latest driver
run nvidia-ami
try switching to dGPU

### Bug Incidence

Always

### nvidia-bug-report.log.gz

[nvidia-bug-report.log.gz](https://github.com/user-attachments/files/20640146/nvidia-bug-report.log.gz)

Binary-Eater, 2025-06-07:
Hi @nekromantik,

From your logs alone, I see, with this driver, we are failing to initialize GSP on the GPU, leading to the symptoms you are seeing.

```
Jun 07 19:37:10 cachyos-g14 kernel: nvidia-nvlink: Nvlink Core is being initialized, major device number 240
Jun 07 19:37:10 cachyos-g14 kernel: NVRM: loading NVIDIA UNIX Open Kernel Module for x86_64  575.57.08  Release Build  (notroot@)  Fri Jun  6 21:24:15 UTC 2025
Jun 07 19:37:10 cachyos-g14 kernel: nvidia-modeset: Loading NVIDIA UNIX Open Kernel Mode Setting Driver for x86_64  575.57.08  Release Build  (notroot@)  Fri Jun  6 21:24:06 UTC 2025
Jun 07 19:37:10 cachyos-g14 kernel: [drm] [nvidia-drm] [GPU ID 0x00006400] Loading driver
Jun 07 19:37:10 cachyos-g14 kernel: NVRM: testIfDsmSubFunctionEnabled: GPS ACPI DSM called before _acpiDsmSupportedFuncCacheInit subfunction = 11.
Jun 07 19:37:10 cachyos-g14 kernel: NVRM: kgspHealthCheck_TU102: ****************************** GSP-CrashCat Report *******************************
Jun 07 19:37:10 cachyos-g14 kernel: NVRM: GPU at PCI:0000:64:00: GPU-b4c029e6-637f-bc9b-6c84-28d005fbad48
Jun 07 19:37:10 cachyos-g14 kernel: NVRM: Xid (PCI:0000:64:00): 120, GSP task exception: load access page fault (cause:0xd) @ pc:0x140c7f6, partition:4#0, task:3
Jun 07 19:37:10 cachyos-g14 kernel: NVRM:     Reported by libos partition:4#5 kernel v3.1 [0] @ ts:10
Jun 07 19:37:10 cachyos-g14 kernel: NVRM:     RISC-V CSR State:
Jun 07 19:37:10 cachyos-g14 kernel: NVRM:         sstatus:0x0000000200000020  sscratch:0xffffffffa30144d0     sie:0x0000000000000220  sip:0x0000000000000000
Jun 07 19:37:10 cachyos-g14 kernel: NVRM:         sepc:0x000000000140c7f6     stval:0x0000000000000000  scause:0x000000000000000d
Jun 07 19:37:10 cachyos-g14 kernel: NVRM:     RISC-V GPR State:
Jun 07 19:37:10 cachyos-g14 kernel: NVRM:         ra:0x000000000140cea2   sp:0x00000045f4c0f5b0   gp:0x0000000000000000   tp:0x0000000000000000
Jun 07 19:37:10 cachyos-g14 kernel: NVRM:         a0:0x0000000000000000   a1:0x00000045eda20530   a2:0x0000000000000004   a3:0x00000045f5241000
Jun 07 19:37:10 cachyos-g14 kernel: NVRM:         a4:0x0000000000000000   a5:0x0000000000000000   a6:0x0000000000001010   a7:0x0000000000000004
Jun 07 19:37:10 cachyos-g14 kernel: NVRM:         s0:0x00000045f4c0f740   s1:0x00000045edc43850   s2:0x0000000000000002   s3:0x00000000017d4e9a
Jun 07 19:37:10 cachyos-g14 kernel: NVRM:         s4:0x000000000411bb70   s5:0x00000000001a8000   s6:0x00000045edb805b0   s7:0x0000000000001500
Jun 07 19:37:10 cachyos-g14 kernel: NVRM:         s8:0x000000000411c088   s9:0x0000000000000000  s10:0x0000000000000000  s11:0x00000045edb7e5b0
Jun 07 19:37:10 cachyos-g14 kernel: NVRM:         t0:0x0000000000000020   t1:0x0000000000000001   t2:0x0000000000000000   t3:0x0000000000000020
Jun 07 19:37:10 cachyos-g14 kernel: NVRM:         t4:0x0000000000000000   t5:0x00000045f4c0f3c1   t6:0x0000000000000020
Jun 07 19:37:10 cachyos-g14 kernel: NVRM:     Stack Trace:
Jun 07 19:37:10 cachyos-g14 kernel: NVRM:         0x000000000140c7f6
Jun 07 19:37:10 cachyos-g14 kernel: NVRM:         0x00000000017d4e9a
Jun 07 19:37:10 cachyos-g14 kernel: NVRM:         0x00000000017dd84e
Jun 07 19:37:10 cachyos-g14 kernel: NVRM:         0x00000000017e001c
Jun 07 19:37:10 cachyos-g14 kernel: NVRM:         0x00000000017d411a
Jun 07 19:37:10 cachyos-g14 kernel: NVRM:         0x000000000141629e
Jun 07 19:37:10 cachyos-g14 kernel: NVRM:         0x0000000001a285da
Jun 07 19:37:10 cachyos-g14 kernel: NVRM:         0x0000000001a4b698
Jun 07 19:37:10 cachyos-g14 kernel: NVRM:         0x0000000001b887c0
Jun 07 19:37:10 cachyos-g14 kernel: NVRM:         0x0000000001a2d332
Jun 07 19:37:10 cachyos-g14 kernel: NVRM:     Local I/O Register State:
Jun 07 19:37:10 cachyos-g14 kernel: NVRM:         0x01450800:0x00000000   0x01450900:0xbadf202b   0x01450a00:0x00000000   0x01450c00:0x00000000
Jun 07 19:37:10 cachyos-g14 kernel: NVRM:         0x01454a00:0x810400d0   0x01454b00:0x010800d0   0x01454c00:0x00080000   0x01400200:0x00000040
Jun 07 19:37:10 cachyos-g14 kernel: NVRM:     ------------[ end crash report ]------------
Jun 07 19:37:10 cachyos-g14 kernel: NVRM: GPU0 GSP RPC buffer contains function 4128 (GSP_POST_NOCAT_RECORD) and data 0x0000000000000005 0x00000000017d4e9a.
Jun 07 19:37:10 cachyos-g14 kernel: NVRM: GPU0 RPC history (CPU -> GSP):
Jun 07 19:37:10 cachyos-g14 kernel: NVRM:     entry function                   data0              data1              ts_start           ts_end             duration actively_polling
Jun 07 19:37:10 cachyos-g14 kernel: NVRM:      0    73   SET_REGISTRY          0x0000000000000000 0x0000000000000000 0x000636ffa1c9106d 0x0000000000000000          y
Jun 07 19:37:10 cachyos-g14 kernel: NVRM:     -1    72   GSP_SET_SYSTEM_INFO   0x0000000000000000 0x0000000000000000 0x000636ffa1c91060 0x0000000000000000           
Jun 07 19:37:10 cachyos-g14 kernel: NVRM: GPU0 RPC event history (CPU <- GSP):
Jun 07 19:37:10 cachyos-g14 kernel: NVRM:     entry function                   data0              data1              ts_start           ts_end             duration during_incomplete_rpc
Jun 07 19:37:10 cachyos-g14 kernel: NVRM:      0    4128 GSP_POST_NOCAT_RECORD 0x0000000000000005 0x00000000017d4e9a 0x000636ffa1d0c3b2 0x000636ffa1d0c3b5      3us y
Jun 07 19:37:10 cachyos-g14 kernel: NVRM: kgspRcAndNotifyAllChannels_IMPL: RC all user channels for critical error 120.
[... cut ...]
Jun 07 19:37:10 cachyos-g14 kernel: NVRM: GPU 0000:64:00.0: RmInitAdapter failed! (0x62:0x40:1941)
Jun 07 19:37:10 cachyos-g14 kernel: NVRM: GPU 0000:64:00.0: rm_init_adapter failed, device minor number 0
Jun 07 19:37:10 cachyos-g14 kernel: [drm:nv_drm_load [nvidia_drm]] *ERROR* [nvidia-drm] [GPU ID 0x00006400] Failed to allocate NvKmsKapiDevice
Jun 07 19:37:10 cachyos-g14 kernel: [drm:nv_drm_register_drm_device [nvidia_drm]] *ERROR* [nvidia-drm] [GPU ID 0x00006400] Failed to register device
```

nekromantik, 2025-06-07:
ok thanks
will it likely not be fixed in the next driver update?

Binary-Eater, 2025-06-07:
Let me provide some quick updates on the bug. We believe that we understand the cause of the issue and have a fix. We are going through the process of integrating the change in a way where it will be available in future releases. I am not providing a statement about which exact release it will come out in since our release process is involved across our different driver branches, but I'll provide an update when its available in a public release to upgrade to.

runningmaster, 2025-06-09:
I’m experiencing the same issue on Fedora 42 with an RTX 5080 (MSI Vector 16 HX AI A2XWIG laptop): 

`$ sudo dmesg | grep 'NVRM'`
```
[   84.810006] NVRM: _kgspBootGspRm: unexpected WPR2 already up, cannot proceed with booting GSP
[   84.810007] NVRM: _kgspBootGspRm: (the GPU is likely in a bad state and may need to be reset)
[   84.810046] NVRM: RmInitAdapter: Cannot initialize GSP firmware RM
[   84.811857] NVRM: iovaspaceDestruct_IMPL: 1 left-over mappings in IOVAS 0x100
[   84.811879] NVRM: GPU 0000:01:00.0: RmInitAdapter failed! (0x62:0x40:1941)
[   84.813774] NVRM: GPU 0000:01:00.0: rm_init_adapter failed, device minor number 0
[   85.552379] NVRM: testIfDsmSubFunctionEnabled: GPS ACPI DSM called before _acpiDsmSupportedFuncCacheInit subfunction = 11.
[   87.526680] NVRM: kgspHealthCheck_TU102: ****************************** GSP-CrashCat Report *******************************
[   87.526688] NVRM: GPU at PCI:0000:01:00: GPU-29689fa2-005b-bfbd-b7ad-94fde3bc2b16
[   87.526689] NVRM: Xid (PCI:0000:01:00): 120, GSP task exception: load access page fault (cause:0xd) @ pc:0x140c7f6, partition:4#0, task:3
[   87.526693] NVRM:     Reported by libos partition:4#5 kernel v3.1 [0] @ ts:2
[   87.526694] NVRM:     RISC-V CSR State:
[   87.526694] NVRM:         sstatus:0x0000000200000020  sscratch:0xffffffffa30144d0     sie:0x0000000000000220  sip:0x0000000000000000
[   87.526695] NVRM:         sepc:0x000000000140c7f6     stval:0x0000000000000000  scause:0x000000000000000d
[   87.526696] NVRM:     RISC-V GPR State:
[   87.526696] NVRM:         ra:0x000000000140cea2   sp:0x00000047a9a0f5b0   gp:0x0000000000000000   tp:0x0000000000000000
[   87.526696] NVRM:         a0:0x0000000000000000   a1:0x00000047a2820530   a2:0x0000000000000004   a3:0x00000047aa041000
[   87.526697] NVRM:         a4:0x0000000000000000   a5:0x0000000000000000   a6:0x0000000000001010   a7:0x0000000000000004
[   87.526697] NVRM:         s0:0x00000047a9a0f740   s1:0x00000047a2d0f130   s2:0x0000000000000002   s3:0x00000000017d4e9a
[   87.526698] NVRM:         s4:0x000000000411bb70   s5:0x00000000001a8000   s6:0x00000047a29805b0   s7:0x0000000000001500
[   87.526699] NVRM:         s8:0x000000000411c088   s9:0x0000000000000000  s10:0x0000000000000000  s11:0x00000047a297e5b0
[   87.526699] NVRM:         t0:0x0000000000000020   t1:0x0000000000000001   t2:0x0000000000000000   t3:0x0000000000000020
[   87.526700] NVRM:         t4:0x0000000000000000   t5:0x00000047a9a0f3c1   t6:0x0000000000000020
[   87.526700] NVRM:     Stack Trace:
[   87.526700] NVRM:         0x000000000140c7f6
[   87.526701] NVRM:         0x00000000017d4e9a
[   87.526701] NVRM:         0x00000000017dd84e
[   87.526701] NVRM:         0x00000000017e001c
[   87.526701] NVRM:         0x00000000017d411a
[   87.526702] NVRM:         0x000000000141629e
[   87.526702] NVRM:         0x0000000001a285da
[   87.526702] NVRM:         0x0000000001a4b698
[   87.526702] NVRM:         0x0000000001b887c0
[   87.526703] NVRM:         0x0000000001a2d332
[   87.526703] NVRM:     Local I/O Register State:
[   87.526703] NVRM:         0x01450800:0x00000000   0x01450900:0xbadf202b   0x01450a00:0x00000000   0x01450c00:0x00000000
[   87.526704] NVRM:         0x01454a00:0x810400d0   0x01454b00:0x010800d0   0x01454c00:0x00080000   0x01400200:0x00000040
[   87.526705] NVRM:     ------------[ end crash report ]------------
[   87.526719] NVRM: GPU0 GSP RPC buffer contains function 4128 (GSP_POST_NOCAT_RECORD) and data 0x0000000000000005 0x00000000017d4e9a.
[   87.526720] NVRM: GPU0 RPC history (CPU -> GSP):
[   87.526721] NVRM:     entry function                   data0              data1              ts_start           ts_end             duration actively_polling
[   87.526722] NVRM:      0    73   SET_REGISTRY          0x0000000000000000 0x0000000000000000 0x000637252a9b0d06 0x0000000000000000          y
[   87.526723] NVRM:     -1    72   GSP_SET_SYSTEM_INFO   0x0000000000000000 0x0000000000000000 0x000637252a9b0d03 0x0000000000000000
[   87.526724] NVRM: GPU0 RPC event history (CPU <- GSP):
[   87.526725] NVRM:     entry function                   data0              data1              ts_start           ts_end             duration during_incomplete_rpc
[   87.526725] NVRM:      0    4128 GSP_POST_NOCAT_RECORD 0x0000000000000005 0x00000000017d4e9a 0x000637252aa43eac 0x000637252aa43ead      1us y
[   87.526728] NVRM: kgspRcAndNotifyAllChannels_IMPL: RC all user channels for critical error 120.
[   87.526736] NVRM: kgspHealthCheck_TU102: **********************************************************************************
[   87.526738] NVRM: nvCheckOkFailedNoLog: Check failed: Reset required [NV_ERR_RESET_REQUIRED] (0x00000062) returned from rpcRecvPoll(pGpu, pRpc, NV_VGPU_MSG_EVENT_GSP_INIT_DONE) @ kernel_gsp.c:4877
[   87.526739] NVRM: nvAssertOkFailedNoLog: Assertion failed: Reset required [NV_ERR_RESET_REQUIRED] (0x00000062) returned from kgspWaitForRmInitDone(pGpu, pKernelGsp) @ kernel_gsp_gh100.c:952
[   87.526746] NVRM: kgspInitRm_IMPL: Max GSP-RM boot attempts exceeded: 4/4
[   87.526795] NVRM: RmInitAdapter: Cannot initialize GSP firmware RM
[   87.528558] NVRM: iovaspaceDestruct_IMPL: 1 left-over mappings in IOVAS 0x100
[   87.528576] NVRM: GPU 0000:01:00.0: RmInitAdapter failed! (0x62:0x62:1941)
[   87.530845] NVRM: GPU 0000:01:00.0: rm_init_adapter failed, device minor number 0
```
`$ sudo dmesg | grep 'nvidia'`
```
$ sudo dmesg | grep 'nvidia'
[... cut ...]
[   13.106022] nvidia 0000:01:00.0: Enabling HDA controller
[   74.693170] nvidia 0000:01:00.0: Enabling HDA controller
[   74.760923] nvidia 0000:01:00.0: Enabling HDA controller
[   77.437172] nvidia 0000:01:00.0: Enabling HDA controller
[   80.131989] nvidia 0000:01:00.0: Enabling HDA controller
[   82.803835] nvidia 0000:01:00.0: Enabling HDA controller
[   85.516847] nvidia 0000:01:00.0: Enabling HDA controller
[   88.235773] nvidia 0000:01:00.0: Enabling HDA controller
[  263.654391] nvidia 0000:01:00.0: Enabling HDA controller
```

Jubijub, 2025-06-09:
Same on a Lenovo Legion Pro 7i laptop, with a RTX 5080 Mobile

```
jun 09 19:26:17 archlinux kernel: nvidia: loading out-of-tree module taints kernel.
jun 09 19:26:17 archlinux kernel: nvidia: module verification failed: signature and/or required key missing - tainting kernel
jun 09 19:26:17 archlinux kernel: nvidia-nvlink: Nvlink Core is being initialized, major device number 240
jun 09 19:26:17 archlinux kernel: 
jun 09 19:26:17 archlinux kernel: nvidia 0000:02:00.0: vgaarb: VGA decodes changed: olddecodes=io+mem,decodes=none:owns=none
jun 09 19:26:17 archlinux kernel: usb 3-2: New USB device found, idVendor=17ef, idProduct=f006, bcdDevice= 0.34
jun 09 19:26:17 archlinux kernel: usb 3-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
jun 09 19:26:17 archlinux kernel: usb 3-2: Product: Idea5003
jun 09 19:26:17 archlinux kernel: usb 3-2: Manufacturer: ZEPHYR
jun 09 19:26:17 archlinux kernel: usb 3-2: SerialNumber: 0123456789ABCDEF
jun 09 19:26:17 archlinux kernel: NVRM: loading NVIDIA UNIX Open Kernel Module for x86_64  575.57.08  Release Build  (root@)  
jun 09 19:26:17 archlinux kernel: usbcore: registered new interface driver usbhid
jun 09 19:26:17 archlinux kernel: usbhid: USB HID core driver
jun 09 19:26:17 archlinux kernel: hid-generic 0003:17EF:F006.0001: hiddev96,hidraw0: USB HID v1.10 Device [ZEPHYR Idea5003] on usb-0000:80:14.0-2/input1
jun 09 19:26:17 archlinux kernel: nvidia-modeset: Loading NVIDIA UNIX Open Kernel Mode Setting Driver for x86_64  575.57.08  Release Build  (root@)  
jun 09 19:26:17 archlinux kernel: [drm] [nvidia-drm] [GPU ID 0x00000200] Loading driver
jun 09 19:26:17 archlinux kernel: usb 3-3: new full-speed USB device number 3 using xhci_hcd
jun 09 19:26:17 archlinux kernel: usb 3-3: New USB device found, idVendor=048d, idProduct=c193, bcdDevice= 0.08
jun 09 19:26:17 archlinux kernel: usb 3-3: New USB device strings: Mfr=1, Product=2, SerialNumber=0
jun 09 19:26:17 archlinux kernel: usb 3-3: Product: Lenovo Lighting 
jun 09 19:26:17 archlinux kernel: usb 3-3: Manufacturer: ITE Tech. Inc.
jun 09 19:26:17 archlinux kernel: hid-generic 0003:048D:C193.0002: hiddev97,hidraw1: USB HID v1.10 Device [ITE Tech. Inc. Lenovo Lighting ] on usb-0000:80:14.0-3/input0
jun 09 19:26:17 archlinux kernel: hid-generic 0003:048D:C193.0003: hiddev98,hidraw2: USB HID v1.10 Device [ITE Tech. Inc. Lenovo Lighting ] on usb-0000:80:14.0-3/input1
jun 09 19:26:17 archlinux kernel: NVRM: testIfDsmSubFunctionEnabled: GPS ACPI DSM called before _acpiDsmSupportedFuncCacheInit subfunction = 11.
jun 09 19:26:17 archlinux kernel: usb 3-4: new full-speed USB device number 4 using xhci_hcd
jun 09 19:26:17 archlinux kernel: usb 3-4: New USB device found, idVendor=048d, idProduct=c197, bcdDevice=18.00
jun 09 19:26:17 archlinux kernel: usb 3-4: New USB device strings: Mfr=1, Product=2, SerialNumber=0
jun 09 19:26:17 archlinux kernel: usb 3-4: Product: ITE Device(8258)
jun 09 19:26:17 archlinux kernel: usb 3-4: Manufacturer: ITE Tech. Inc.
jun 09 19:26:17 archlinux kernel: input: ITE Tech. Inc. ITE Device(8258) Keyboard as /devices/pci0000:80/0000:80:14.0/usb3/3-4/3-4:1.0/0003:048D:C197.0004/input/input3
jun 09 19:26:17 archlinux kernel: input: ITE Tech. Inc. ITE Device(8258) as /devices/pci0000:80/0000:80:14.0/usb3/3-4/3-4:1.0/0003:048D:C197.0004/input/input4
jun 09 19:26:17 archlinux kernel: input: ITE Tech. Inc. ITE Device(8258) Wireless Radio Control as /devices/pci0000:80/0000:80:14.0/usb3/3-4/3-4:1.0/0003:048D:C197.0004/input/input5
jun 09 19:26:17 archlinux kernel: hid-generic 0003:048D:C197.0004: input,hiddev99,hidraw3: USB HID v1.10 Keyboard [ITE Tech. Inc. ITE Device(8258)] on usb-0000:80:14.0-4/input0
jun 09 19:26:17 archlinux kernel: usb 3-11: new high-speed USB device number 5 using xhci_hcd
jun 09 19:26:17 archlinux kernel: NVRM: kgspHealthCheck_TU102: ****************************** GSP-CrashCat Report *******************************
jun 09 19:26:17 archlinux kernel: NVRM: GPU at PCI:0000:02:00: GPU-af62c0b6-aa09-087a-8baa-a6a903ccebea
jun 09 19:26:17 archlinux kernel: NVRM: Xid (PCI:0000:02:00): 120, GSP task exception: load access page fault (cause:0xd) @ pc:0x140c7f6, partition:4#0, task:3
jun 09 19:26:17 archlinux kernel: NVRM:     Reported by libos partition:4#5 kernel v3.1 [0] @ ts:115
jun 09 19:26:17 archlinux kernel: NVRM:     RISC-V CSR State:
jun 09 19:26:17 archlinux kernel: NVRM:         sstatus:0x0000000200000020  sscratch:0xffffffffa30144d0     sie:0x0000000000000220  sip:0x0000000000000000
jun 09 19:26:17 archlinux kernel: NVRM:         sepc:0x000000000140c7f6     stval:0x0000000000000000  scause:0x000000000000000d
jun 09 19:26:17 archlinux kernel: NVRM:     RISC-V GPR State:
jun 09 19:26:17 archlinux kernel: NVRM:         ra:0x000000000140cea2   sp:0x00000047f240f5b0   gp:0x0000000000000000   tp:0x0000000000000000
jun 09 19:26:17 archlinux kernel: NVRM:         a0:0x0000000000000000   a1:0x00000047eb220530   a2:0x0000000000000004   a3:0x00000047f2a41000
jun 09 19:26:17 archlinux kernel: NVRM:         a4:0x0000000000000000   a5:0x0000000000000000   a6:0x0000000000001010   a7:0x0000000000000004
jun 09 19:26:17 archlinux kernel: NVRM:         s0:0x00000047f240f740   s1:0x00000047eb444c90   s2:0x0000000000000002   s3:0x00000000017d4e9a
jun 09 19:26:17 archlinux kernel: NVRM:         s4:0x000000000411bb70   s5:0x00000000001a8000   s6:0x00000047eb380550   s7:0x0000000000001500
[... cut ...]
jun 09 19:26:17 archlinux kernel: NVRM: iovaspaceDestruct_IMPL: 1 left-over mappings in IOVAS 0x200
jun 09 19:26:17 archlinux kernel: NVRM: GPU 0000:02:00.0: RmInitAdapter failed! (0x62:0x40:1941)
jun 09 19:26:17 archlinux kernel: NVRM: GPU 0000:02:00.0: rm_init_adapter failed, device minor number 0
jun 09 19:26:17 archlinux kernel: [drm:nv_drm_load [nvidia_drm]] *ERROR* [nvidia-drm] [GPU ID 0x00000200] Failed to allocate NvKmsKapiDevice
jun 09 19:26:17 archlinux kernel: [drm:nv_drm_register_drm_device [nvidia_drm]] *ERROR* [nvidia-drm] [GPU ID 0x00000200] Failed to register device
```

Mario156090, 2025-06-09:
Same Issue using MSI Vector 16 HX AI a2xwhg RTX5070TI.

runningmaster, 2025-06-10:
> You can check using:

```bash
$ cat /proc/driver/nvidia/gpus/0000:01:00.0/power
Runtime D3 status:          Disabled by default
Video Memory:               Active

GPU Hardware Support:
 Video Memory Self Refresh: Not Supported
 Video Memory Off:          Supported

S0ix Power Management:
 Platform Support:          Supported
 Status:                    Disabled

Notebook Dynamic Boost:     Supported
```

Mario156090, 2025-06-10:
> ```bash
> $ cat /proc/driver/nvidia/gpus/0000:01:00.0/power
> Runtime D3 status:          Disabled by default
> Video Memory:               Active
>
> GPU Hardware Support:
>  Video Memory Self Refresh: Not [...]

Ok, thanks. Same issue to me. It's a driver problem too. RTD3 is not working.

Rabcor, 2025-06-11:
Same on 5080 Max-Q on MSI Vector A18 HX A9WIG

[nvidia-bug-report.log.gz](https://github.com/user-attachments/files/20683499/nvidia-bug-report.log.gz)

The bug report is from the beta driver but I had the issue on 575.57.08 as well.+

To be honest I am completely shocked that this is happening on a supposedly stable driver, it is clear they didn't even test it on all available cards before releasing it as stable.

arbobendik, 2025-06-11:
> Apparently happens on 5090 as well [#872](https://github.com/NVIDIA/open-gpu-kernel-modules/issues/872)

Can confirm this happens with both 575.57.08 and 575.51.02 on my 5090 mobile. I also tested different kernel versions 6.14.x and the recent 6.15. Downgrading to the last 570.153.02 with open kernel modules consistently makes the driver pick up on the GPU across all configurations I've tested. RTD3 is consistently not working on any driver version either which is a huge loss on any mobile device.

Jubijub, 2025-06-12:
>  Downgrading to the last 570.153.02 with open kernel modules consistently makes the driver pick up on the GPU across all configurations I've tested. RTD3 is consistently not working on any driver version either which is a huge loss on any mobile device.

Same here. But it's a temporary fix, on Archlinux (as on any rolling distro), it won't be long until a kernel update or any other update breaks things

Binary-Eater, 2025-06-12:
@nekromantik does the RDT3 issue you are seeing only occur when you have `nvidia-drm` `modeset=1` (and implicitly `fbdev=1`)? If so, this is identical to #759.

nekromantik, 2025-06-12:
> [@nekromantik](https://github.com/nekromantik) does the RDT3 issue you are seeing only occur when you have `nvidia-drm` `modeset=1` (and implicitly `fbdev=1`)? If so, this is identical to [#759](https://github.com/NVIDIA/open-gpu-kernel-modules/issues/759).

I not tried it but the posters above talking about RDT3 not working. They can reply to your question.

Mario156090, 2025-06-12:
> [@nekromantik](https://github.com/nekromantik) does the RDT3 issue you are seeing only occur when you have `nvidia-drm` `modeset=1` (and implicitly `fbdev=1`)? If so, this is identical to [#759](https://github.com/NVIDIA/open-gpu-kernel-modules/issues/759).

Hello, no. This happen by default. You can check my thread on nvidia forums.

https://forums.developer.nvidia.com/t/new-portatil-2025-nvidia-rtx-5070-ti-mobile-rtd3-disable-by-default-or-not-supported-arch-linux-kernel-6-14-5-driver-version-570-144/332848

Mario156090, 2025-06-13:
> [@Mario156090](https://github.com/Mario156090) would you mind sharing the output from the following commands?
> 
> ```
> sudo cat /sys/module/nvidia_drm/parameters/fbdev 
> sudo cat /sys/module/nvidia_drm/parameters/modeset
> ```

Hello, OK, I moved to stable 570.153.02 version driver because how do you know with another recent versions the RTX5070 TI is not detected:

By default the output were:

[image]

Now I am going to change fbdev to 0 and reboot:

"Edit on a moment"

sk0rabu, 2025-06-16:
Can we plz get an ETA on the fix / next release? It's unacceptable a major driver release has been plaguing a whole product family for 2+ weeks without a clear way out in sight.

Same issue with RTD3 for me. I hope it's not sth Nvidia decided to just pull from the Linux drivers for good, like the ability to adjust TDP (anyone having their laptop GPU capped at 40W?)

mdx0111, 2025-06-17:
> Can someone check if this has been fixed on 575.64 ? In archlinux I have pushed these packages into the testing repostiory. On CachyOS they should follow soon into stable.

I don't suppose it is fixed as this issue is still open, no one from Nvidia has mentioned it is fixed and the changelog doesn't mention this bug. But, what do I know?! It's Nvidia!

Mario156090, 2025-06-17:
> I'm sorry this wasn't called out explicitly in the changelog, but the GSP initialization failures on some blackwell notebooks should be fixed in both of today's releases (570.169, 575.64).
> 
> I recommend tracking the RTD3 problem separately.

Thank's a lot. I will Open a New thread for RTD3 bug.

Mario156090, 2025-06-17:
I opened a bug for the RTD3 problem.

https://github.com/NVIDIA/open-gpu-kernel-modules/issues/882

Rabcor, 2025-06-18:
> Can someone check if this has been fixed on 575.64 ? In archlinux I have pushed these packages into the testing repostiory. On CachyOS they should follow soon into stable.

I am on cachy, and have tested it, confirmed it is working.

[image]

sk0rabu, 2025-06-18:
Can confirm the update is working and I got "Current Power Limit" and "Requested Power Limit" readings correct, unlike on previous versions.

ptr1337, 2025-06-22:
> > > [@aritger](https://github.com/aritger) For me the latest open driver still does not detect my 5070 ti
> > 
> > 
> > Which version?
> 
> 575.64 from cachyos normal repo not a branch

Please add a sudo nvidia-bug-report.sh

nekromantik, 2025-06-22:
> > > > [@aritger](https://github.com/aritger) For me the latest open driver still does not detect my 5070 ti
> > > 
> > > 
> > > Which version?
> > 
> > 
> > 575.64 from cachyos normal repo not a branch
> 
> Please add a sudo nvidia-bug-report.sh

attached 

[nvidia-bug-report.log.gz](https://github.com/user-attachments/files/20853972/nvidia-bug-report.log.gz)

nekromantik, 2025-06-23:
> > > > [@aritger](https://github.com/aritger) For me the latest open driver still does not detect my 5070 ti
> > > 
> > > 
> > > Which version?
> > 
> > 
> > 575.64 from cachyos normal repo not a branch
> 
> Please add a sudo nvidia-bug-report.sh

Did you have a look?

ptr1337, 2025-06-27:
> [@ptr1337](https://github.com/ptr1337) hi any updates?

Sorry, im not a NVIDIA maintainer. Ive just asked, so that the Maintainers do not ask you then again. :)

nekromantik, 2025-06-27:
@Mario156090 hi did you have a look at bug report I posted?
Newest version on cachy os still won't detect my 5070 ti

Mario156090, 2025-06-27:
> [@Mario156090](https://github.com/Mario156090) hi did you have a look at bug report I posted? Newest version on cachy os still won't detect my 5070 ti

Hello, I checked this but I can't found the issue. :/

can you test remove the catchy repo driver and install the drivers from nvidia all?
https://github.com/Frogging-Family/nvidia-all

For example, in the another threat speaking about RTD3, coarse mode does not work using Arch Linux official driver but compile from nvidia all works perfectly.

Mario156090, 2025-06-27:
> @Mario156090 ignore me I was being stupid I had set graphic mode to integrated only haha

Well, You have the power, close the threat as solved.

closed 2025-06-27 by nekromantik

YuanYuYuan, 2026-09-02:
This thread is closed, but 16 people reported the same GSP init failure in it, and the fix @aritger announced above **regressed again**. Posting once for anyone still subscribed.

**The fix landed in 570.169 and 575.64, then broke by 570.190.** I bisected it on a GB206M (RTX 5070 Laptop, `10de:2d58`):

| driver | result |
|---|---|
| **570.169** | **works** |
| **570.172.08** | **works** |
| 570.190 → 570.211.01 | fail |
| 575.64, 580.173.02, 595.84, 610.43.03, 610.57.04 | fail |

Ten versions, two working. So if you upgraded past 570.172.08 hoping for the fix, you moved onto the regression.

**Check your error code first — there are two different bugs in this thread.**

- **`RmInitAdapter failed! (0x62:0x1f:*)`** — the regression above. Try 570.169 or 570.172.08. Tracked at #1332, with the evidence that it is the GSP-RM firmware image rather than the driver.
- **`(0x62:0x40:2168)` on a 595.x driver** — probably a different regression. [bazzite#4494](https://github.com/ublue-os/bazzite/issues/4494) bisects that one to 595.45.04 good, 595.58.03 broken, and reproduces it on an **Ada** GPU. Rolling back to 595.45.04 is the fix there, not 570.169.

**One thing that costs people days: read `dmesg` from the top, not the tail.** `unexpected WPR2 already up` dominates every log here and is quoted in most of the reports above. On my machine it is an artifact of a retry. The first attempt fails earlier with a different code, and that is the real failure. `NVreg_RegistryDwords=RmGspBootRetryAttempts=1` removes the retries and shows it.

Full write-up: https://gist.github.com/YuanYuYuan/1b3c7290a3dbbf3029a326e1ad7ff644

Different laptops here have different dies, so a downgrade may not help everyone. It is cheap to test.