NVIDIA/open-gpu-kernel-modules #879: Driver 575: second external display missing and driver errors

arroyo-pl, 2025-06-09:
### NVIDIA Open GPU Kernel Modules Version

575.57.08

### Please confirm this issue does not happen with the proprietary driver (of the same version). This issue tracker is only for bugs specific to the open kernel driver.

- [ ] I confirm that this does not happen with the proprietary driver package.

### Operating System and Version

Archlinux

### Kernel Release

Linux 6.14.9-arch1-1 #1 SMP PREEMPT_DYNAMIC

### Please confirm you are running a stable release kernel (e.g. not a -rc). We do not accept bug reports for unreleased kernels.

- [x] I am running on a stable kernel release.

### Hardware: GPU

GPU 0: NVIDIA GeForce RTX 4070 Laptop GPU (UUID: GPU-85fade4b-34aa-55e3-2fe2-caacc7060d25)

### Describe the bug

After upgrading to version 575.57.08, only one (out of two) external display is detected.
Additionally, there are errors related to the NVIDIA driver logged in the system.
After reverting to version 570.153.02, both external displays work as expected, and there are no nvidia_drm errors in the dmesg output.
Attached logs for both 575 and 570 for comparison.

System:
Dell G16 7630 (Laptop)
GeForce RTX 4070
Archlinux

The displays are connected via HDMI ports provided by a USB-C hub.
The integrated Intel GPU is disabled in the BIOS.

Error found for driver 575:
```
[   48.520393] [drm:nv_drm_event_callback [nvidia_drm]] *ERROR* [nvidia-drm] [GPU ID 0x00000100] Failed to add encoder for NvKmsKapiDisplay 0x00000001
...
[   48.536896] Call Trace:
[   48.536896]  <TASK>
[   48.536898]  ? up+0x54/0xa0
[   48.536902]  ? __pfx__main_loop+0x10/0x10 [nvidia_modeset b6a132ebbd8d22d47c4eca2f95727419b05218ff]
[   48.536927]  nv_drm_event_callback+0x83/0x90 [nvidia_drm 467bc9be7922a385de83130999189aa615bf84d5]
[   48.536930]  nvKmsKapiHandleEventQueueChange+0xa7/0xe0 [nvidia_modeset b6a132ebbd8d22d47c4eca2f95727419b05218ff]
[   48.536947]  _main_loop+0x90/0x150 [nvidia_modeset b6a132ebbd8d22d47c4eca2f95727419b05218ff]
[   48.536958]  ? __pfx__main_loop+0x10/0x10 [nvidia_modeset b6a132ebbd8d22d47c4eca2f95727419b05218ff]
[   48.536968]  kthread+0xec/0x230
[   48.536971]  ? __pfx_kthread+0x10/0x10
[   48.536973]  ret_from_fork+0x31/0x50
[   48.536975]  ? __pfx_kthread+0x10/0x10
[   48.536977]  ret_from_fork_asm+0x1a/0x30
[   48.536979]  </TASK>
```

### To Reproduce

Just restart the system.

### Bug Incidence

Always

### nvidia-bug-report.log.gz

[575-nvidia-bug-report.log.gz](https://github.com/user-attachments/files/20653424/575-nvidia-bug-report.log.gz) - current version, only one display works, errors in dmesg

[570-nvidia-bug-report.log.gz](https://github.com/user-attachments/files/20653423/570-nvidia-bug-report.log.gz) - prev. version, two displays work, no error

arroyo-pl, 2025-06-20:
The problem persists in the new version:
nvidia-open 575.64
Linux borealis 6.15.2-arch1-1 

```
[15:45:08] [drm:nv_drm_load [nvidia_drm]] *ERROR* [nvidia-drm] [GPU ID 0x00000100] Failed to add connector for NvKmsKapiDisplay 0x00000100
[15:45:18] [drm:nv_drm_load [nvidia_drm]] *ERROR* [nvidia-drm] [GPU ID 0x00000100] Failed to get dynamic displays
[15:45:22] [drm:nv_drm_atomic_commit [nvidia_drm]] *ERROR* [nvidia-drm] [GPU ID 0x00000100] Flip event timeout on head 0
[15:45:28] [drm:nv_drm_event_callback [nvidia_drm]] *ERROR* [nvidia-drm] [GPU ID 0x00000100] Failed to add encoder for NvKmsKapiDisplay 0x00000001
[15:45:28] WARNING: CPU: 11 PID: 1217 at nvidia-drm/nvidia-drm-crtc.h:335 __nv_drm_handle_flip_event+0x1ad/0x1c0 [nvidia_drm]

```

kuch3n, 2025-07-19:
Got this error on proprietary drivers, too.
```
yay -Q nvidia linux
nvidia 575.64.03-4
linux 6.15.7.arch1-1
```

```
[kuch3n@TuxStation ~]$ sudo dmesg | grep nvidia
[    0.000000] Command line: initrd=\eb9b8373b9224ba4a3dc8ae31652ef10\6.15.7-arch1-1\initrd nvme_load=YES nowatchdog rw root=UUID=012070d0-81be-48da-9a59-d371e69efa3a resume=UUID=b7ac4cea-5d65-441a-95de-36b81ea55487 nvidia_drm.modeset=1 nvidia_drm.fbdev=1 rw root=UUID=012070d0-81be-48da-9a59-d371e69efa3a resume=UUID=b7ac4cea-5d65-441a-95de-36b81ea55487 nvidia_drm.modeset=1 nvidia_drm.fbdev=1 systemd.machine_id=eb9b8373b9224ba4a3dc8ae31652ef10
[    0.036368] Kernel command line: initrd=\eb9b8373b9224ba4a3dc8ae31652ef10\6.15.7-arch1-1\initrd nvme_load=YES nowatchdog rw root=UUID=012070d0-81be-48da-9a59-d371e69efa3a resume=UUID=b7ac4cea-5d65-441a-95de-36b81ea55487 nvidia_drm.modeset=1 nvidia_drm.fbdev=1 rw root=UUID=012070d0-81be-48da-9a59-d371e69efa3a resume=UUID=b7ac4cea-5d65-441a-95de-36b81ea55487 nvidia_drm.modeset=1 nvidia_drm.fbdev=1 systemd.machine_id=eb9b8373b9224ba4a3dc8ae31652ef10
[    3.239383] nvidia: loading out-of-tree module taints kernel.
[    3.239389] nvidia: module license 'NVIDIA' taints kernel.
[    3.239393] nvidia: module verification failed: signature and/or required key missing - tainting kernel
[    3.239395] nvidia: module license taints kernel.
[    3.693231] nvidia-nvlink: Nvlink Core is being initialized, major device number 240
[    3.696605] nvidia 0000:10:00.0: vgaarb: VGA decodes changed: olddecodes=io+mem,decodes=none:owns=none
[    3.922141] nvidia_uvm: module uses symbols nvUvmInterfaceDisableAccessCntr from proprietary module nvidia, inheriting taint.
[    3.934957] nvidia-modeset: Loading NVIDIA Kernel Mode Setting Driver for UNIX platforms  575.64.03  Wed Jun 25 18:07:15 UTC 2025
[    4.018948] [drm] [nvidia-drm] [GPU ID 0x00001000] Loading driver
[   14.813681] [drm:nv_drm_load [nvidia_drm]] *ERROR* [nvidia-drm] [GPU ID 0x00001000] Failed to add connector for NvKmsKapiDisplay 0x00000400
[   24.901517] [drm:nv_drm_load [nvidia_drm]] *ERROR* [nvidia-drm] [GPU ID 0x00001000] Failed to get dynamic displays
[   24.907246] [drm] Initialized nvidia-drm 0.0.0 for 0000:10:00.0 on minor 0
[   24.907449] nvidia 0000:10:00.0: vgaarb: deactivate vga console
[   24.912805] nvidia 0000:10:00.0: [drm] Cannot find any crtc or sizes
[   34.997518] [drm:nv_drm_event_callback [nvidia_drm]] *ERROR* [nvidia-drm] [GPU ID 0x00001000] Failed to add encoder for NvKmsKapiDisplay 0x00000001
[   35.767934] fbcon: nvidia-drmdrmfb (fb0) is primary device
[   35.767989] nvidia 0000:10:00.0: [drm] fb0: nvidia-drmdrmfb frame buffer device
```

```lspci | grep VGA
10:00.0 VGA compatible controller: NVIDIA Corporation GP106 [GeForce GTX 1060 6GB] (rev a1)
30:00.0 VGA compatible controller: Advanced Micro Devices, Inc. [AMD/ATI] Cezanne [Radeon Vega Series / Radeon Vega Mobile Series] (rev c9)
```

Plasma/X11 works with two monitors on daisy chain, while wayland only works with one of them (last one). 
On boot up the screen stays black for around half a minute, then shows the login screen on both monitors.

pauloconstanca, 2025-08-09:
Hello. I've been affected by this bug as well since before 575 release (late 570 started it).

I use 3 external monitors (2 HDMI 1080p@75hz) trough a DOCK connected in USB-C and 1 Display Port (DP 1440p @ 180hz) trough a USB-C Adapter), and every time i boot my system with all of them connected the same monitor will not turn on and the driver will always spit "Failed to add connector for NvKmsKapiDisplay" in the system logs.

Also, every time, if i boot with the dock unpluged (but my DP Monitor plugged in), the system boots up fine, and after boot if i plug the dock all screens turn on just fine.

I can confirm it happened in late 570, 575 and now in 580 BETA as well.

Bellow logs and details:

### NVIDIA Open GPU Kernel Modules Version
580.65.06-1 [BETA DKMS]

    * [x]  I confirm that this does not happen with the proprietary driver package.

 ### Operating System and Version
 Archlinux

 ### Kernel Release
Linux 6.15.9-arch1-1
     * [x]  I am running on a stable kernel release.

### Hardware: GPU
System: Legion Pro 5 16ARX8 (AMD Ryzen 7945HX + 32GB RAM + Nvidia 4070 Laptop GPU
Note: CPU iGPU is disabled in BIOS

### Describe the bug
Plug in a USB-C Dock with 2 HDMI screens + one USB-C adapter with a Display Port screen with the system powered off. 
Boot the system
One of the HDMI displays (always the same one) will not get detected by the system.
Logs will show "Failed to add connector for NvKmsKapiDisplay" as follows:

`
[   14.763698] [drm:nv_drm_dev_load [nvidia_drm]] *ERROR* [nvidia-drm] [GPU ID 0x00000100] Failed to add connector for NvKmsKapiDisplay 0x00000200
[   23.124223] [UFW BLOCK] IN=enp5s0f3u2u2 OUT= MAC=01:00:5e:00:00:01:cc:19:a8:2c:27:3f:08:00 SRC=192.168.70.254 DST=224.0.0.1 LEN=32 TOS=0x00 PREC=0x00 TTL=1 ID=4255 DF PROTO=2 
[   23.180677] [UFW BLOCK] IN=wlan0 OUT= MAC=c0:35:32:67:49:c9:cc:19:a8:2c:27:3f:08:00 SRC=192.168.70.254 DST=224.0.0.1 LEN=32 TOS=0x00 PREC=0x00 TTL=1 ID=4255 DF PROTO=2 
[   24.851727] [drm:nv_drm_dev_load [nvidia_drm]] *ERROR* [nvidia-drm] [GPU ID 0x00000100] Failed to get dynamic displays
[   24.870303] [drm] Initialized nvidia-drm 0.0.0 for 0000:01:00.0 on minor 1
[   24.894455] Console: switching to colour dummy device 80x25
[   24.912015] nvidia 0000:01:00.0: vgaarb: deactivate vga console
[   24.984097] fbcon: nvidia-drmdrmfb (fb0) is primary device
[   26.191904] Console: switching to colour frame buffer device 160x45
[   29.235753] [drm:nv_drm_atomic_commit [nvidia_drm]] *ERROR* [nvidia-drm] [GPU ID 0x00000100] Flip event timeout on head 0
[   32.243677] [drm:nv_drm_atomic_commit [nvidia_drm]] *ERROR* [nvidia-drm] [GPU ID 0x00000100] Flip event timeout on head 1
[   32.245472] nvidia 0000:01:00.0: [drm] fb0: nvidia-drmdrmfb frame buffer device
[   33.436580] Bluetooth: RFCOMM TTY layer initialized
[   33.436586] Bluetooth: RFCOMM socket layer initialized
[   33.436588] Bluetooth: RFCOMM ver 1.11
[   34.939374] systemd-journald[526]: Time jumped backwards, rotating.
[   34.971377] [drm:nv_drm_event_callback [nvidia_drm]] *ERROR* [nvidia-drm] [GPU ID 0x00000100] Failed to add encoder for NvKmsKapiDisplay 0x00000001
[   35.060411] ------------[ cut here ]------------
[   35.060414] WARNING: CPU: 2 PID: 726 at nvidia-drm/nvidia-drm-crtc.h:335 __nv_drm_handle_flip_event+0x1ad/0x1c0 [nvidia_drm]
`
 
 ### To Reproduce
Cold start the system with 3 external monitors connected or, if already booted with them plugged in, restart the system.

### Bug Incidence
Always reproducible with the same bug manifesting the same way

### nvidia-bug-report.log.gz
[nvidia-bug-report.log.gz](https://github.com/user-attachments/files/21698142/nvidia-bug-report.log.gz)

### More Info
If i recall correctly, this issue started in late 570 point release builds.

arroyo-pl, 2025-09-14:
@jcubic 
Do you have the following line in your dmesg? I don't see it in the logs you provided.
`Failed to add connector for NvKmsKapiDisplay`

jcubic, 2025-09-14:
@arroyo-pl no, so it seems to be a different issue.

amrit1711, 2025-12-29:
Hi All,
Apologize for delayed response, we have a bug 5772758 internally filed for tracking purpose.
Team is currently trying to duplicate issue in lab, shall keep posted on the further updates.