NVIDIA/open-gpu-kernel-modules #880: RTX 5090 often freezes presentation for 3-5 seconds in certain games

matte-schwartz, 2025-06-13:
### NVIDIA Open GPU Kernel Modules Version

570.123.18

### Please confirm this issue does not happen with the proprietary driver (of the same version). This issue tracker is only for bugs specific to the open kernel driver.

- [x] I confirm that this does not happen with the proprietary driver package.

### Operating System and Version

CachyOS (Arch Linux)

### Kernel Release

Linux blackwell 6.12.33-1-cachyos-lts #1 SMP PREEMPT_DYNAMIC Tue, 10 Jun 2025 14:43:36 +0000 x86_64 GNU/Linux

### Please confirm you are running a stable release kernel (e.g. not a -rc). We do not accept bug reports for unreleased kernels.

- [x] I am running on a stable kernel release.

### Hardware: GPU

GPU 0: NVIDIA GeForce RTX 5090 (UUID: GPU-addf70ec-1a4e-f8db-39bc-59d1ab271798)

### Describe the bug

When playing certain titles on my RTX 5090 rig, I experience frequent "freezes" where the game window will freeze but some audio continues to play in the background. During the freezes, I am able to use everything else on my PC normally without any indications of a GPU hang. After 3-5 seconds, the game "unfreezes" and then catches up to where it should be. 

This happens with every compositor I have tried: KWin wayland, cosmic-comp, and gamescope. I have tried every 570 and 575 driver release and experience this issue on all of them. 

Games I experience this freezing with include:

* Clair Obscur: Expedition 33 (DX12)
* The First Berserker: Khazan (DX11 and DX12)
* Star Citizen (DX11)

### To Reproduce

KWin repro steps:
 
1. Launch Clair Obscur: Expedition 33 on an RTX 5090 (settings should not matter)
2. Play the game for 5-20 minutes
> At some random point in that timespan, you should start running into this freezing bug.

Gamescope repro steps:

1. Launch Steam inside of gamescope with `gamescope -e -f -h <height> -w <width> -r <refresh rate> --mangoapp --adaptive-sync -- steam -steamdeck -steamos3 -steampal -gamepadui`
2. Launch Clair Obscur: Expedition 33 (make sure the only launch option you are using is `SteamDeck=0 %command%`)
3. Play the game for 5-20 minutes
4. When you encounter a freeze, try pulling up one of the Steam side menus with the Xbox logo or Shift + Tab on your keyboard

> You should see a similar effect to the gameplay clip I posted below

### Bug Incidence

Always

### nvidia-bug-report.log.gz

[nvidia-bug-report.log.gz](https://github.com/user-attachments/files/20719823/nvidia-bug-report.log.gz)

[steam-1903340.log](https://github.com/user-attachments/files/20719830/steam-1903340.log)

### More Info

One interesting tidbit I found is that I am able to "unfreeze" the game if I pull up a separate DRM plane, like one of the Steam side menus when running Steam's GamepadUI inside of gamescope. An example of this is here:

https://github.com/user-attachments/assets/c406a8c4-8ea3-461c-8e5e-18e1cb467015

You can see at the end of the clip it gets stuck for a few seconds but the instant I pull up the side menu, it unfreezes.

matte-schwartz, 2025-06-14:
Apparently another way to unfreeze presentation is to pull up a different app window (like [ulauncher](https://github.com/Ulauncher/Ulauncher)) on top of the main game window if you're using a desktop environment to game. Haven't been able to reproduce this issue on my 4000 series laptop in dGPU mode yet, only on Blackwell.

Arulan, 2025-06-16:
I think I'm getting the same thing here with Kingdom Come: Deliverance II on a 5090.

https://forums.developer.nvidia.com/t/570-release-feedback-discussion/321956/554?u=arulan7106

mgruberb, 2025-06-17:
FYI, I had this exact issue on Clair Obscur on my 5080 when using Proton Experimental Bleeding Edge about 3 weeks ago. Switching back to the "stable" Proton 10.0-1 beta fixed it for me. Not sure about the exact causal relation, but maybe this makes it easier to find the trigger.

matte-schwartz, 2025-06-17:
that's very interesting, I hadn't considered that. I am running a 9800X3D on my NVIDIA rig with an AM5 motherboard (ASUS B850M-PLUS)

SimpleHeuristics, 2025-06-19:
So since Bazzite testing updated to 575.64 I enabled the smooth motion feature for Clair Obscur since that game doesn't have frame Gen support.

The game now doesn't do the random visual hang described here.

I haven't tested it with smooth motion off yet.

matte-schwartz, 2025-06-19:
Interesting find. 575.64 makes no difference here on its own, but I've yet to experience a freeze if I enable smooth motion. 

Maybe will help nvidia track down the issue?

Arulan, 2025-06-20:
Interesting. I wonder if some experimenting with DXVK_NVAPI logging/flag could yield something. I've tried to set DXVK_NVAPI_GPU_ARCH to Ada (AD100), but that didn't seem to do anything.

https://github.com/jp7677/dxvk-nvapi/blob/master/README.md

matte-schwartz, 2025-06-23:
None of the NVAPI settings seem to make a difference here. I'm fairly certain the reason NvPresent avoids the freeze is because of the second swapchain it creates alongside the primary game swapchain.

matte-schwartz, 2025-07-01:
got another hang in Clair Obscur: Expedition 33 after ~20 minutes of gameplay on the new `575.64.03` driver:

[nvidia-bug-report.log.gz](https://github.com/user-attachments/files/21008682/nvidia-bug-report.log.gz)

Did not have Smooth Motion enabled at the time.

vuryn, 2025-07-09:
I'm having similar issues with classic Mists of Pandaria. The game will randomly freeze for a few seconds (sometimes showing an older frame or just black) and then "catch up" to a current frame. Pulling up any kind of other window (alt-tab, start menu etc.) will often unfreeze the game.

I'm on Fedora 42 (kernel 6.15.4) with a 5090 FE (driver version 575.64.03). Logs at the time of the freeze below.

```
info:vkd3d-proton:vkd3d_pipeline_library_disk_thread_main: Pipeline cache marked dirty. Flush is scheduled.
info:vkd3d-proton:vkd3d_pipeline_library_disk_thread_main: Flushing disk cache (wakeup counter since last flush = 152). It seems like application has stopped creating new PSOs for the time being.
info:vkd3d-proton:vkd3d_pipeline_library_disk_thread_main: Pipeline cache marked dirty. Flush is scheduled.
info:vkd3d-proton:vkd3d_pipeline_library_disk_thread_main: Flushing disk cache (wakeup counter since last flush = 81). It seems like application has stopped creating new PSOs for the time being.
info:vkd3d-proton:vkd3d_pipeline_library_disk_thread_main: Pipeline cache marked dirty. Flush is scheduled.
info:vkd3d-proton:vkd3d_pipeline_library_disk_thread_main: Flushing disk cache (wakeup counter since last flush = 2). It seems like application has stopped creating new PSOs for the time being.
err:vkd3d-proton:d3d12_command_allocator_Reset: There are still 1 pending command lists awaiting execution from command allocator iface 00000000887dbbd0!
err:vkd3d-proton:d3d12_command_allocator_Reset: There are still 1 pending command lists awaiting execution from command allocator iface 00000000887dd9d0!
err:vkd3d-proton:d3d12_command_allocator_Reset: There are still 1 pending command lists awaiting execution from command allocator iface 0000000088af6f20!
err:vkd3d-proton:d3d12_command_allocator_Reset: There are still 1 pending command lists awaiting execution from command allocator iface 0000000088af7c40!
err:vkd3d-proton:d3d12_command_allocator_Reset: There are still 1 pending command lists awaiting execution from command allocator iface 0000000088af8d20!
err:vkd3d-proton:d3d12_command_allocator_Reset: There are still 1 pending command lists awaiting execution from command allocator iface 0000000088af9680!
err:vkd3d-proton:d3d12_command_allocator_Reset: There are still 1 pending command lists awaiting execution from command allocator iface 0000000088af9c20!
err:vkd3d-proton:d3d12_command_allocator_Reset: There are still 1 pending command lists awaiting execution from command allocator iface 0000000096528a20!
err:vkd3d-proton:d3d12_command_allocator_Reset: There are still 1 pending command lists awaiting execution from command allocator iface 00000000887dbf90!
info:vkd3d-proton:vkd3d_get_vk_version: vkd3d-proton - applicationVersion: 2.14.0.
info:vkd3d-proton:vkd3d_instance_init: vkd3d-proton - build: ...
info:vkd3d-proton:vkd3d_memory_info_decide_hvv_usage: Topology: No more than 1 device local heap, HVV access is viable.
info:vkd3d-proton:vkd3d_memory_info_upload_hvv_memory_properties: Topology: HVV usage is allowed, using DEVICE_LOCAL | HOST_COHERENT for UPLOAD.
info:vkd3d-proton:vkd3d_memory_info_init_budgets: Applying resizable BAR budget to memory types: 0x10.
info:vkd3d-proton:vkd3d_bindless_state_get_bindless_flags: Enabling fast paths for advanced ExecuteIndirect() graphics and compute (EXT_dgc).
info:vkd3d-proton:vkd3d_bindless_state_get_bindless_flags: Device supports VK_EXT_mutable_descriptor_type.
info:vkd3d-proton:vkd3d_bindless_state_add_binding: Device supports VK_EXT_descriptor_buffer!
info:vkd3d-proton:vkd3d_bindless_state_add_binding: Device supports VK_EXT_descriptor_buffer!
info:vkd3d-proton:vkd3d_bindless_state_add_binding: Device supports VK_EXT_descriptor_buffer!
info:vkd3d-proton:d3d12_device_caps_init_shader_model: Enabling support for SM 6.6.
info:vkd3d-proton:d3d12_device_caps_init_shader_model: Enabling support for SM 6.7.
info:vkd3d-proton:d3d12_device_caps_init_shader_model: Enabling support for SM 6.8.
fixme:vkd3d-proton:d3d12_device_caps_init_feature_options1: TotalLaneCount = 16320, may be inaccurate.
info:vkd3d-proton:d3d12_device_determine_ray_tracing_tier: DXR support enabled.
info:vkd3d-proton:d3d12_device_determine_ray_tracing_tier: DXR 1.1 support enabled.
info:vkd3d-proton:d3d12_device_caps_init_feature_level: DX Ultimate supported!
info:vkd3d-proton:vkd3d_pipeline_library_init_disk_cache: Remapping VKD3D_SHADER_CACHE to: vkd3d-proton.cache.
info:vkd3d-proton:vkd3d_pipeline_library_init_disk_cache: Attempting to load disk cache from: vkd3d-proton.cache.
info:vkd3d-proton:vkd3d_pipeline_library_disk_thread_main: Performing async setup of stream archive ...
info:vkd3d-proton:vkd3d_pipeline_library_disk_cache_merge: Merging disk caches.
info:vkd3d-proton:vkd3d_pipeline_library_disk_cache_merge: Done merging shader caches, existing entries: 1147, new entries: 76.
info:vkd3d-proton:vkd3d_pipeline_library_disk_cache_merge: Successfully replaced shader cache with merged cache.
info:vkd3d-proton:vkd3d_pipeline_library_disk_cache_initial_setup: Merging pipeline libraries took 2.288 ms.
info:vkd3d-proton:vkd3d_pipeline_library_disk_cache_initial_setup: Mapping read-only cache took 0.054 ms.
info:vkd3d-proton:vkd3d_pipeline_library_disk_cache_initial_setup: Parsing stream archive took 0.262 ms.
info:vkd3d-proton:vkd3d_pipeline_library_disk_thread_main: Done performing async setup of stream archive.
[... cut ...]
info:vkd3d-proton:dxgi_vk_swap_chain_init_sync_objects: Ensure maximum latency of 3 frames with KHR_present_wait.
info:vkd3d-proton:dxgi_vk_swap_chain_init_waiter_thread: Enabling present wait path for frame latency.
info:vkd3d-proton:dxgi_vk_swap_chain_init_sleep_state: Timer interval is 1.0 ms.
info:vkd3d-proton:dxgi_vk_swap_chain_recreate_swapchain_in_present_task: Got 4 swapchain images.
info:vkd3d-proton:dxgi_vk_swap_chain_recreate_swapchain_in_present_task: Got 4 swapchain images.
info:vkd3d-proton:dxgi_vk_swap_chain_recreate_swapchain_in_present_task: Got 4 swapchain images.
```

matte-schwartz, 2025-07-11:
depending on your setup moving the mouse may be enough to "wake" it up per se, similar to how pulling up an overlay unfreezes it for me.

admchatm, 2025-08-05:
I have been trying to troubleshoot this problem for a few weeks now. I was getting the same issues, FPS would drop to nothing and would unfreeze after a few moments or bring another application into focus. Trying changes listed in this thread didn't yield much. 

So next i decided to turn off any overclocks. starting with DOCP/XMP. After disabling overclock on my memory, a game that would do it a few times per hour hasn't done it once in three hours. I am going to continue testing but my case seems to have been stemming from an unstable overclock on the memory controller. (Also running on an 9800X3D)

matte-schwartz, 2025-08-05:
That was one of the first things I tried before reporting the issue, unfortunately did not seem to make a difference here. I will check again though.

admchatm, 2025-08-09:
Hi Matte. Was curious if you saw any success. After several days I haven't had a single drop to performance to what we were seeing. I did find something else interesting though, that could also be worth investigation. While I was having issues, lsfg-vk was running in the background latching onto GameThread, while troubleshooting I disabled that background process which might have also had something to do with it. When I re-enabled it I got drops to 0 but they only lasted for a split second.

cjbottaro, 2025-08-30:
I'm so glad I found this issue and that I'm not the only one, I was going crazing. I'm experiencing the exact same issue; game freezes, and alt-tabbing (focusing another window) unfreezes it. During the freeze, I still get sound.

I also have a 9800X3D and RTX 5090.

I have also tried multiple RAM vendors (G.Skill and Crucial) and also tried lowering down to JEDEC specs.

Forcing PCIe to gen4 seems to lessen the frequency for me, but it still happens.

Please let me know what other info I can provide.

grumd, 2025-08-31:
Same issue here.

- RTX 5080
- Ryzen 9800X3D
- Linux 6.15.9-116.bazzite.fc42.x86_64
- KDE Plasma 6.4.4 (Wayland)
- Drivers 580.76.05, also was happening with 575.

- Clair Obscur Expedition 33 - very frequent freezes
- Genshin Impact - very rare freezes

Every time instantly unfreezes if I alt-tab. Unfreezes on its own if I wait maybe 10 seconds. Sound doesn't freeze.

matte-schwartz, 2025-09-02:
tested the new `580.82.07` driver, and still experienced a freeze in Star Citizen (DX11): [nvidia-bug-report.log.gz](https://github.com/user-attachments/files/22103608/nvidia-bug-report.log.gz)

admchatm, 2025-09-02:
So, I thought it stopped for me but it recently came back with a vengeance. game would freeze for a few seconds but I can rescue it sooner by bringing something else into focus. Only difference between when I wasn't having the problem and when it came back was when I connected a new bluetooth controller. I had also switched to ethernet from wifi back when I initially had the issue so I don't know if that also is something. So I stopped my WiFi and Bluetooth driver and the problem was gone.

I then tried a bluetooth headset and began noticing stuff like videos started hitching until I again stopped the bluetooth driver. I am using on onboard bluetooth adapter as opposed to a an external one.

Looking up the issue I found that Jay in this video recently experienced something similar under windows (not exactly the same but it did have the sudden FPS to 0 for a few seconds) Symptom. Maybe its the same issue with a few differences under linux? Try going into the BIOS and stopping the the wifi and bluetooth controller and see if that has an impact? Or maybe a driver utility that can monitor connections but I dont know anything under Linux.
https://www.youtube.com/watch?v=YWTZkB9rVU0

cjbottaro, 2025-09-03:
I've had my wifi and bluetooth disabled in bios from the beginning. Doesn't do anything to help.

cjbottaro, 2025-09-05:
I tested NVK and for Helldivers 2, I played for over an hour with no freezes at all. But with KCD2, it froze within a minute and alt-tabbing didn't help at all. I had to quit the game via Steam.

Not sure if this a useful datapoint, but dual booting into Windows and using the latest Nvidia Game Ready drivers, everything works perfectly fine (i.e. the card/system is not defective).

EugeneSlepov, 2025-09-05:
> I tested NVK and for Helldivers 2, I played for over an hour with no freezes at all. But with KCD2, it froze within a minute and alt-tabbing didn't help at all. I had to quit the game via Steam. [...]

This is interesting, in KCD2 I observed 2 types of freezes with 580.76.05. Recoverable, i.e. the ones where Alt-Tab helps and unrecoverable, i.e. the one you describe. When unrecoverable ones happened dmesg showed XID 109 error, recoverable ones didn't yield any messages in dmesg.

adoseofreel, 2025-09-18:
Having the same issue on games like Assetto Corsa and FiveM. I did find that there was at one point event viewer sending errors for a chromium based file called libcef.dll. I once replaced the file in Assetto Corsa, next game, no freezing, after time though, the freezes came back so i don't think that's the issue. But could lead us to something that may be remaining in the games were playing? Worth a shot. 9800X3D, 5090 FE, 96GB DDR5. Now i will say as well, my system is running at x8 as I'm running 3 NVME's.

grumd, 2025-11-02:
> Anyone have any workarounds while real fix is being developed? It's a real bummer having to boot into Windows... :/

I've been able to play Clair Obscur for multiple hours without a freeze by running the game through Lutris, couldn't play it in Steam due to constant freezes just like described here (instantly unfreezes on alt+tab or win button press)

DoomPenguin9, 2025-11-07:
I get this issue with VKD3D games during long play sessions (2+ hours). The game will freeze for 5-10 seconds, audio will continue playing, and everything other than the game is fully responsive. Alt+Tab doesn't fix it, I just have to wait it out. After it recovers, I can continue to play the game normally for hours.

Nothing like this has happened in VULKAN games like DOOM The Dark Ages do not exhibit this behaviour after 4+ hours of gameplay.

matte-schwartz, 2025-11-07:
I noticed something similar with Star Citizen. The DX11 backend freezes as described in this issue but the Vulkan backend has been flawless.

cjbottaro, 2025-11-16:
> Not sure VRR is the culprit.

Yeah. I just tried after turning off Gnome's VRR in display settings. The freeze happened within 5 mins of playing then once again later in about a 40 min playing session.

EugeneSlepov, 2025-12-27:
> > I'm afraid they haven't even acknowledged the problem, let alone started working on a fix.
> 
> So I'm guessing 590.48.01 didn't fix the issue then... 😩

It did not, I tested it the day it was released.

yokelpole, 2025-12-31:
I've been able to trigger this pretty frequently (1-2 minutes between freezes) with the following specs:

- CachyOS w/ Niri window manager
- Linux 6.18.2-2-cachyos
- Intel(R) Core(TM) Ultra 9 275HX (24) @ 5.40 GHz
- NVIDIA GeForce RTX 5090 Max-Q / Mobile [Discrete]
- Driver version 590.48.01

The game that seems to trigger it most frequently for me is Pizza Possum, which definitely isn't a super demanding game. It also frequently happens in other games such as Sonic Racing: Crossworlds.

**However, I have more to add than just another "this happens to me too!" report.** The freezes seem to fully subside when I do not use the dedicated fullscreen mode in niri. If I make the window 100% width and disable my waybar so that the window takes up all of the screen then I don't get any freezes. So while it's not optimal to have the window border visible, it seems like the issue may be related to dedicated full screen in one way or another?

yokelpole, 2025-12-31:
**Potential workaround:** I use Niri as a window manager, and I find that if I just set the window for games like Pizza Possum to full width rather than using the fullscreen mode that the freezes seem to completely disappear.

Maybe the problem is related to dedicated fullscreen in one way or another?

cjbottaro, 2026-01-16:
@EugeneSlepov I don't think @amrit1711 works for Nvidia... or anyone.

emichael, 2026-01-18:
Just to add to the list of people experiencing the exact same symptoms with Intel CPUs: I'm running an i5-12500 with an RTX 5080. I'm on Ubuntu 24.04 with the `nvidia-driver-580-open` package.

Interestingly, I'm seeing the same intermittent freezes in Chrome during video playback. They don't tend to last as long as the multi-second pauses during games, but videos will freeze for a half second while audio playback is unaffected.

cjbottaro, 2026-01-22:
> > [@EugeneSlepov](https://github.com/EugeneSlepov) I don't think [@amrit1711](https://github.com/amrit1711) works for Nvidia... or anyone.
> 
> I am pretty sure he does: https://forums.developer.nvidia.com/t/590-release-feedback-discussion/353310/202 I am wondering if Nvidia supports open source kernel modules at all at this point.

Ha, fair... but it's a very sus GitHub user... https://github.com/amrit1711

:D

Hyrules, 2026-02-17:
I've been having the same issue with Satisfactory on DX12. I'm on a 4090 though. Drivers 590.48.01-4. arch plasma kde Endeavour OS. After playing a while with the game the presentation get stuck for a few seconds at certain intervals. In my case it varies from 3 - 30 seconds or more of hang time. The audio keeps playing meanwhile. To restore the presentation faster, my workaround is to press the (winkey) menu so the main menu open in plasma and then return to the game. It generally resume after a few seconds. Otherwise running Satisfactory with the vulkan engine (use the -vukan flag at launch) removes the hang completely.<

> OS: EndeavourOS Linux x86_64 
> Host: MS-7E85 1.0 
> Kernel: 6.18.9-arch1-2 
> Uptime: 2 hours, 45 mins 
> Packages: 1742 (pacman) 
> Shell: bash 5.3.9 
> Resolution: 3840x2160 
> DE: Plasma 6.5.5 
> WM: kwin 
> Theme: Breeze-Dark [GTK2], Breeze [GTK3] 
> Icons: breeze-dark [GTK2/3] 
> Terminal: konsole 
> CPU: AMD Ryzen 7 9800X3D (16) @ 5.271GHz 
> GPU: NVIDIA GeForce RTX 4090 
> Memory: 9711MiB / 63669MiB 
 
Nvidia info : 

> 2026-02-17 09:44:55: Info: Currently installed packages related to Nvidia:
> 2026-02-17 09:44:55:       egl-gbm 1.1.3-1
> 2026-02-17 09:44:55:       egl-wayland 4:1.1.21-1
> 2026-02-17 09:44:55:       egl-wayland2 1.0.0.rc.r57.g1893c37-1
> 2026-02-17 09:44:55:       egl-x11 1.0.4-1
> 2026-02-17 09:44:55:       ffnvcodec-headers 13.0.19.0-1
> 2026-02-17 09:44:55:       lib32-nvidia-utils 590.48.01-1
> 2026-02-17 09:44:55:       libvdpau 1.5-4
> 2026-02-17 09:44:55:       libxnvctrl 590.48.01-1
> 2026-02-17 09:44:55:       linux-firmware-nvidia 20260110-1
> 2026-02-17 09:44:55:       nvidia-hook 1.5.3-1
> 2026-02-17 09:44:55:       nvidia-inst 26.1.1-1
> 2026-02-17 09:44:55:       nvidia-open-dkms 590.48.01-4
> 2026-02-17 09:44:55:       nvidia-settings 590.48.01-1
> 2026-02-17 09:44:55:       nvidia-utils 590.48.01-4

Immow, 2026-02-22:
I am experiencing the same.

System Information:

    GPU: NVIDIA GeForce RTX 5070 (Discrete)
    Driver Version: 590.48.01 (Open Kernel Modules)
    Kernel: Linux 6.18.8-3-cachyos (x86_64)
    OS: CachyOS
    DE: KDE Plasma 6.6.0 (Wayland)
    CPU: AMD Ryzen 7 7800X3D

Affected Games (Unreal Engine 5):
    Satisfactory,  StarRupture.

Compatibility Layer: Proton Version: proton-cachyos-10.0-20260207 (Steam Linux Runtime build)

Current Launch Options / Workarounds tried:
I am currently using the following string to mitigate the issue, though the freezes still occur (with and without VKD3D_CONFIG=no_upload_hvv):
__GL_MaxFramesAllowed=1 __GL_SHADER_DISK_CACHE_SKIP_CLEANUP=1 __GL_SHADER_DISK_CACHE_SIZE=12000000000 PROTON_DLSS_UPGRADE=1 game-performance %command%

My friend has the exact issue running Bazzite with an older NVIDIA GPU, suggesting this is a regression across the 590.xx series driver affecting UE5's DX12-to-Vulkan translation.

Immow, 2026-02-23:
All I can say is that the following games run flawless:

- Borderlands 3 (slight stutters sometimes but not nearly as severe as the issue being discussed) -> UE4 game
- Helldivers 2
- OverWatch
- No Rest of the Wicked
- Mahjong Soul

Only been using Linux for a month but these are my current findings.

I fed AI my game list to check if there are any UE5 titles and I will be testing Once Human next.

cjbottaro, 2026-03-13:
Issue still persists with 595.45.04 playing RE:Requiem.

> It seems like Nvidia managed to reproduce the issue on their side

That's something, at least.

You know who doesn't have issues? My gf with a 3060 on the exact same Linux system (Bazzite). The first time I have ever splurged on a top-of-the-line card and it's a worse experience than a budget card. Sorry I'm ranting, but damn, it's so frustrating and demoralizing.

asheriif, 2026-05-03:
Folks please also let Nvidia know in their official release threads. At least they sometimes respond there. They were actually able to reproduce this issue we're having two months ago in the 590 release thread but we've had no updates since https://forums.developer.nvidia.com/t/590-release-feedback-discussion/353310/429

According to that comment, their internal tracking number for this is bug 5376205.

Probably best to post here in the new 595 thread though https://forums.developer.nvidia.com/t/595-release-feedback-discussion/362561. Make sure to include your driver version and all other useful information (and you can reference their bug number, the comment in the 590 thread I linked above, and even this github issue). It's pretty easy to create an account on the forums and post.

ZakButler, 2026-05-14:
I have also commented in https://forums.developer.nvidia.com/t/595-release-feedback-discussion/362561

If you're experiencing this issue I recommend you (politely!) post in that thread too. Squeaky wheel gets the grease.

admchatm, 2026-05-23:
Out of curiosity. if Direct Scanoutis WORKING, I shouldn't be seeing the Kwin "Compositing" message in games right? Does this indicate a problem with Direct Scanout?

notpentadactyl, 2026-08-17:
Operating System: CachyOS Linux
KDE Plasma Version: 6.7.4
KDE Frameworks Version: 6.28.0
Qt Version: 6.11.1
Kernel Version: 7.1.5-1-cachyos (64-bit)
GPU: NVIDIA GeForce RTX **5080** (**610.57.04**)

I can reproduce this issue in **Warframe** with both d3d11 and d3d12 and with both xwayland and wine-wayland.
DLSS on/off, graphic settings, vrr off/on etc seem to not matter.

**The game freezes in random intervals for a few seconds (music continues).**

**You can instantly unfreeze it by just quickly pressing the `Win` key twice (brings up the KDE Plasma Application launcher).**

FYI: **I have tested `nouveau` (`mesa-git`) with `nvk` and can't reproduce it there!**

jpleclerc, 2026-08-24:
> > Facing similar issues. The game/screen will freeze entirely, mangohud aswell. Pressing the win key or alt+tab will instantly unfreeze.
> > Fedora 44 KDE, up to date RTX 5070 Ti, Driver 610.57.04 GE-Proton11-5, forcing wayland with `PROTON_ENABLE_WAYLAND=1`
> > I have also experienced the freeze on earlier GE-Proton and nvidia driver versions. Running the game under xwayland does not trigger the issue.
> > I ran `nvidia-smi dmon` while playing. During the freeze you can see the sm and mem drop to 0 before instantly recovering as soon as the freeze stops(either by itself or by pressing win/alt+tab). Power consumtion also drops, but the reporting seems to be slightly delayed.
> > > gpu    pwr  gtemp  mtemp     sm    mem    enc    dec    jpg    ofa   mclk   pclk
> > > Idx      W      C      C      %      %      %      %      %      %    MHz    MHz
> > > 0    164     65      -     63     29      0      0      0      0  14001   2812
> > > 0    170     65      -     65     34      0      0      0      0  14001   2812
> > > 0    156     61      -      **0      0**      0      0      0      0  14001   2820
> > > 0     56     59      -      **0      0**      0      0      0      0  14001   2835
> > > 0     93     64      -     63     32      0      0      0      0  14001   2767
> > > 0    169     65      -     65     35      0      0      0      0  14001   2812
> 
> i might be saying some totally unrelated dumb stuff so please take it with a grain of salt. i think its somehow related to the inactivity detection of linux. os might be thinking that as it is not getting any input from keyboard or mouse the computer is ideal and undisturbed and might be going into standby mode or sleep mode. thats why the gpu just stops rendering and after it detect that oh no the computer is still on it resumes its tasks.
> 
> i set my computer to stay awake all the time and the issue it gone for me atleast cant said for others. just putting it out there. btw i was testing this with tekken 8 so it could be that it only solved the freezing issue for that particular game.

You might be onto something here. I noticed that the freeze doesn't happen when I have a twitch stream opened on my second monitor, which prevents sleep. I'll gather some data in my next gaming session to confirm that.

cjbottaro, 2026-08-24:
> might be going into standby mode or sleep mode

> I noticed that the freeze doesn't happen when I have a twitch stream opened on my second monitor, which prevents sleep.

I'm on Bazzite which automatically sets Caffeine to run when a game starts. It doesn't help at all. I get this microfreezes pretty often.

dsine-de, 2026-08-24:
I get those freezes in Hogwarts Legacy. Didn't try other UE5 games yet. Other games run fine.

The only suspicious log entries I find:
```
x86/split lock detection: #AC: HogwartsLegacy./9145 took a split_lock trap at address: 0x14be8ac77
```

Carrotze, 2026-08-25:
> You might be onto something here. I noticed that the freeze doesn't happen when I have a twitch stream opened on my second monitor, which prevents sleep. I'll gather some data in my next gaming session to confirm that.

I have the same problem in Hunt Showdown. Played for 3 hours yesterday and put a twitch stream on my second monitor. And it didn't freeze a single time. Always on settings didn't do anything.

ValorousVisage, 2026-08-28:
> [@amrit1711](https://github.com/amrit1711) Any update on what Nvidia is doing to address this? It's been more than a year.

It was acknowledged by amrits in the 595 release feedback & discussion thread on the Nvidia Developer forum back in may where they said "This is still under investigation from engineering end." and again 4 days ago in the 610 thread where someone had a similar (if not identical) issue, the same moderator said "I will highlight it to the engineering team for visibility and follow up to help drive progress."

Unfortunately, it doesn't sound like we're getting a fix soon. Unsure if the cause is that elusive, or if this just isn't a priority, or both. Probably both.

KartikKawalia, 2026-09-06:
ok new error just appeared. if we use the twitch stream opened on the second monitor method after around 4hrs i got this error and the game crashed. the game didnt froze at any moment when i was gaming but the game straight up crashed. 

LowLevelFatalError [File:G:\p4\eagle\v302_xx-build_pkg_win\ue\Engine\Source\Runtime\RenderCore\Private\RenderingThread.cpp] [Line: 1310] GameThread timed out waiting for RenderThread after 120.00 secs

i am putting here what the epic developer community forum had to say about it:

This crash looks like a GPU Timeout - the GPU likely entered an infinite loop or taken an extremely long time to complete a task.

The user seems to have suspended TDR (Timeout Detection and Recovery) that would normally reset the GPU in this case, leading to UE detecting and crashing due to the unresponsive Render Thread. Normal behavior would have been a GPU crash after 5 seconds, in line with the LogD3D12RHI Warning messages.

With these types of crashes, the callstack and crash dumps do not provide useful information.

Are you able to reproduce the crash yourself? If so, the following GPU debugging options might help you get more information to debug the crash.

-d3ddebug -rdgdebug -rdgimmediate -forcerhibypass -gpucrashdebugging -gpuvalidationThis talk from Unreal Fest 2023 goes into more details on GPU crash debugging. The GPU Breadcrumbs feature it discusses can help you narrow down what pass is causing the issue.

KartikKawalia, 2026-09-07:
i will be deleting my comment it was a proton version specific issue

KartikKawalia, 2026-09-11:
ok i think i have found a permanent solution to this issue. i turned off vfr variable frame rate (not variable refresh rate which is vrr) and the problem is solved. with vfr on the system tries to save resources and stops send frames to the display which causes the game to freeze that is why opening the twitch stream and chat on the side solved the problem. it keeps forces wayland to stay active all the time hence no freezes and stutters. i am testing this for like 1 hour and not a single freeze. letsgoooooooooooo. finally solved

alenart1160, 2026-09-11:
> ok i think i have found a permanent solution to this issue. i turned off vfr variable frame rate (not variable refresh rate which is vrr) and the problem is solved. with vfr on the system tries to [...]

Well... I don't have VRR turned on and I still have this freezes :/ but it's good that you have found solution!

KartikKawalia, 2026-09-11:
btw i used hyprmod to generate a config lua file. hyprmod named it hyprland-gui.lua and i made these changes in it and it solved the problem:

-- Generated by HyprMod

-- Settings
hl.config({
    misc = {
       
    },
    debug = {
        vfr = false, 
    }
})

-- Monitors
hl.monitor({
    output = "DP-3",
    disabled = false,
    mode = "2560x1440@180.00Hz",
    position = "0x0",
    scale = 1,
    vrr = 1, 
    cm = "srgb",
})

KartikKawalia, 2026-09-11:
mind u this solution works for hyprland. dont know how to turn off vfr for other window managers and desktop environments. turning off vfr can lead to 1-2% more cpu usage and gpu temp can increase a little. by 2-5 degrees

Alien3294, 2026-09-11:
> ok i think i have found a permanent solution to this issue. i turned off vfr variable frame rate (not variable refresh rate which is vrr) and the problem is solved. with vfr on the system tries to save resources and stops send frames to the display which causes the game to freeze that is why opening the twitch stream and chat on the side solved the problem. it keeps forcing wayland to stay active all the time hence no freezes and stutters. i am testing this for like 1 hour and not a single freeze. letsgoooooooooooo. finally solved

How to disable vfr in KDE?

KartikKawalia, 2026-09-11:
lol it seems like this also solved the lowlevelfatalerror game crash in some proton versions. basically if any unreal engine 5 game is running in a different workspace but your focus is in another workspace the game crashes with a lowlevelfatalerror. basically linux stop generating any frames because the game is not in focus because of which the gamethread keeps waiting for frame but the renderthread is not generating any frame and the game crashes. letsgooo 2 birds with 1 stone.

spathi85, 2026-09-12:
> How to disable vfr in KDE?

As far as I can see, there is no option to disable VFR in KDE :(

But at least it looks like I've found a workaround: 
Leaving the KDE System Monitor (History tab) open on my second screen seems to prevent the freezes. I have played for several hours over the last few days and haven't had a single issue. Some people here have already mentioned that having an open Twitch stream would help, and this is probably pointing in the same direction.

Setup for reference:
CPU: Ryzen 7 9800X3D
GPU: RTX 5070 Ti
Dual monitor: 2K@144Hz + 2K@60Hz
Nobara 44 (KDE)

KartikKawalia, 2026-09-12:
> As far as I can see, there is no option to disable VFR in KDE :(
>
> But at least it looks like I've found a workaround: Leaving the KDE System Monitor (History tab) open on my second screen seems to [...]

yeah u cant turn it off kde because it handles that internally unlike hyprland

KartikKawalia, 2026-09-13:
Ok just putting it here if anyone has the specific issue I was facing. If u running any other proton versions like cachyos, ge proton or other proton versions instead of the default one somes games might crash after a massive 2 minute freeze with vfr off so stick to proton experimental and the official proton releases

mhogenson, 2026-09-15:
I'm running Arch Linux with GNOME and I was able to get this to work by running my game with `NVPRESENT_ENABLE_SMOOTH_MOTION=1`.  `NVPRESENT_ENABLE_SMOOTH_MOTION=1 %command%` in Steam launch options.  It enables NVIDIA’s Smooth Motion Vulkan layer (VK_LAYER_NV_present), which takes over frame presentation from the game/DXVK/VKD3D-Proton and, by default, presents from a separate asynchronous compute queue instead of the game’s normal present path.

Gunther-Schulz, 2026-09-29:
Same GPU, same symptom class, with tracing that narrows where the time goes. Posting in case it helps.

**System:** RTX 5090 (VBIOS 98.02.2E.00.03, PCIe 5.0 x16), open kernel module 615.71.09, Linux 7.2.8 (CachyOS), Ryzen 9 9950X3D, GNOME/mutter 50.5 on Wayland (also seen on X11). Game: Marvel Rivals (UE5, D3D12) via Proton (proton-cachyos 11.0-20260703) + vkd3d-proton 3.0.1.

**Symptom:** the game stops producing frames for ~1–5.5 s, then continues. Most long ones land at 5.0–5.5 s. They happen in match and in the focused main menu. Rate is strongly per-launch: across 78 launches (83 h) some launches freeze every ~10–20 min for hours, others stay clean for hours (dispersion test p < 0.0001). Rarely the stall does not recover and the game dies with Xid 13 (`SKEDCHECK06_REGISTER_COUNT`, `SKEDCHECK16_CTA_THREAD_DIMENSION_ZERO`, class `cec0`) followed ~60 s later by UE's render-fence watchdog — reported separately as HansKristian-Work/vkd3d-proton#3305. The common freeze logs nothing in the kernel log (no NVRM, no Xid, no AER).

**What the tracing shows** (bpftrace on syscalls, ntsync, winevulkan entry/return, nvidia ioctls, NVIDIA IRQ counts; `nvidia-smi` sampled every 200 ms). Consistent over 25+ freezes, most recently a 5.04 s one on 2026-09-29:

1. The game submits its last frame normally. `vkQueueSubmit2`, `vkQueuePresentKHR` and `vkAcquireNextImageKHR` all **return** within ~0.2 ms of the last frame. No Vulkan call is in flight during the hold.
2. The presenting queue's submission that the D3D12 fence worker waits on (via `vkWaitForFences`) depends on nothing outstanding. Its semaphore chain walks back to a head that waits on no other queue and no swapchain image. Same result in all 24 freezes of one launch.
3. That fence then signals **~0.5–5.4 s later with no CPU input in between**. There are 0 submits/signals between the last frame and fence-ready. In the 5.43 s case the first new submit came 4.8 ms *after* the fence turned ready.
4. During the hold, GPU utilization reads 0–1%. The GPU stays in P0 with memory at full clock at the start and only downclocks ~2 s in. NVIDIA interrupts drop to ~9% of normal. No nvidia ioctl from any process spans the hold.
5. All game render/RHI threads are asleep waiting on that fence; nothing is spinning on the CPU side.
6. UE's own 5 s fence-wait timeout is not what ends it: the timeout expires, the thread re-waits, and the stall ends only when the fence completes (+48 ms and +428 ms after the timeout in two cases).

So from userspace it looks like **the GPU sits on already-submitted, dependency-free work and completes it seconds later**, with nothing on the CPU side holding it. That puts the hold inside the driver's closed userspace component, GSP firmware scheduling, or the GPU itself. We cannot see further: `NVreg_EnableGpuFirmwareLogs=1` needs `gsp_log_*.bin`, which is not distributed (discussion #490), and `_issueRpcAndWait` cannot be kprobed on this kernel.

**Ruled out by testing:** ntsync on/off, `DXVK_NVAPI_VKREFLEX` / `PROTON_VKD3D_LOWLATENCY`, Wayland vs X11, a custom V/F curve (undervolt off: still 5 s freezes), locked GPU/memory clocks (worse, not better), thermal/power slowdown flags (none active). It occurs on every kernel used since onset (7.2.5–7.2.8).

**Onset:** first recorded on 615.71.09 (installed 2026-09-14, first freeze logged the same day). Earlier episodes on older drivers are remembered but were not recorded.

Happy to run a specific capture or a debug build if that would help. The traces (per-freeze timelines, GPU state logs) are available on request.

<sub>Analysis and draft prepared with Claude (AI assistant); measurements are from my machine.</sub>

KartikKawalia, 2026-10-03:
I was recently reading through the proton debug log files and made some foundings. At times proton is falling back to mesa drivers for some reason and giving this error:

MESA-EGL: warning: pci id for fd 50: 10de:2d04, driver (null)
MESA-EGL: warning: egl: failed to create dri2 screen

Wine’s native Wayland driver (winewayland.drv) is attempting to instantiate EGL contexts via Mesa's generic loader rather than NVIDIA's proprietary EGL vendor library (10_nvidia.json). On NVIDIA GPUs, this driver mismatch prevents proper Wayland surface creation and leads to rendering freezes—especially during workspace switches or window state changes.

 __EGL_VENDOR_LIBRARY_FILENAMES=/usr/share/glvnd/egl_vendor.d/10_nvidia.json PROTON_LOG=1 VKD3D_CONFIG=descriptor_heap %command%

Used this launch option to force Nvidia EGL on the game at all times. Game is running somehow better and smoother than before. Inputs feel way snappier for some reason. Way better than before, and I have also not experienced any stutters in my 4 hour testing. But I am not sure that this is the final solution to this problem. I'm asking everyone who is reading his to please try this launch option with the proton log on. If the game freezes or stutters even after using this launch option, quit the game and read the log file for any suspicious entries like falling back to mesa or something like that. I think this might be the main culprit.

KartikKawalia, 2026-10-08:
I have been testing for like 3 days and I think I have finally fixed it.

[10 more replies]