NVIDIA/open-gpu-kernel-modules #881: Need kernel-6.16 support

jeamieofqidan, 2025-06-15:
### NVIDIA Open GPU Kernel Modules Version

575.57.08

### Operating System and Version

Fedora

### Kernel Release

6.16.0-0.rc1.250613g27605c8c0f69.21.fc43.x86_64

### Please confirm you are running a stable release kernel (e.g. not a -rc). We do not accept bug reports for unreleased kernels.

- [ ] I am running on a stable kernel release.

### Build Command

open kernel modules compilation broken under kernel-6.16

### Terminal output/Build Log

  CC [M]  nvidia/nv-platform.o
nvidia/nv-platform.c:261:5: 警告：‘nv_soc_register_irqs’先前没有原型 [-Wmissing-prototypes]
  261 | int nv_soc_register_irqs(nv_state_t *nv)
      |     ^~~~~~~~~~~~~~~~~~~~
nvidia/nv-platform.c:332:6: 警告：‘nv_soc_free_irqs’先前没有原型 [-Wmissing-prototypes]
  332 | void nv_soc_free_irqs(nv_state_t *nv)
      |      ^~~~~~~~~~~~~~~~
nvidia/nv-platform.c:469:23: 警告：‘nv_soc_device_reset’先前没有原型 [-Wmissing-prototypes]
  469 | NV_STATUS NV_API_CALL nv_soc_device_reset(nv_state_t *nv)
      |                       ^~~~~~~~~~~~~~~~~~~
nvidia/nv-platform.c:525:23: 警告：‘nv_soc_mipi_cal_reset’先前没有原型 [-Wmissing-prototypes]
  525 | NV_STATUS NV_API_CALL nv_soc_mipi_cal_reset(nv_state_t *nv)
      |                       ^~~~~~~~~~~~~~~~~~~~~
nvidia/nv-platform.c:1330:5: 警告：‘nv_platform_count_devices’先前没有原型 [-Wmissing-prototypes]
 1330 | int nv_platform_count_devices(void)
      |     ^~~~~~~~~~~~~~~~~~~~~~~~~
nvidia/nv-platform.c:1343:5: 警告：‘nv_platform_register_driver’先前没有原型 [-Wmissing-prototypes]
 1343 | int nv_platform_register_driver(void)
      |     ^~~~~~~~~~~~~~~~~~~~~~~~~~~
nvidia/nv-platform.c:1353:6: 警告：‘nv_platform_unregister_driver’先前没有原型 [-Wmissing-prototypes]
 1353 | void nv_platform_unregister_driver(void)
      |      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
nvidia/nv-platform.c:1362:26: 警告：‘nv_soc_fuse_register_read’先前没有原型 [-Wmissing-prototypes]
 1362 | unsigned int NV_API_CALL nv_soc_fuse_register_read (unsigned int addr)
      |                          ^~~~~~~~~~~~~~~~~~~~~~~~~
nvidia/nv-platform.c:1373:11: 警告：‘nv_get_valid_window_head_mask’先前没有原型 [-Wmissing-prototypes]
 1373 | NV_STATUS nv_get_valid_window_head_mask(nv_state_t *nv, NvU64 *window_head_mask)
      |           ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
nvidia/nv-platform.c:1402:23: 警告：‘nv_soc_i2c_hsp_semaphore_acquire’先前没有原型 [-Wmissing-prototypes]
 1402 | NV_STATUS NV_API_CALL nv_soc_i2c_hsp_semaphore_acquire(NvU32 ownerId, NvBool bAcquire, NvU64 timeout)
      |                       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
nvidia/nv-platform.c:1408:8: 警告：‘nv_get_hdcp_enabled’先前没有原型 [-Wmissing-prototypes]
 1408 | NvBool nv_get_hdcp_enabled(nv_state_t *nv)
      |        ^~~~~~~~~~~~~~~~~~~
  CC [M]  nvidia/nv-dsi-parse-panel-props.o
nvidia/nv-dsi-parse-panel-props.c:798:1: 警告：‘nv_dsi_is_panel_connected’先前没有原型 [-Wmissing-prototypes]
  798 | nv_dsi_is_panel_connected
      | ^~~~~~~~~~~~~~~~~~~~~~~~~
nvidia/nv-dsi-parse-panel-props.c:828:1: 警告：‘nv_dsi_parse_panel_props’先前没有原型 [-Wmissing-prototypes]
  828 | nv_dsi_parse_panel_props
      | ^~~~~~~~~~~~~~~~~~~~~~~~
nvidia/nv-dsi-parse-panel-props.c:852:1: 警告：‘nv_dsi_panel_enable’先前没有原型 [-Wmissing-prototypes]
  852 | nv_dsi_panel_enable
      | ^~~~~~~~~~~~~~~~~~~
nvidia/nv-dsi-parse-panel-props.c:894:1: 警告：‘nv_dsi_panel_reset’先前没有原型 [-Wmissing-prototypes]
  894 | nv_dsi_panel_reset
      | ^~~~~~~~~~~~~~~~~~
nvidia/nv-dsi-parse-panel-props.c:942:6: 警告：‘nv_dsi_panel_disable’先前没有原型 [-Wmissing-prototypes]
  942 | void nv_dsi_panel_disable
      |      ^~~~~~~~~~~~~~~~~~~~
nvidia/nv-dsi-parse-panel-props.c:980:6: 警告：‘nv_dsi_panel_cleanup’先前没有原型 [-Wmissing-prototypes]
  980 | void nv_dsi_panel_cleanup
      |      ^~~~~~~~~~~~~~~~~~~~
  CC [M]  nvidia/nv-bpmp.o
  CC [M]  nvidia/nv-gpio.o
nvidia/nv-gpio.c:42:23: 警告：‘nv_gpio_get_pin_state’先前没有原型 [-Wmissing-prototypes]
   42 | NV_STATUS NV_API_CALL nv_gpio_get_pin_state
      |                       ^~~~~~~~~~~~~~~~~~~~~
nvidia/nv-gpio.c:69:18: 警告：‘nv_gpio_set_pin_state’先前没有原型 [-Wmissing-prototypes]
   69 | void NV_API_CALL nv_gpio_set_pin_state
      |                  ^~~~~~~~~~~~~~~~~~~~~
nvidia/nv-gpio.c:83:23: 警告：‘nv_gpio_set_pin_direction’先前没有原型 [-Wmissing-prototypes]
   83 | NV_STATUS NV_API_CALL nv_gpio_set_pin_direction
      |                       ^~~~~~~~~~~~~~~~~~~~~~~~~
nvidia/nv-gpio.c:121:23: 警告：‘nv_gpio_get_pin_direction’先前没有原型 [-Wmissing-prototypes]
  121 | NV_STATUS NV_API_CALL nv_gpio_get_pin_direction
      |                       ^~~~~~~~~~~~~~~~~~~~~~~~~
nvidia/nv-gpio.c:148:23: 警告：‘nv_gpio_get_pin_number’先前没有原型 [-Wmissing-prototypes]
  148 | NV_STATUS NV_API_CALL nv_gpio_get_pin_number
      |                       ^~~~~~~~~~~~~~~~~~~~~~
nvidia/nv-gpio.c:191:20: 警告：‘nv_gpio_get_pin_interrupt_status’先前没有原型 [-Wmissing-prototypes]
  191 | NvBool NV_API_CALL nv_gpio_get_pin_interrupt_status
      |                    ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
nvidia/nv-gpio.c:226:23: 警告：‘nv_gpio_set_pin_interrupt’先前没有原型 [-Wmissing-prototypes]
  226 | NV_STATUS NV_API_CALL nv_gpio_set_pin_interrupt
      |                       ^~~~~~~~~~~~~~~~~~~~~~~~~
  CC [M]  nvidia/nv-backlight.o
nvidia/nv-backlight.c:31:23: 警告：‘nv_get_tegra_brightness_level’先前没有原型 [-Wmissing-prototypes]
   31 | NV_STATUS NV_API_CALL nv_get_tegra_brightness_level
      |                       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
nvidia/nv-backlight.c:56:23: 警告：‘nv_set_tegra_brightness_level’先前没有原型 [-Wmissing-prototypes]
   56 | NV_STATUS NV_API_CALL nv_set_tegra_brightness_level
      |                       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
  CC [M]  nvidia/nv-imp.o
  CC [M]  nvidia/nv-platform-pm.o
nvidia/nv-platform-pm.c:53:23: 警告：‘nv_soc_pm_unpowergate’先前没有原型 [-Wmissing-prototypes]
   53 | NV_STATUS NV_API_CALL nv_soc_pm_unpowergate(
      |                       ^~~~~~~~~~~~~~~~~~~~~
nvidia/nv-platform-pm.c:106:23: 警告：‘nv_soc_pm_powergate’先前没有原型 [-Wmissing-prototypes]
  106 | NV_STATUS NV_API_CALL nv_soc_pm_powergate(
      |                       ^~~~~~~~~~~~~~~~~~~
  CC [M]  nvidia/nv-ipc-soc.o
  CC [M]  nvidia/nv.o
  CC [M]  nvidia/nv-pci.o
  CC [M]  nvidia/nv-dmabuf.o
nvidia/nv-dmabuf.c: 在函数‘nv_dma_buf_map’中:
nvidia/nv-dmabuf.c:844:9: 错误：隐式声明函数‘dma_buf_attachment_is_dynamic’ [-Wimplicit-function-declaration]
  844 |         dma_buf_attachment_is_dynamic(attachment) &&
      |         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
make[5]: *** [/usr/src/kernels/6.16.0-0.rc1.250613g27605c8c0f69.21.fc43.x86_64/scripts/Makefile.build:287：nvidia/nv-dmabuf.o] 错误 1
make[4]: *** [/usr/src/kernels/6.16.0-0.rc1.250613g27605c8c0f69.21.fc43.x86_64/Makefile:2052：.] 错误 2
make[3]: *** [/usr/src/kernels/6.16.0-0.rc1.250613g27605c8c0f69.21.fc43.x86_64/Makefile:260：__sub-make] 错误 2
make[3]: 离开目录“/home/lamarujian/open-gpu-kernel-modules-main/kernel-open”
make[2]: *** [Makefile:260：__sub-make] 错误 2
make[2]: 离开目录“/usr/src/kernels/6.16.0-0.rc1.250613g27605c8c0f69.21.fc43.x86_64”
make[1]: *** [Makefile:140：modules] 错误 2
make[1]: 离开目录“/home/lamarujian/open-gpu-kernel-modules-main/kernel-open”
make: *** [Makefile:59：modules] 错误 2

gilvbp, 2025-06-16:
@jeamieofqidan on cachy kernel is working, you can find more info and patches here: https://github.com/CachyOS/CachyOS-PKGBUILDS/pull/686

luisalvarado, 2025-06-16:
Ubuntu 25.04 fails if I upgrade to 6.16. So it also fails there.

jeamieofqidan, 2025-06-16:
> [@jeamieofqidan](https://github.com/jeamieofqidan) on cachy kernel is working, you can find more info and patches here: [CachyOS/CachyOS-PKGBUILDS#686](https://github.com/CachyOS/CachyOS-PKGBUILDS/pull/686)

Thx.Refer to https://github.com/CachyOS/CachyOS-PKGBUILDS/pull/686/commits/8070900bed4727dab62b9322bfa6627ff26a353d, my problem is solved

jeamieofqidan, 2025-06-16:
> Ubuntu 25.04 fails if I upgrade to 6.16. So it also fails there.
refer to:Refer to https://github.com/CachyOS/CachyOS-PKGBUILDS/pull/686/commits/8070900bed4727dab62b9322bfa6627ff26a353d, my problem is solved

Fjodor42, 2025-06-16:
On my Ubuntu 24.10 and on my Ubuntu 25.04 systems, the CachyOS patches makes the code compile, and also lets me use my DE (Cinnamon) *in general*, but *some* graphical apps, such as Ghostty (GNOME Terminal seems to work fine), freak out with long delays, trouble updating the view, etc.

It would probably be wrong to report that to the CachyOS project, as my distribution is Ubuntu, as opposed to theirs, but something seems to be amiss...

ptr1337, 2025-06-17:
> On my Ubuntu 24.10 and on my Ubuntu 25.04 systems, the CachyOS patches makes the code compile, and also lets me use my DE (Cinnamon) _in general_, but _some_ graphical apps, such as Ghostty (GNOME [...]

This is more likely a 6.16 Kernel issue, since these issues are also present on AMD

davidebeatrici, 2025-06-19:
Undefining `NV_CAN_CALL_VMA_START_WRITE` should be enough:

https://github.com/NVIDIA/open-gpu-kernel-modules/blob/fade1f7b2056a637ea3b7d6d4e78cc680cf358dc/kernel-open/common/inc/nv-mm.h#L300-L331

ptr1337, 2025-06-19:
@davidebeatrici This is not related to 6.16. This discussion youve sent was related to 6.15, which has been fixed in the recent 570 and 575 release.
6.16 needs different changes, see the patches linked above.

davidebeatrici, 2025-06-19:
Sorry, I didn't realize another separate issue came up. Thank you for pointing that out.

Fjodor42, 2025-06-29:
> > > On my Ubuntu 24.10 and on my Ubuntu 25.04 systems, the CachyOS patches makes the code compile, and also lets me use my DE (Cinnamon) _in general_, but _some_ graphical apps, such as Ghostty (GNOME Terminal seems to work fine), freak out with long delays, trouble updating the view, etc.
> > > It would probably be wrong to report that to the CachyOS project, as my distribution is Ubuntu, as opposed to theirs, but something seems to be amiss...
> > 
> > 
> > This is more likely a 6.16 Kernel issue, since these issues are also present on AMD
> 
> Would it be possible to share links to any threads discussing this, [@ptr1337](https://github.com/ptr1337)?
> 
> I don't mean that dismissively at all, rather just to be able to follow what's going on on my own :-)

Incidentally, it would seem, at least to me, that whatever causes this has now been back-ported into 6.15.4...

ptr1337, 2025-07-01:
Issue has been fixed in 575.64.03.

Here can be closed

closed 2025-07-02 by jeamieofqidan