NVIDIA/open-gpu-kernel-modules #882: RTX 5000 series RTD3 does not work

Mario156090, 2025-06-17:
### NVIDIA Open GPU Kernel Modules Version

575.64

### Please confirm this issue does not happen with the proprietary driver (of the same version). This issue tracker is only for bugs specific to the open kernel driver.

- [x] I confirm that this does not happen with the proprietary driver package.

### Operating System and Version

Arch Linux

### Kernel Release

Linux archlinux 6.15.2-zen1-1-zen #1 ZEN SMP PREEMPT_DYNAMIC Tue, 10 Jun 2025 21:32:14 +0000 x86_64 GNU/Linux

### Please confirm you are running a stable release kernel (e.g. not a -rc). We do not accept bug reports for unreleased kernels.

- [x] I am running on a stable kernel release.

### Hardware: GPU

GPU 0: NVIDIA GeForce RTX 5070 Ti Laptop GPU (UUID: GPU-846a94a0-8570-27c4-9943-057ea9ee7cea)

### Describe the bug

Hello everyone, I recently purchased an MSI laptop with a Core i7 255HX and an RTX5070 TI Mobile. I initially installed the drivers with the proprietary module, but the card wasn’t detected by the system. After that, I realized that the drivers were needed, and with that, the card now works, at least in a basic way. What I noticed was that RTD3 for the graphics card isn’t available. On the first attempt, it appeared disabled by default when running the query:

cat /proc/driver/nvidia/gpus/0000:01:00.0/power

It also appears as D0 when running the query:

cat /sys/class/drm/card*/device/power_state

### To Reproduce

Install nvidia drivers.
Check RTD3 status using "cat /proc/driver/nvidia/gpus/0000:01:00.0/power" and "cat /sys/class/drm/card*/device/power_state"

### Bug Incidence

Always

### nvidia-bug-report.log.gz

[nvidia-bug-report.log.gz](https://github.com/user-attachments/files/20785360/nvidia-bug-report.log.gz)

### More Info

This happen with stable branch too.
This does not happen with another RTX series, for example, I have a RTX3060 mobile with RTD3 working correctly with the same driver version.

nekromantik, 2025-06-18:
Plus 1 cachy OS

Mario156090, 2025-06-18:
+ @aritger

ptr1337, 2025-06-18:
FYI, Im not sure if this affects blackwell, but we are enabling Dynamic Power Management on CachyOS OOB:
https://github.com/CachyOS/CachyOS-Settings/blob/master/usr/lib/modprobe.d/nvidia.conf#L35

Just as an extra information

Mario156090, 2025-06-18:
> FYI, Im not sure if this affects blackwell, but we are enabling Dynamic Power Management on CachyOS OOB: https://github.com/CachyOS/CachyOS-Settings/blob/master/usr/lib/modprobe.d/nvidia.conf#L35
> 
> Just as an extra information

Yeah, that troublehooting was the first thing I tested. But after that We get Not supported. :/

[image]

Mario156090, 2025-06-22:
> Was the ooen source driver installed and not the proprietary one? I asl since starting with the 5000 series you can only use the open source one. 

Yes, this problem is Open source driver.

Binary-Eater, 2025-06-24:
To avoid losing track of this issue, I have opened NVBug 5356833 for this.

Neutron3529, 2025-06-26:
> > FYI, Im not sure if this affects blackwell, but we are enabling Dynamic Power Management on CachyOS OOB: https://github.com/CachyOS/CachyOS-Settings/blob/master/usr/lib/modprobe.d/nvidia.conf#L35
> > Just as an extra information
> 
> Yeah, that troublehooting was the first thing I tested. But after that We get Not supported. :/
> 
> [image]
> 
> [nvidia-bug-report.log.gz](https://github.com/user-attachments/files/20802507/nvidia-bug-report.log.gz)

```
[neutron@9955HX3D-5090 me]$ cat /proc/driver/nvidia/gpus/0000:01:00.0/power 
Runtime D3 status:          Enabled (coarse-grained)
Video Memory:               Active

GPU Hardware Support:
 Video Memory Self Refresh: Not Supported
 Video Memory Off:          Supported

S0ix Power Management:
 Platform Support:          Supported
 Status:                    Disabled

Notebook Dynamic Boost:     Supported
[neutron@9955HX3D-5090 me]$ nvidia-smi
Fri Jun 27 01:40:06 2025       
+-----------------------------------------------------------------------------------------+
| NVIDIA-SMI 575.64                 Driver Version: 575.64         CUDA Version: 12.9     |
|-----------------------------------------+------------------------+----------------------+
| GPU  Name                 Persistence-M | Bus-Id          Disp.A | Volatile Uncorr. ECC |
| Fan  Temp   Perf          Pwr:Usage/Cap |           Memory-Usage | GPU-Util  Compute M. |
|                                         |                        |               MIG M. |
|=========================================+========================+======================|
|   0  NVIDIA GeForce RTX 5090 ...    On  |   00000000:01:00.0  On |                  N/A |
| N/A   47C    P5             55W /   95W |    2816MiB /  24463MiB |     78%      Default |
|                                         |                        |                  N/A |
+-----------------------------------------+------------------------+----------------------+
```

Mario156090, 2025-06-26:
> .

Hello and thanks You for your comment. I did not tested coarse mode, only fine.

Did You feel the inital problem using fine and get that New result using coarse on your RTX5090?

Mario156090, 2025-06-26:
@Neutron3529, after testing what you reported in your finding, I was indeed able to see that RTD3 is working through the main query. However, in the next query, where we can verify PCI power support, we always see D0, which means the card cannot be shut down. In other words, RTD3 is still not working. In any case, I thank you for this finding, as we can now confirm that the problem is only with the fine-grained RTD3.

[image]

ptr1337, 2025-06-27:
@Mario156090 I think we are using for sddm already the wayland, if you have "cachyos-kde-settings" installed:
https://github.com/CachyOS/cachyos-kde-settings/blob/develop/usr/lib/sddm/sddm.conf.d/zz-wayland.conf

brynmailer, 2025-07-02:
Also encountering this issue on an RTX 5070 ti mobile. Very annoying. My setup seems very similar to @Mario156090 (same driver version, using sddm etc). Although, even after following the advice in this thread and setting runtime D3 status to course grained, as well as forcing sddm to use the wayland backend and hence no longer run on the dGPU. The cards power_state remains fixed at D0. 

[image]

Mario156090, 2025-07-03:
Hello everyone, after the last update. Fine grained is working correctly.

[image]

But there is a visual glitch when Nvidia is not active.

[image]

Well, it is a progress.

closed 2025-07-03 by Mario156090

JaneSmith, 2025-09-09:
How did you get this working? I'm having this issue with my RTX 5090 FE. Am I doing something wrong?

```
❯ cat /sys/bus/pci/devices/0000:2d:00.0/power_state                 
D0
```
```
❯ cat /proc/driver/nvidia/gpus/0000:2d:00.0/power                   
Runtime D3 status:          Not supported
Video Memory:               Active

GPU Hardware Support:
 Video Memory Self Refresh: Not Supported
 Video Memory Off:          Supported

S0ix Power Management:
 Platform Support:          Not Supported
 Status:                    Disabled

Notebook Dynamic Boost:     Not Supported
```
```
❯ cat /proc/driver/nvidia/params | grep DynamicPowerManagement
DynamicPowerManagement: 2
DynamicPowerManagementVideoMemoryThreshold: 200
```
```
❯ nvidia-smi
Tue Sep  9 04:35:15 2025       
+-----------------------------------------------------------------------------------------+
| NVIDIA-SMI 575.64.05              Driver Version: 575.64.05      CUDA Version: 12.9     |
|-----------------------------------------+------------------------+----------------------+
| GPU  Name                 Persistence-M | Bus-Id          Disp.A | Volatile Uncorr. ECC |
| Fan  Temp   Perf          Pwr:Usage/Cap |           Memory-Usage | GPU-Util  Compute M. |
|                                         |                        |               MIG M. |
|=========================================+========================+======================|
|   0  NVIDIA GeForce RTX 5090        On  |   00000000:2D:00.0 Off |                  N/A |
|  0%   40C    P8             16W /  575W |       1MiB /  32607MiB |      0%      Default |
|                                         |                        |                  N/A |
+-----------------------------------------+------------------------+----------------------+
                                                                                         
+-----------------------------------------------------------------------------------------+
| Processes:                                                                              |
|  GPU   GI   CI              PID   Type   Process name                        GPU Memory |
|        ID   ID                                                               Usage      |
|=========================================================================================|
|  No running processes found                                                             |
+-----------------------------------------------------------------------------------------+

```

Mario156090, 2025-09-09:
> How did you get this working? I'm having this issue with my RTX 5090 FE. Am I doing something wrong?
>
> ```
> ❯ cat /sys/bus/pci/devices/0000:2d:00.0/power_state
> D0
> ```
> ```
> ❯ cat [...]
Hello, upgrade to 580 drivers and test. 

If RTD3 does not work Open a new threat with a new bug report.