NVIDIA/open-gpu-kernel-modules #883: RTX 5000 Mobile Series w/ External Mon - Crash after Screen Off / Suspend (Plasma Wayland)

sk0rabu, 2025-06-18:
### NVIDIA Open GPU Kernel Modules Version

575.64

### Please confirm this issue does not happen with the proprietary driver (of the same version). This issue tracker is only for bugs specific to the open kernel driver.

- [ ] I confirm that this does not happen with the proprietary driver package.

### Operating System and Version

Arch Linux 6.15.2

### Kernel Release

6.15.2-arch1-1 #1 SMP PREEMPT_DYNAMIC Tue, 10 Jun 2025 21:32:33 +0000 x86_64 GNU/Linux

### Please confirm you are running a stable release kernel (e.g. not a -rc). We do not accept bug reports for unreleased kernels.

- [x] I am running on a stable kernel release.

### Hardware: GPU

NVIDIA GeForce RTX 50X0 Laptop GPU

### Describe the bug

I've mainly been using Linux on my new Arrow Lake + RTX 5000 series laptop since I bought it earlier this year. I use it docked all the time (laptop screen off) with a DP over USBC connection to an external monitor (QHD @ 180Hz). I stayed on 570.153.02 for a long time as the last release didn't play nice with my laptop GPU.

I use KDE Plasma on Wayland, and suspend / hibernate is meh. Suspend mostly works with minor issues (after suspend / screen off, the screen resolution becomes super low and I need to unplug / plug the USBC cable so that resolution becomes normal). Sometimes neither the screens would have signal and I would have to switch to a separate TTY to gracefully shutdown my laptop. Hibernate is hit or miss; I guess if I made major changes e.g. a full upgrade it mostly would fail so I used it less and less.

On the recent driver release I was met with a full crash when I tried to wake up my laptop from screen-off status (not suspended yet). Please see the logs attached.

Many thanks in advance and please let me know if you need anything else.

### To Reproduce

Wait for the laptop to turn off its screen after idling for some time, or suspend it (Not sure if laptop screen off + output to external monitor is a necessity)

### Bug Incidence

Sometimes

### nvidia-bug-report.log.gz

[nvidia-bug-report.log](https://github.com/user-attachments/files/20798644/nvidia-bug-report.log)

Neutron3529, 2025-06-26:
You might try https://download.nvidia.com/XFree86/Linux-x86_64/435.17/README/dynamicpowermanagement.html
I use the proposed `80-nvidia-pm.rules`, and set `options nvidia "NVreg_DynamicPowerManagement=0x01"` rather than the proposed `options nvidia "NVreg_DynamicPowerManagement=0x02"`. 

At least for me, after this setting, the GPU won't hang anymore.

sk0rabu, 2025-07-12:
Believe this was pretty much resolved by the new drivers.