NVIDIA/open-gpu-kernel-modules #887: NVIDIA crashes after resume (from sleep or hibernation)

jotkauser, 2025-06-22:
### NVIDIA Open GPU Kernel Modules Version

575.64-2 (Latest one from Arch package: nvidia-open-lts)

### Please confirm this issue does not happen with the proprietary driver (of the same version). This issue tracker is only for bugs specific to the open kernel driver.

- [ ] I confirm that this does not happen with the proprietary driver package.

### Operating System and Version

Description:	Arch Linux

### Kernel Release

Linux asus-main 6.12.34-1-lts #1 SMP PREEMPT_DYNAMIC Thu, 19 Jun 2025 15:05:14 +0000 x86_64 GNU/Linux

### Please confirm you are running a stable release kernel (e.g. not a -rc). We do not accept bug reports for unreleased kernels.

- [x] I am running on a stable kernel release.

### Hardware: GPU

NVIDIA GeForce RTX 3050 Laptop GPU

### Describe the bug

So, i'm using Arch on my ASUS laptop. And sometimes when i sleep or hibernate, after resuming the NVIDIA driver doesn't work anymore. When i check dmesg i see this `[ 4331.789906] nvidia 0000:01:00.0: can't suspend (nv_pmops_runtime_suspend [nvidia] returned -5)`. After a reboot everything is okay. Sometimes this error prevents the kernel from suspend and putting me in a broken state. I noticed this issue sometimes show up, sometimes everything works fine.

### To Reproduce

- Install latest Arch Linux with LTS kernel.
- Install `nvidia-open-lts` package
- Configure Hibernation
- Try to sleep for some time (there is a chance it won't go to suspend state) (dmesg will show nvidia error)
- Check if driver works after resume from hibernation (nvidia-smi doesnt return unknown error)

### Bug Incidence

Sometimes

### nvidia-bug-report.log.gz

[nvidia-bug-report.log.gz](https://github.com/user-attachments/files/20854264/nvidia-bug-report.log.gz)

### More Info

Exact laptop: ASUS ROG Strix G15 (G15IC)
I'm using Wayland. I don't know if something would change on Xorg. NVIDIA is still a discrete GPU. Doesn't render image.
This problem also occurs on proprietary driver. But also occurs on the Open driver.

amrit1711, 2025-07-03:
Hi @jotkauser 
Thanks for reporting issue, could you please apply below patch and see if it fixes the issue.

https://github.com/NVIDIA/open-gpu-kernel-modules/commit/c7e72135da83ff027755b4a61a3ff09a32fe00c3

jotkauser, 2025-07-03:
> Hi [@jotkauser](https://github.com/jotkauser) Thanks for reporting issue, could you please apply below patch and see if it fixes the issue.
> 
> [c7e7213](https://github.com/NVIDIA/open-gpu-kernel-modules/commit/c7e72135da83ff027755b4a61a3ff09a32fe00c3)

So should i build the kernel module with this patch?

amrit1711, 2025-07-03:
Yes, you can build the open kernel with that patch.

jotkauser, 2025-07-07:
[nvidia-bug-report.log.gz](https://github.com/user-attachments/files/21101374/nvidia-bug-report.log.gz)

Driver stopped working without any errors related to NVIDIA in dmesg

lumingzh, 2025-07-09:
> [nvidia-bug-report.log.gz](https://github.com/user-attachments/files/21101374/nvidia-bug-report.log.gz)
> 
> Driver stopped working without any errors related to NVIDIA in dmesg

According to my test so far, the rule file is not needed. Add the nvidia.conf kernel module .conf file and change the value to 0x01 is enough to fix suspend issue. However, the NVRM still will show errors like "out of memory", so the bug is not fixed in fact.

alexmo1997, 2025-07-24:
Is this patch going to land one of the next releases?