NVIDIA/open-gpu-kernel-modules #889: /sys/class/drm/*/enabled reports “disabled” for enabled monitors

RMTT, 2025-06-24:
### NVIDIA Open GPU Kernel Modules Version

570.169

### Please confirm this issue does not happen with the proprietary driver (of the same version). This issue tracker is only for bugs specific to the open kernel driver.

- [ ] I confirm that this does not happen with the proprietary driver package.

### Operating System and Version

NixOS

### Kernel Release

6.15.2

### Please confirm you are running a stable release kernel (e.g. not a -rc). We do not accept bug reports for unreleased kernels.

- [x] I am running on a stable kernel release.

### Hardware: GPU

NVIDIA GeForce RTX 5070 Ti Laptop GPU

### Describe the bug

That's old problem which had been reported in https://forums.developer.nvidia.com/t/sys-class-drm-enabled-reports-disabled-for-enabled-monitors/51187 , but it doesn't been solved until now.

I found that kernel updates `/sys/class/drm/*/enabled` by(https://github.com/torvalds/linux/blob/master/drivers/gpu/drm/drm_sysfs.c#L251):
```
	struct drm_connector *connector = to_drm_connector(device);
	bool enabled;

	enabled = READ_ONCE(connector->encoder);

	return sysfs_emit(buf, enabled ? "enabled\n" : "disabled\n");
```
Seems just check the existence of  field `connector->encoder` . I skimmed nvidia-drm sources, seems there is no assignment for `connector->encoder`, and it only uses `connector->nv_detected_encoder` .

Some softwares which depend on /sys/class/drm/*/enabled to detect monitor's status cannot work correctly when use nvidia card, for example, logind in laptop with nvidia card.

### To Reproduce

`cat /sys/class/drm/*/enabled`

### Bug Incidence

Once

### nvidia-bug-report.log.gz

[nvidia-bug-report.log.gz](https://github.com/user-attachments/files/20884280/nvidia-bug-report.log.gz)

Binary-Eater, 2025-07-27:
As a FYI, it's a bug if a DRM driver explicitly sets `connector->encoder`.

```
/**
 * drm_connector_attach_encoder - attach a connector to an encoder
 * @connector: connector to attach
 * @encoder: encoder to attach @connector to
 *
 * This function links up a connector to an encoder. Note that the routing
 * restrictions between encoders and crtcs are exposed to userspace through the
 * possible_clones and possible_crtcs bitmasks.
 *
 * Returns:
 * Zero on success, negative errno on failure.
 */
int drm_connector_attach_encoder(struct drm_connector *connector,
				 struct drm_encoder *encoder)
{
	/*
	 * In the past, drivers have attempted to model the static association
	 * of connector to encoder in simple connector/encoder devices using a
	 * direct assignment of connector->encoder = encoder. This connection
	 * is a logical one and the responsibility of the core, so drivers are
	 * expected not to mess with this.
	 *
	 * Note that the error return should've been enough here, but a large
	 * majority of drivers ignores the return value, so add in a big WARN
	 * to get people's attention.
	 */
	if (WARN_ON(connector->encoder))
		return -EINVAL;

	connector->possible_encoders |= drm_encoder_mask(encoder);

	return 0;
}
EXPORT_SYMBOL(drm_connector_attach_encoder);
```

RMTT, 2025-07-27:
> Looking into resolving this

Thanks! This bug is very crucial for improving linux desktop experience with NVIDIA in laptops.

Binary-Eater, 2025-08-02:
Wanted to update we have a fix for this issue internally.

RMTT, 2025-08-02:
> Wanted to update we have a fix for this issue internally.

Thanks for your work! Will this fix be included in next release?

abchauhan-nv, 2025-08-13:
Hi all, the issue was tracked on NVBug #5420364. This is now resolved and the fix will be available in a future driver release. Thanks

ShadiestGoat, 2025-08-26:
Hi - is this fix supposed to be in current release? I seem to still be able to reproduce this on latest version

Binary-Eater, 2025-09-03:
We just had a new release today.

https://www.nvidia.com/en-us/drivers/details/253003/

Binary-Eater, 2025-09-24:
I think the correct thing for logind to do is account for both sysfs nodes. It is expected for atomic modesetting drivers to not implement the legacy helper that populates the information for the legacy sysfs node. That sysfs node is supported by drivers just to not break uapi from the time those drivers supported the legacy modeset path. The difference for X is with the sunsetting of X development, DRM driver developers implemented a X modesetting driver that can use DRM kernel drivers to drive the X display. However, this removes the opportunity for enhancing performance/improving behavior for GLX support. We want to have an enhanced GLX experience, so we still ship a custom X userspace driver and do not plumb through DRM with this driver.

Binary-Eater, 2025-09-24:
If you wanted, you could use the X modesetting userspace driver which would make use of `nvidia-drm` in place of the NVIDIA X userspace driver, but you would be choosing to throw away GLX performance.

ShadiestGoat, 2025-09-24:
I see, sad. I guess this is a sign to move to Wayland anyways