NVIDIA/open-gpu-kernel-modules #893: "libspdm expects LKCA but found stubs" on recent kernel versions

snir911, 2025-06-29:
### NVIDIA Open GPU Kernel Modules Version

570.158.01

### Please confirm this issue does not happen with the proprietary driver (of the same version). This issue tracker is only for bugs specific to the open kernel driver.

- [x] I confirm that this does not happen with the proprietary driver package.

### Operating System and Version

Fedora 41

### Kernel Release

6.15.3-100.fc41

### Please confirm you are running a stable release kernel (e.g. not a -rc). We do not accept bug reports for unreleased kernels.

- [x] I am running on a stable kernel release.

### Hardware: GPU

NVIDIA H100 NVL

### Describe the bug

Nvidia driver is failing to operate in Confidential mode on recent kernels, it's failing with the following error:
`libspdm_check_crypto_backend: Error - libspdm expects LKCA but found stubs`

### To Reproduce

Compile and install driver on 6.13+ kernel and try to set conf compute to ready state: `nvidia-smi conf-compute -srs 1` on confidential machine.

### Bug Incidence

Always

### nvidia-bug-report.log.gz

[nvidia-bug-report.log.gz](https://github.com/user-attachments/files/20966810/nvidia-bug-report.log.gz)

### More Info

The issue appears to originate from a failed `compile_check_conftest` for LKCA ECC support (which is available and should be supported), specifically the `ecc_digits_from_bytes` function.
Observed Error (during driver build/conftest):
```
/lib/modules/6.15.3-100.fc41.x86_64/build//include/crypto/internal/ecc.h:53:13: error: variably modified 'r' at file scope 
  53 |        u64 r[ECC_MAX_DIGITS]; 
     |            ^ 
/lib/modules/6.15.3-100.fc41.x86_64/build//include/crypto/internal/ecc.h:54:13: error: variably modified 's' at file scope 
  54 |        u64 s[ECC_MAX_DIGITS]; 
     |            ^
```
This seems to cause the conftest to skip LKCA support, which results in the driver defaulting to stub crypto logic, leading to the error mentioned above, although the used kernel supports all the required crypto APIs.

steven-bellock, 2025-07-14:
This was tracked internally as bug 5160312.

@snir911 the fix for this will be present in the upcoming `r570` update. Needed to `#include <linux/math.h>` in `conftest.sh`.

steven-bellock, 2025-07-17:
This should be resolved now with https://github.com/NVIDIA/open-gpu-kernel-modules/blob/4d6c416e7fd20dba9b24e6b1d352284473af7812/kernel-open/conftest.sh#L6967

mtijanic, 2025-08-04:
Closing per above.

closed 2025-08-04 by mtijanic