NVIDIA/open-gpu-kernel-modules #895: My 4060 won't accept more than 120Hz

desamont, 2025-07-01:
### NVIDIA Open GPU Kernel Modules Version

575.64

### Please confirm this issue does not happen with the proprietary driver (of the same version). This issue tracker is only for bugs specific to the open kernel driver.

- [x] I confirm that this does not happen with the proprietary driver package.

### Operating System and Version

Arch Linux

### Kernel Release

6.15.4-arch2-1

### Please confirm you are running a stable release kernel (e.g. not a -rc). We do not accept bug reports for unreleased kernels.

- [x] I am running on a stable kernel release.

### Hardware: GPU

NVIDIA GeForce RTX 4060

### Describe the bug

my monitor supports 165 Hz, but if I set it higher than 120 (144 or 165) - then a black screen. I installed the driver from the Arch repositories, and here, still the same problem, and only on this distro

### To Reproduce

I install the driver from your installer, reboot the computer, everything works, but it doesn't go above 120 Hz

### Bug Incidence

Always

### nvidia-bug-report.log.gz

[nvidia-bug-report.log](https://github.com/user-attachments/files/21005552/nvidia-bug-report.log)

wilsoft-gt, 2025-07-23:
Same issue for Ubuntu 24.04 kernel 6.8.0-64-generic, my issue is happening with both, private an open drivers.

a5xwin, 2025-07-25:
on arch linux with the open nvidia driver (575.64), setting refresh rate above 120hz on my rtx 4060 just gives a black screen. monitor supports 165hz fine. tried the proprietary driver (same version) and everything works there. seems like the open driver can't handle higher refresh rates properly... maybe missing edid stuff or bad mode setting. everything else installs fine. definitely a bug with the open modules...

luisalvarado, 2025-07-25:
> on arch linux with the open nvidia driver (575.64), setting refresh rate above 120hz on my rtx 4060 just gives a black screen. monitor supports 165hz fine. tried the proprietary driver (same [...]

If it helps using Ubuntu 25.04 I use 144hz

luisalvarado, 2025-07-27:
> [@luisalvarado](https://github.com/luisalvarado) yeah same here man, open driver just blanks out anything above 120hz on my 4060. tried everything.. xrandr, modsetting, kernel params, nothing helped. but switching to the proprietary driver fixed it instantly. definitely feels like the open one doesn’t handle high refresh properly yet.

That is important to know. Thank you. What driver version was it?

closed 2025-08-13 by desamont