NVIDIA/open-gpu-kernel-modules #902: Display rotation via kernel argument does not work

Nivalux, 2025-07-18:
### NVIDIA Open GPU Kernel Modules Version

575.64.03

### Please confirm this issue does not happen with the proprietary driver (of the same version). This issue tracker is only for bugs specific to the open kernel driver.

- [ ] I confirm that this does not happen with the proprietary driver package.

### Operating System and Version

NixOS 25.05

### Kernel Release

6.15.6

### Please confirm you are running a stable release kernel (e.g. not a -rc). We do not accept bug reports for unreleased kernels.

- [x] I am running on a stable kernel release.

### Hardware: GPU

NVIDIA GeForce RTX 3080

### Describe the bug

I am using two main monitors, one of which is a vertical screen.
The vertical screen has the wrong rotation during boot and the TUI login manager (I use `ly`).

The kernel has "video=..." and "fbcon=..." options, to change the display rotation. Setting the "video=..." option had no effect with the open-gpu kernel module.

### To Reproduce

1) Determine, which port the Display is on. I used `xrandr --query` for this.
2) Set the kernel parameter to one of the following: (Change the DP-3 according to your Port)
"video=DP-3:rotate=90"
"video=DP-3:panel_orientation=right_side_up"
"video=rotate=90"
"video=rotate=270"
"fbcon=rotate:3"
"fbcon=rotate:1"

Only the last two options have an effect. But they rotate all displays, not just one.

### Bug Incidence

Always

### nvidia-bug-report.log.gz

[nvidia-bug-report.log.gz](https://github.com/user-attachments/files/21315114/nvidia-bug-report.log.gz)

Binary-Eater, 2025-07-26:
Hi @Nivalux,

I just spent some time debugging the issue on my local setup and found that the DRM subsystem's `modedb` support is working with our DRM fbdev implementation. However, I also struggled understanding how to specify the kernel parameter correctly without reading the linux kernel source code related to this logic.

* https://docs.kernel.org/fb/modedb.html
* https://elixir.bootlin.com/linux/v6.15.8/source/drivers/gpu/drm/drm_modes.c#L2337
* https://elixir.bootlin.com/linux/v6.15.8/source/drivers/gpu/drm/drm_connector.c#L166

When you read through the source code, you will see that specifying a connector and mode explicitly along with `rotate` behaves as expected.

Here is an example that worked for me, `video=DP-3:3440x1440@60D,rotate=180`

You can check for viable modelines using `cat /sys/class/drm/card1-DP-3/modes` where you will replace `card1-DP-3` as appropriate.

Thanks,
Rahul Rameshbabu

Nivalux, 2025-07-28:
Hi @Binary-Eater ,

thank you for investigating this!

I tried out different combinations with these results:
worked:        `video=DP-3:1920x1080@60D,rotate=180`
didn't work: `video=DP-3:1920x1080@60D,rotate=90`
didn't work: `video=DP-3:1920x1080@60D,rotate=270`
didn't work: `video=DP-3:1080x1920@60D,rotate=270`

I also tried, if swapping x and y would change anything, but that produced the following error, but otherwise didn't do anything.
`nvidia 0000:0b:00.0: [drm] User-defined mode not supported: "1080x1920": 60 178312 1080 1168 1288 1496 1920 1921 1924 1987 0x20 0x6`

Conclusion, for the moment: It seems only 180 degree rotations work. 90 degree left or right don't work.
In my case, I have the monitor with the right side up.
Does a 90 degree rotation work on your end?

```
cat /sys/class/drm/card1-DP-3/modes
1920x1080
1920x1080
1920x1080
1920x1080
1920x1080
1920x1080
1920x1080
1680x1050
1280x1024
1280x1024
1440x900
1280x960
1280x800
1152x864
1280x720
1280x720
1280x720
1024x768
1024x768
1024x768
800x600
800x600
800x600
800x600
720x576
720x480
640x480
640x480
640x480
640x480
```

Binary-Eater, 2025-07-28:
Hi @Nivalux,

That's quite interesting. After I tested 180 degrees, I immediately tested `video=DP-3:3440x1440@60D,rotate=90` and that worked for me as expected.

Would you mind sharing the generated logs from `nvidia-bug-report.sh` for the `rotate=180` case (that works) and the `rotate=90` care (that does not work)?

Binary-Eater, 2025-08-09:
@Nivalux thanks for sharing the logs. They were helpful for confirming that the parameter was being applied even though specifying 90 degree rotation is not helping. I wrote a `bpftrace` script to help debug this issue further since I do not have a reproduction.

https://github.com/Binary-Eater/bpftrace-scripts/blob/trunk/drm/drm_client_rotation.bpftrace

You can use this script once you enter whatever environment you prefer (a Wayland compositor, X server session, etc.), with the `video=DP-3:1920x1080@60D,rotate=90` kernel parameter in use.
Once the script is running, can you perform a VT switch to a framebuffer console? That will generate useful output on the script once you VT switch back to where the script is running.

Here is an example usage that I ran with `video=DP-3:3440x1440@60D,rotate=90`. `0x2` corresponds to `DRM_MODE_ROTATE_90`.

```
~/Documents/bpftrace-scripts/drm trunk
❯ sudo bpftrace ./drm_client_rotation.bpftrace
Attaching 1 probe...
drm_client_rotation: rotation: 0x2, return: 0
drm_client_rotation: rotation: 0x0, return: 0
drm_client_rotation: rotation: 0x0, return: 0
drm_client_rotation: rotation: 0x0, return: 0
drm_client_rotation: rotation: 0x2, return: 0
drm_client_rotation: rotation: 0x0, return: 0
drm_client_rotation: rotation: 0x0, return: 0
drm_client_rotation: rotation: 0x0, return: 0
drm_client_rotation: rotation: 0x2, return: 0
drm_client_rotation: rotation: 0x0, return: 0
drm_client_rotation: rotation: 0x0, return: 0
drm_client_rotation: rotation: 0x0, return: 0
drm_client_rotation: rotation: 0x2, return: 0
drm_client_rotation: rotation: 0x0, return: 0
drm_client_rotation: rotation: 0x0, return: 0
drm_client_rotation: rotation: 0x0, return: 0
drm_client_rotation: rotation: 0x2, return: 0
drm_client_rotation: rotation: 0x0, return: 0
drm_client_rotation: rotation: 0x0, return: 0
drm_client_rotation: rotation: 0x0, return: 0
drm_client_rotation: rotation: 0x2, return: 0
drm_client_rotation: rotation: 0x0, return: 0
drm_client_rotation: rotation: 0x0, return: 0
drm_client_rotation: rotation: 0x0, return: 0
drm_client_rotation: rotation: 0x2, return: 0
drm_client_rotation: rotation: 0x0, return: 0
drm_client_rotation: rotation: 0x0, return: 0
drm_client_rotation: rotation: 0x0, return: 0
^C
```

Nivalux, 2025-08-10:
Hi @Binary-Eater 

thank you for investigating this!
I ran the bpftrace command on wayland, switched to tty3, waited for 5 seconds, and switched back to wayland.
This is the result:

```
Attaching 1 probe...
drm_client_rotation: rotation: 0x1, return: 1
drm_client_rotation: rotation: 0x2, return: 0
drm_client_rotation: rotation: 0x0, return: 0
drm_client_rotation: rotation: 0x0, return: 0
drm_client_rotation: rotation: 0x1, return: 1
drm_client_rotation: rotation: 0x2, return: 0
drm_client_rotation: rotation: 0x0, return: 0
drm_client_rotation: rotation: 0x0, return: 0
drm_client_rotation: rotation: 0x1, return: 1
drm_client_rotation: rotation: 0x2, return: 0
drm_client_rotation: rotation: 0x0, return: 0
drm_client_rotation: rotation: 0x0, return: 0
drm_client_rotation: rotation: 0x1, return: 1
drm_client_rotation: rotation: 0x2, return: 0
drm_client_rotation: rotation: 0x0, return: 0
drm_client_rotation: rotation: 0x0, return: 0
^C
```

Binary-Eater, 2025-08-10:
Would you mind also sharing the output with the working 180 degree rotation case?

Binary-Eater, 2025-08-10:
Also, would you mind testing 90 degree rotation with a single monitor connected after that?

Nivalux, 2025-08-10:
So the result of that test is, that it does work, when I only have the vertical monitor connected. Then it rotates. In my case 270 degrees is the correct rotation for that one.
After that, I also tested if the port order made any difference. I connected the vertical to DP-1 (before DP-3) and the main monitor to DP-2 (before DP-1) and it didn't change anything on the fb console.

```
180 degree, working (with all monitors connected, unchanged to before)

Attaching 1 probe...
drm_client_rotation: rotation: 0x1, return: 1
drm_client_rotation: rotation: 0x4, return: 1
drm_client_rotation: rotation: 0x0, return: 0
drm_client_rotation: rotation: 0x0, return: 0
drm_client_rotation: rotation: 0x1, return: 1
drm_client_rotation: rotation: 0x4, return: 1
drm_client_rotation: rotation: 0x0, return: 0
drm_client_rotation: rotation: 0x0, return: 0
drm_client_rotation: rotation: 0x1, return: 1
drm_client_rotation: rotation: 0x4, return: 1
drm_client_rotation: rotation: 0x0, return: 0
drm_client_rotation: rotation: 0x0, return: 0
drm_client_rotation: rotation: 0x1, return: 1
drm_client_rotation: rotation: 0x4, return: 1
drm_client_rotation: rotation: 0x0, return: 0
drm_client_rotation: rotation: 0x0, return: 0
^C

270 degree, only the one vertical monitor connected, working! (90 degree also worked)
Attaching 1 probe...
drm_client_rotation: rotation: 0x8, return: 0
drm_client_rotation: rotation: 0x0, return: 0
drm_client_rotation: rotation: 0x0, return: 0
drm_client_rotation: rotation: 0x0, return: 0
drm_client_rotation: rotation: 0x8, return: 0
drm_client_rotation: rotation: 0x0, return: 0
drm_client_rotation: rotation: 0x0, return: 0
drm_client_rotation: rotation: 0x0, return: 0
drm_client_rotation: rotation: 0x8, return: 0
drm_client_rotation: rotation: 0x0, return: 0
drm_client_rotation: rotation: 0x0, return: 0
drm_client_rotation: rotation: 0x0, return: 0
drm_client_rotation: rotation: 0x8, return: 0
drm_client_rotation: rotation: 0x0, return: 0
drm_client_rotation: rotation: 0x0, return: 0
drm_client_rotation: rotation: 0x0, return: 0
^C
```

Binary-Eater, 2025-08-10:
Thanks @Nivalux . That confirms my theories. 90 degree and 270 degree rotations are handled purely by kernel SW without involving the `nvidia-drm` driver/utilizing HW to perform the rotation. I suspect there is a bug here with regards to multi-monitor support in the SW rotation path I will look into.

Binary-Eater, 2025-10-26:
I have confirmed the exact nature of the issue. Both `i915` and `nvidia-drm` do not support HW rotations for `DRM_MODE_ROTATE_90` and `DRM_MODE_ROTATE_270`.

* https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/tree/drivers/gpu/drm/i915/display/i9xx_plane.c?id=72761a7e31225971d0b29d9195e0ffa986b77867#n1100
* https://github.com/NVIDIA/open-gpu-kernel-modules/blob/2b436058a616676ec888ef3814d1db6b2220f2eb/src/nvidia-modeset/src/nvkms-evo3.c#L6238-L6247

`fbcon` supports software rotation fallback when hardware is not capable of the requested rotations.
However, this support can only support one rotation type at a time.
Therefore, mixed rotations on different monitors are not supported.
The reason 180 degree rotation works is because the driver/HW supports it in the case of `i915` and `nvidia-drm`.

https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/tree/drivers/gpu/drm/drm_fb_helper.c?id=72761a7e31225971d0b29d9195e0ffa986b77867#n1807

Nivalux, 2025-11-06:
Thank you for investigating it and finding the reason!
Now I now, that for the moment I can't use mixed rotations between multiple monitors in the Framebuffer console.

Should this issue be closed now that the reason is found or should it remain open, in case someone want's to implement the solution to this issue?