NVIDIA/open-gpu-kernel-modules #903: Frozen monitor and eventual crash when opening Steam on Ubuntu 24.04

uyha, 2025-07-18:
### NVIDIA Open GPU Kernel Modules Version

570.169

### Please confirm this issue does not happen with the proprietary driver (of the same version). This issue tracker is only for bugs specific to the open kernel driver.

- [ ] I confirm that this does not happen with the proprietary driver package.

### Operating System and Version

Description:    Ubuntu 24.04.2 LTS

### Kernel Release

Linux 6.14.0-24-generic #24~24.04.3-Ubuntu SMP PREEMPT_DYNAMIC Mon Jul  7 16:39:17 UTC 2 x86_64 x86_64 x86_64 GNU/Linux

### Please confirm you are running a stable release kernel (e.g. not a -rc). We do not accept bug reports for unreleased kernels.

- [x] I am running on a stable kernel release.

### Hardware: GPU

GPU 0: NVIDIA GeForce RTX 5080 Laptop GPU (UUID: GPU-83a57f68-c8a4-b38c-555d-91d8cf646250)

### Describe the bug

I have 2 monitors connected my laptop via an HDMI port and a display port. When opening Steam and Steam games, the monitors would freeze for a bit, and then the monitor connected to the display port would unfreeze and work normally. However, the monitor connected to the HDMI port would stay frozen. I can unfreeze it by going to "Settings" and change the monitor arrangement. However, the freeze would continue to occur when I play games and eventually crashes the system. 

### To Reproduce

1. Install Ubuntu 24.04
2. Open "Additional Drivers"
3. Select the option `Using NVIDIA driver (open kernel) metapackage from nvidia-driver-570-open (proprietary)`
4. Update all the packages
5. Restart the machine
6. Open Steam / Steam games.

### Bug Incidence

Always

### nvidia-bug-report.log.gz

[nvidia-bug-report.log.gz](https://github.com/user-attachments/files/21315849/nvidia-bug-report.log.gz)

gatti2602, 2025-08-09:
Same here

[nvidia-bug-report.log.gz](https://github.com/user-attachments/files/21693474/nvidia-bug-report.log.gz)

Binary-Eater, 2025-09-07:
In both reported logs, I did notice the following.

```
Aug 08 19:41:00 ngatti-Legion-5 kernel: nvidia-modeset: WARNING: GPU:0: Unable to read EDID for display device DP-4
Aug 08 19:41:00 ngatti-Legion-5 kernel: nvidia-modeset: WARNING: GPU:0: Unable to read EDID for display device DP-4
```

For the DisplayPort monitor, are you using some kind of adapter for native DisplayPort on a laptop GPU or does this laptop have native DisplayPort. Looking at the logs, I suspect this laptop is a Lenovo Legion 5 and Lenovo Legion 7 Pro?

gatti2602, 2025-09-07:
It has native display port its a Lenovo Legion 5i

Saludos,
Nicolás Gatti.

uyha, 2025-09-07:
I'm using the Lenovo Legion Pro 7i 10th gen. It has a native display port. However, the freezing does not happen on the monitor connected to the display port, rather the HDMI monitor.

uyha, 2025-10-16:
This currently still happens with 580.65.06.