NVIDIA/open-gpu-kernel-modules #905: nvidia_drm prevents gpu from staying in suspended state

imaGuru, 2025-07-20:
### NVIDIA Open GPU Kernel Modules Version

575.64.03

### Please confirm this issue does not happen with the proprietary driver (of the same version). This issue tracker is only for bugs specific to the open kernel driver.

- [ ] I confirm that this does not happen with the proprietary driver package.

### Operating System and Version

Arch Linux

### Kernel Release

6.15.7-arch1-1 #1 SMP PREEMPT_DYNAMIC Thu, 17 Jul 2025 21:05:29 +0000 x86_64 GNU/Linux

### Please confirm you are running a stable release kernel (e.g. not a -rc). We do not accept bug reports for unreleased kernels.

- [x] I am running on a stable kernel release.

### Hardware: GPU

NVIDIA GeForce RTX 4060 Laptop GPU (UUID: GPU-a0338cc2-52ca-fc18-d87f-0a5df1c5ca22)

### Describe the bug

Laptop: Lenovo Legion Slim 5 16ahp9
CPU: AMD Ryzen 7 8845HS w/ Radeon 780M Graphics

GPU can enter suspended state and turn off memory but won't stay cold. After 1s GPU is woken up. Waiting 15s causes GPU to sleep again, only to repeat the cycle. This behaviour exists even when not booting into graphical interface (gdm/gnome/wayland/xorg) and no applications are using the GPU

nvidia_drm module can be removed with `modprobe -r` when not running gnome (wayland/xorg). After removal the GPU suspends normally. Starting GDM and gnome with nvidia_drm removed works - running apps on dGPU also seems to work. Enabling nvidia-persistenced, nvidia-powerd or modprobing nvidia_drm brings back the bug with 15s turn on/off cycle

These look like the same or very similar issues:
  - https://forums.developer.nvidia.com/t/nvidia-gpu-fails-to-power-off-prime-razer-blade-14-2022/250023/40
  - https://forums.developer.nvidia.com/t/4070-555-and-560-drivers-wont-stay-in-d3cold-lenovo-legion-slim-5/302967
  - https://www.markwatkinson.com/knowledge/linux/nvidia-dgpu-power/#known-issues

### To Reproduce

1. Boot into system
2. Wait 15s for GPU to go to sleep. 
3. GPU wakes up after 1s. 
4. Cycle continues every 15s

Removing nvidia_drm module allows the GPU to power down

### Bug Incidence

Always

### nvidia-bug-report.log.gz

[nvidia-bug-report.log.gz](https://github.com/user-attachments/files/21334982/nvidia-bug-report.log.gz)

Binary-Eater, 2025-07-25:
This sounds like its similar in nature to the problem I am trying to fix here: https://github.com/NVIDIA/open-gpu-kernel-modules/issues/759#issuecomment-2565928704

If you load `nvidia_drm` with the kernel parameters `modeset=1 fbdev=0`, are you able to reproduce the issue?

Binary-Eater, 2025-07-26:
Thanks @imaGuru. That suggests to me that there may be other paths in `nvidia-drm` where we may be making allocations that have certain requirements, preventing the GPU from being able to stay in RTD3. It would help me if we could conduct the following test.

1. Reload `nvidia-drm` with `modeset=0` parameter
1. Launch GNOME (using Xorg)
1. Re-conduct the experiment to see if the GPU now stays in RTD3

If my theory is correct, I would expect this to enable the GPU to stay in RTD3.

Binary-Eater, 2025-07-28:
Thanks @imaGuru ,

That's quite peculiar. The reason I wanted to run this experiment is because with `modeset=0`, `nvidia-drm` does almost nothing. Would you mind sharing a `nvidia-bug-report.sh` dump in the repro-ing case with `modeset=0`?

imaGuru, 2025-07-29:
Sure, here you go `modeset=0`:

[nvidia-bug-report.log.gz](https://github.com/user-attachments/files/21484578/nvidia-bug-report.log.gz)

To recap: first d3cold after boot or laptop suspend to ram holds, subsequent do not (last only 1s then wakeup). Removing nvidia_drm and stoping nvidia_powerd (removing just nvidia_drm with nvidia_powerd left running, keeps the gpu in the wakeup cycle) allows GPU to stay d3cold after first wakeup

imaGuru, 2025-07-30:
@francoism90 no, modset or fbdev do not change anything. I think I have tried every possible configuration of Nvidia driver, xorg and wayland, nothing works unfortunately. RTD3 works only the first time it triggers, after first wakeup it does enter rtd3cold but can never stay cold longer than 1s. The result is that it constantly powers off and on and seems to waste even more power, so it's better to disable rt3d or take great care not to wake the GPU after first sleep since boot/laptop suspend.

imaGuru, 2025-08-04:
@francoism90 Can you link the issue with nvidia-powerd not being compatible? Search for "Error in processParam call" doesn't bring anything up in the issue tracker nor google. For me, aside from these errors, nvidia-powerd seems to work - I get depending on laptop mode: quiet 55W, balanced 60W and performance 115W (you can check in nvidia-smi).

As for laptop sleep/suspend to ram, for me, it usually works. However sometimes the GPU gets "stuck" and I get a black screen of death right before suspend or shutdown and laptop refuses to shutdown. Only way to power off is by holding the power button. Is this what you are experiencing also?

francoism90, 2025-08-04:
@imaGuru I cannot find it either, maybe the comment has been removed?

This is where I found out about your issue: https://github.com/NVIDIA/open-gpu-kernel-modules/issues/432

I still have it when enabling nvida-powered, so you may need to create an issue:

```bash
aug 04 13:18:46 lenovo systemd[1]: Started nvidia-powerd.service - nvidia-powerd service.
aug 04 13:18:46 lenovo nvidia-powerd[376985]: nvidia-powerd version:2.0 (build 1)
aug 04 13:18:46 lenovo nvidia-powerd[376985]: ERROR! Failed to allocate GPU device handle 0x59
aug 04 13:18:46 lenovo systemd[1]: nvidia-powerd.service: Deactivated successfully.
```
It now fails, I think that's what it has to do now?

Mulukulum, 2025-08-17:
> Only semiworkable solution so far is suspending the laptop and after wakeup the first d3cold state holds correctly, subsequent do not.

Weird, for me D3Cold seems to hold correctly indefinitely after a single suspend-wakeup cycle even after the GPU is woken up again by applications. 

I have powerd enabled, but persistenced disabled. However, after manually starting persistenced after boot, the GPU still seems to go into D3Cold indefinitely even after being woken up. Same laptop and driver version but on Kernel 6.15.9 on Fedora 42.

[nvidia-bug-report.log.gz](https://github.com/user-attachments/files/21820673/nvidia-bug-report.log.gz)

imaGuru, 2025-08-17:
> You could try that, but since this issue doesn't happen on the Open Kernel Module, I don't see why the BIOS would matter here.

Did you mean the "proprietary driver"? For me it happens on both: open and closed source drivers. It does however seem to be some kind of software issue, because your laptop "almost" works... (maybe some archlinux patches). Do you have any specific nvidia configurations? Which desktop environment do you use?

It seems that you can change the version number in this link to download older versions of the BIOS (link is from this site
https://download.lenovo.com/consumer/mobiles/nrcn24ww.exe

I will try downgrading the BIOS sometime later and see if has any effect.

Mulukulum, 2025-08-17:
Yes, sorry I meant the proprietary driver. I've edited my original comment. I use GNOME (Fedora 42) but the issue also occurs on KDE (Bazzite) but it also occurs without a DE altogether. 

Curious that the issue still occurs on the proprietary drivers for you. I'm not sure how it works on Arch but Fedora's (more specifically, RPM Fusion's)  Nvidia package made me think I was installing the nonfree proprietary drivers package but actually installs the free package.

francoism90, 2025-08-17:
@Mulukulum How did you installed the driver? I'm also on Fedora Kionite, and they should be the closed ones.

Bazzite uses a different upstream for nvidia, not rpmfusion (if I'm correct).

francoism90, 2025-08-17:
@Mulukulum Thanks! Going to try, and check if this actually fixes the issue.

Hmm, so this https://github.com/CheariX/silverblue-akmods-keys/issues/14, isn't needed anymore?

Mulukulum, 2025-08-17:
Is the kmod actually getting rebuilt?

Maybe try
 
```
sudo akmods --rebuild --force --akmod nvidia
```

francoism90, 2025-08-17:
@Mulukulum Hmm, it seems to fail:

```bash
2025/08/17 20:03:02 akmods: Installing newly built rpms
2025/08/17 20:03:02 akmods: DNF not found, using YUM instead.
/usr/sbin/akmods: line 362: yum: command not found
2025/08/17 20:03:02 akmods: Could not install newly built RPMs. You can find them and the logfile in:
```

imaGuru, 2025-08-25:
@amrit1711 Here are my logs. Drivers from proprietary pacman package nvidia 580.76.05-4.
`NVreg_DynamicPowerManagementVideoMemoryThreshold=512` : 

[nvidia-bug-report-threshold512.log.gz](https://github.com/user-attachments/files/21964346/nvidia-bug-report-threshold512.log.gz)

`NVreg_DynamicPowerManagementVideoMemoryThreshold=0` : 

[nvidia-bug-report-threshold0.log.gz](https://github.com/user-attachments/files/21964348/nvidia-bug-report-threshold0.log.gz)

Video of the bug with threshold 512:

https://github.com/user-attachments/assets/3b7685da-4621-413e-b25c-83c74fe19f67

Kimiblock, 2025-11-11:
I found out this problem too. Though in my case the 40 series GPU only periodically wakes up instead of waking up right after suspend. Something is definitely broken there.

ryanabx, 2026-03-17:
> > Someone with the affected hardware (i.e. a Lenovo legion 16aph8 or 16aph9) needs to confirm that the issue is fixed.
> 
> In my case unfortunately, your script does not work (Razer Blade 14 2021 with 3080 - RZ09-0370). I'm investigating a ACPI workaround as proposed [here](https://github.com/NVIDIA/open-gpu-kernel-modules/issues/860#issuecomment-4071793570), i'm waiting for OP's answer.

I think there may be similar, but separate issues at play here.

OP, myself and several others here seem to have an issue with the dGPU not going to D3Cold in a very specific pattern, where it will try every 10-15 seconds to suspend and immediately come back. The issue seems to be specific to the 16APH8 and 16APH9 legion devices. The workaround script I made fixes this issue.

Other people have come onto this thread with similar issues, but affecting other vendors and GPUs that are not the 4060 (of course all nvidia though), they won't see any benefit from the script I put together.

I wonder if we need to open a separate issue for whatever other commenters are seeing so we don't dilute either issue, as they appear to be separate issues.

ryanabx, 2026-03-26:
> **I'm almost certain _E10 GPE are triggered by the hdmi sound subdevice and snd_hda_intel module**

This would probably reveal my lack of knowledge on debugging this thing, but why is `snd_hda_intel` in use on an AMD system? 😅

imaGuru, 2026-03-29:
You are right that it should work out of the box, but forcing lenovo and nvidia to fix it doesn't seem like a viable option. 
It does seem that to have windows equivalent support there would probably have to be changes to BIOS, kernel and the driver. I can leave it open unless someone from nvidia says that it is strictly a firmware problem.

The workaround I posted above does seem to solve all issues though. For me going into advanced bios menu was not an option as it is unavailable on new lenovo bioses. I even extracted new ifr forms from my BIOS and collected the GUIDs, but they are gated differently on the new bios and would require finding a new unlock hack. Also Ive reconstructed the new forms and they do not have this D3Cold setting anymore.

I do not see however how enabling D3cold in bios would change anything, because d3cold can already be achieved without it on dgpu and other pcie devices (therefore there is nothing to enable). It could be that you changing PCI-e resource handling could affect triggering _E10 and disable dGPU hotplug functionality which would essentially be my dsdt fix but from the other side. Whatever works I guess.

Mulukulum, 2026-08-20:
Hi @ryanabx, if it is relevant, on my laptop it seems like the DMI System Version is 16AHP9 instead of 16APH9 as in the commits referenced above. It seems like 16APH8 is correct though.
```
System Information
	Manufacturer: LENOVO
	Product Name: 83DH
	Version: Legion Slim 5 16AHP9
```

ryanabx, 2026-08-20:
> Hi @ryanabx, if it is relevant, on my laptop it seems like the DMI System Version is 16AHP9 instead of 16APH9 as in the commits referenced above. It seems like 16APH8 is correct though.
> ```
> System Information
> 	Manufacturer: LENOVO
> 	Product Name: 83DH
> 	Version: Legion Slim 5 16AHP9
> ```

That does help, thank you! I've been working with a kernel maintainer to get a patch ready to fix this with a DMI quirk.

Lyude, 2026-08-20:
You may have already guessed, but me and Ryan have been working on getting a proper fix for this into the upstream kernel. It should be ready for testing, it's the top patch here (you're on your own for instructions on building the kernel, I've got too much other work to do today):

https://gitlab.freedesktop.org/lyudess/linux/-/commit/7c48d301679ce43668cc25eacbb0eb67c0b89581

This isn't for OpenRM specifically, but it should fix any nvidia hardware driver including OpenRM. This disables the PM notifier for the culprit GPE here, which -should- be fine From nouveau's perspective the only thing we care about wakeup events for is connector hotplugging, and those come across as `ACPI_VIDEO` events for us which aren't blocked by this patch). I'm not sure if OpenRM uses the same `ACPI_VIDEO` event as us, but I definitely don't think it uses the GPE either (as evidenced by the fact people here have had success with simply disabling the GPE from userspace). However, testing would be appreciated. The things to test:

* Making sure the GPU goes into D3Cold when it's inactive and stays there as long as it remains inactive, obv.
* Making sure it still goes into D3Cold when it's inactive even after a full system suspend/resume cycle (and hibernate, if you can test it).
* Testing display connector hotplugging while the GPU is in D3Cold. As in, if your laptop has connectors which you know are routed directly to the discrete nvidia GPU, allow the GPU to idle into D3Cold and then try plugging a monitor into it to make sure it wakes up accordingly and detects the monitor. This is pretty much the only actual wake-up event we care about from the GPU, and it shouldn't be affected by this patch but we should definitely make sure.

Once I can hear back from ryan or someone else regarding testing on the nouveau side, I will be sending this upstream. Testing of this patch with OpenRM would also be appreciated.

Mulukulum, 2026-08-21:
Would this patch require us to set
```
options nvidia NVreg_DynamicPowerManagementVideoMemoryThreshold=0
```
to prevent oscillation between D3Cold and D0?  Or is this expected to fix that as well?

Mulukulum, 2026-08-27:
@ryanabx, After some more testing I can confirm that the bug I was seeing was not due to this patch, and was another problem in some other software. The problem seems to have gone away with a recent package update on my distro, and I am able to observe the correct behavior with both the latest Nvidia OpenRM (Driver 610) and nouveau with this patch on kernel 7.1.10. The GPU consistently goes into D3Cold with just `options nvidia NVreg_DynamicPowerManagement=0x02` set, and I observe no oscillation.

Lyude, 2026-08-27:
> [@ryanabx](https://github.com/ryanabx), After some more testing I can confirm that the bug I was seeing was not due to this patch, and was another problem in some other software. The problem seems [...]

Oh that's awesome! Thank you for confirming this

Mulukulum, 2026-09-07:
I believe it's safe to say that the above patch closes this issue. 

Would there be any further testing required of the patch before it can be put into the kernel? I'd be happy to help with testing any other drivers/scenarios etc.

ryanabx, 2026-09-07:
> I believe it's safe to say that the above patch closes this issue. 
> 
> Would there be any further testing required of the patch before it can be put into the kernel? I'd be happy to help with testing any other drivers/scenarios etc.

I think it just needs to be reviewed, but I'm new to kernel development, so I don't know when it will be reviewed 🤷‍♂️

jindalpratik, 2026-09-18:
> You may have already guessed, but me and Ryan have been working on getting a proper fix for this into the upstream kernel. It should be ready for testing, it's the top patch here (you're on your own [...]

Hi i seem to be facing the same bug on my Lenovo LOQ 15AHP9 (83DX), which has the same RTX 4060 (10de:28e0). The dGPU runtime-suspends and wakes again after about 1.5 s. Every time, exactly one ACPI GPE (gpe10, the wake event for GPP0) fires. Disabling it with echo disable > /sys/firmware/acpi/interrupts/gpe10 keeps the GPU in D3cold. Could the quirk table be extended with:
```
	{
		.matches = {
			DMI_EXACT_MATCH(DMI_SYS_VENDOR, "LENOVO"),
			DMI_MATCH(DMI_PRODUCT_VERSION, "LOQ 15AHP9"),
		},
	},
```

ryanabx, 2026-09-18:
> Hi i seem to be facing the same bug on my Lenovo LOQ 15AHP9 (83DX), which has the same RTX 4060 (10de:28e0). The dGPU runtime-suspends and wakes again after about 1.5 s. Every time, exactly one ACPI [...]

I don't see why not. Have you built the patch with your hardware matched with the quirk to verify it works? If so, @Lyude maybe we do a v2?

jindalpratik, 2026-09-21:
> I don't see why not. Have you built the patch with your hardware matched with the quirk to verify it works? If so, [@Lyude](https://github.com/Lyude) maybe we do a v2?

Yep already built it and confirmed it works.

inigo001, 2026-09-24:
> You may have already guessed, but me and Ryan have been working on getting a proper fix for this into the upstream kernel. It should be ready for testing, it's the top patch here (you're on your own [...]

I have a Lenovo Legion Slim 5 16APH8 (82Y9) but in this case with a GeForce RTX 4070 Max-Q / Mobile [10de:2860] that has been facing the same issue. I have applied the patch changing the Device ID to the one my GPU has (2860) and it seems to be working properly now, with it staying in D3Cold until I execute something that requires the dGPU.

```
DECLARE_PCI_FIXUP_CLASS_FINAL(PCI_VENDOR_ID_NVIDIA, 0x2860,
                              PCI_CLASS_DISPLAY_VGA, 8,
                              quirk_nvidia_dgpu_broken_gpe);
``` 

Could this also be added to take the 4070 into account?

Thank you very much for finding the fix for this bug.