NVIDIA/open-gpu-kernel-modules #912: Tag for 580.65.06 missing

sndirsch, 2025-08-04:
### NVIDIA Open GPU Kernel Modules Version

575.57.08

### Please confirm this issue does not happen with the proprietary driver (of the same version). This issue tracker is only for bugs specific to the open kernel driver.

- [x] I confirm that this does not happen with the proprietary driver package.

### Operating System and Version

openSUSE Leap 16.0

### Kernel Release

6.12.0

### Please confirm you are running a stable release kernel (e.g. not a -rc). We do not accept bug reports for unreleased kernels.

- [x] I am running on a stable kernel release.

### Hardware: GPU

Device: pci 0x25b9 "GA107GLM [RTX A1000 Laptop GPU]"

### Describe the bug

github Tag for 580.65.06 driver release is still missing. 

### To Reproduce

Search in github for a release 580.65.06

### Bug Incidence

Once

### nvidia-bug-report.log.gz

...

ptr1337, 2025-08-04:
It commonly takes few hours till it is available.

sndirsch, 2025-08-04:
I see. Thanks a lot! I was just wondering because I've seen that the date of the driver packages for SUSE in the [CUDA repository](https://developer.download.nvidia.com/compute/cuda/repos/sles15/x86_64/) were from July 28. So I thought: maybe they forgot to tag it (again).